prusa3d/Prusa-Firmware #3447: [BUG] Printer permanently crashes and is not able to finish print

devproof, 2022-03-18:
**Printer type** - MK3S, MK3, MK2.5S, MK2.5, MK2S, MK2
**Printer firmware version** -  3.10.1 

**MMU upgrade** - no MMU

**SD card or USB/Octoprint** SD Card

**Describe the bug**
The printer randomly crashes at different points and I was never able to finish the attached print. I tried it four times and even resliced and moved the position of the object. The slicer is PrusaSlicer 2.4.0. This was introduced with 3.10.1 and only happens on this firmware version. After rolling back to 3.10.0, I was able to print this model 6 times in a row without any failure.
  
**To Reproduce**
Print attached gcode with PETG. Sometimes it fails after 6 hours, sometimes after 20. I believe it ends up in a crash and reboots the printer.
  
**Expected behavior**
No crash.
  
**G-code**
  Please attach a G-code. This will make it easier for us to replicate the error.

**Video**
  Please attach a video. It usually helps to solve the problem.
[Kassetten_0.2mm_PETG_MK3S_22h13m.zip](https://github.com/prusa3d/Prusa-Firmware/files/8306663/Kassetten_0.2mm_PETG_MK3S_22h13m.zip)

wavexx, 2022-03-19:
How did it crash? By self-restarting showing a "FIRMWARE CRASH!"?

djcb52, 2022-03-19:
Hi 
I have this issue. At various stages the print stops feeding filament and the print has to be stopped or stops before finishing. Printed the same file with the same filament on my Flashforge Adventurer 3 perfectly. I even disconnected my MMU2 thinking it was causing the issue. No difference.
Very frustrating. Going back to previous firmware.

wavexx, 2022-03-22:
These behaviors can be quite different internally, so it would be helpful to be extra-precise if you saw this happening.

A printer crash should self-restart and show a "FIRMWARE CRASH!" message.

If the printer stops feeding filament, does it keep on moving or it completely stops?

djcb52, 2022-03-22:
Hello

The printer head keeps moving.

I'm going to re burn the printer and MMU firmware.

Any other suggestions welcomed.

Regards
Dale J Brown
Email: ***@***.***
Mobile: 0403 055 485
Postal Address : PO Box 421 Maldon Vic 3463

wavexx, 2022-03-22:
It helps tremendously if you write down everything you do. After flashing the FW, the printer resets.

From this point, make a note of every single action (push encoder -> enter sd menu -> select file) until the printer starts printing. If you can record a video while starting the print that's also excellent.

The reset is important, as it brings the printer to a known state. If you do something unrelated before, reset the printer before recording.

Keep the original gcode you used to start the print and attach it as well.
When it crashes, also make a video if possible, so that we can try to understand the crash behavior.

We suspect these crashes are due to some other bug which is triggered before the print actually starts, but finding where is not easy. I tried to print the above gcode twice, but it didn't result in a crash. Any extra detail helps.

djcb52, 2022-03-22:
Thanks for the tip!

I'm on holidays for a few days. When I get back, I'll try what you
suggested.

Regards
Dale J Brown
Email: ***@***.***
Mobile: 0403 055 485
Postal Address : PO Box 421 Maldon Vic 3463

ldboehm, 2022-03-27:
I just updated my PRUSA MK3S, and am having this same issue.
However, it only seems to happen on very wide prints.  It crashes after a few layers but always in the same spot when the carriage is always to the extreme far right side

wavexx, 2022-03-27:
@ldboehm the same as above applies. Please attach the gcode of the print, and if you can also take a video of how the printer crashes. If there's a "FIRMWARE CRASH!" message we can also get the internal crash log from the printer with your help.

djcb52, 2022-03-27:
Will do.

Regards
Dale J Brown
Email: ***@***.***
Mobile: 0403 055 485
Postal Address : PO Box 421 Maldon Vic 3463

ldboehm, 2022-03-28:
> 

I just got it to happen again.
Print stops
Error on the printer is: "ERR: PRINT FAN ERROR"
This print is a PETG print with no cooling turned on for the extruder. the cooling fan for the carriage was on and spinning.

wavexx, 2022-03-28:
@ldboehm that's completely different from what I was expecting. Can you attach the gcode of the print that shows this?

ldboehm, 2022-03-28:
And it happens at the same spot.
I'm trying it now again with filament cooling turned on.
This is basically a rectangular shape, so it passed this position many times as it raises up.
[Archive.zip](https://github.com/prusa3d/Prusa-Firmware/files/8366190/Archive.zip)

No obstruction on the fans.

ldboehm, 2022-03-29:
> @ldboehm that's completely different from what I was expecting. Can you attach the gcode of the print that shows this?forget my post for now.  I found the fan problem. A wire disconnected from the front fan on the fan board itself.

closed 2024-07-09 by github-actions[bot]