prusa3d/Prusa-Firmware #3449: [ENHANCEMENT]<A way to bypass filament checking procedure(Tc) and successfully print flexible with MMU2s>

5jvm0u4, 2022-03-23:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

Enter what type of printer or upgrade the enhancement applies to.
**Printer type** - [MK3S]
**MMU Upgrade** - [MMU2S]

**Is your enhancement related to a problem? Please describe.**
   I'm always frustrated when printing flexible filament with MMU2s, since every time the filament will get tangled because of the filament checking procedure the extruder preforms, which contain some very high speed movement on the filament. After a lot of try-and-error , I found out that this line of gcode [ Tc ; ] is doing that procedure.
So I found a way to reliably load flexible to the nozzle:
```
Start G-code：
M862.3 P "[printer_model]" ; printer model check
M862.1 P[nozzle_diameter] ; nozzle diameter check
M115 U3.10.1 ; tell printer latest fw version
G90 ; use absolute coordinates
M83 ; extruder relative mode
M104 S[first_layer_temperature] ; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
M109 S[first_layer_temperature] ; wait for extruder temp
Tx ; //Brian 20220323
M190 S[first_layer_bed_temperature] ; wait for bed temp
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling
;go outside print area
G1 Y-3 F1000
G1 Z0.4 F1000
G92 E0; reset extruder position //Brian 20220323
G1 E49 F1000; load filament to the nozzle //Brian 20220323
; purge line
G1 X55 F2000
G1 Z0.3 F1000
G92 E0
G1 X240 E25 F1800
G1 Y-2 F1000
G1 X55 E25 F1400
G1 Z0.2 F1000
G1 X5 E4 F1000
M221 S{if layer_height<0.075}100{else}95{endif}
G92 E0
G1 E-1 F2100
; Don't change E values below. Excessive value can damage the printer.
{if print_settings_id=~/.*(DETAIL @MK3|QUALITY @MK3).*/}M907 E430 ; set extruder motor current{endif}
{if print_settings_id=~/.*(SPEED @MK3|DRAFT @MK3).*/}M907 E538 ; set extruder motor current{endif}
```
by only performing Tx and manually load the filament to nozzle, it'll always success, but some changes need to be made：
 **First** you need to enter MMU2's service menu and calibrate the filament s.t. it loads pass the idler about 2mm, like this
[image]
 **Second** you need to loosen your tension on the extruder idler to it's minimum possible to reliably push filament
This is all you need, next I'll explain some changes I've made to the start G-code to make this work.
```
M104 S[first_layer_temperature] ; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
M109 S[first_layer_temperature] ; wait for extruder temp
Tx ; //Brian 20220323
M190 S[first_layer_bed_temperature] ; wait for bed temp
```
Here I changed the order when Tx is called ,because if the nozzle is preheated to the correct temperature, when Tx is called, the extruder gears will turn and try to load the filament through the gears, but if not preheated, it won't turn and not load the filament at this state.
```
;go outside print area
G1 Y-3 F1000
G1 Z0.4 F1000
G92 E0; reset extruder position //Brian 20220323
G1 E49 F1000; load filament to the nozzle //Brian 20220323
```
here I load the filament manually 49mm into the nozzle
```
M221 S{if layer_height<0.075}100{else}95{endif}
G92 E0
G1 E-1 F2100
```
here I retract a little bit so you might want to add this line to the Before Layer Change G-code like this
```
;BEFORE_LAYER_CHANGE
G92 E0.0
;[layer_z]
{if layer_num == 1}G1 E1 F1500{endif}; //this line

```
  
**Describe the solution you'd like**
  A clear and concise description of what you want to happen.

5jvm0u4, 2024-01-17:
I'm not sure if this problem got fixed in the latest firmware or not, if so, this solution might be outdated.

3d-gussner, 2024-08-08:
@5jvm0u4 Can you please check if that is still an issue with FW 3.14.0 + MMU 3.0.2 firmware? If not please close this issue.

closed 2024-10-15 by github-actions[bot]