prusa3d/Prusa-Firmware #3452: [BUG] Incorrect M112 Thermal Runaway during print - ONLY if manually preheating bed *before* print

Andos, 2022-03-26:
Printer type:   MK2.5S
Firmware version:  3.8.1
Running from:  USB/Octoprint

**Background**
Heating the bed takes longer to heat than the nozzle (especially for ABS printing).
To prevent the nozzle from oozing while the bed warms up, I usually manually set the bed temperature to 100 C in Octoprint a few minutes before I start the print. This usually means they reach their target temperature at around the same time.

**Describe the bug**
During the initial phase of the print (usually a short while after the bed temp is upped to 110, before the bed actually reaches it), M112 is triggered with the message "THERMAL RUNAWAY (PREHEAT HEATBED)"
[image]

There are a few odd things to notice here:
- The error message itself: This in itself seems a bit strange. The bed has already been preheated. Now it should just be ... heating..
- The bed temperature is not 15 degrees away from the target temperature and the temp curves shows it's slowly approaching the target.
- Always happens shortly after the GCODE issues a temp increase to 110 degrees (over my previously manual set temperature of 100)
- This error doesn't happen if I DON'T set the temperature to 100 before I press "Print")

I have double checked the actual temperature with an IR temperature sensor and closely monitored the temperature in Octoprint - nothing seems to be going wrong. Cables have been inspected and seems intact all the way.

**Speculation**
This seems to be a bug in the firmware where the previously manually set temperature (100C) still lingers somewhere in the firmware's memory.
When the GCODE issues the 110C temp change the heatbed begins to increase in temperature.
The firmware maybe compares the target temperature to the *previous* 100C temperature instead of the 110 it should be comparing it against. The printer then sees the temperature keep climbing above it's (supposed) target, panics and issues the M112.
Seems to happen whenever the bed temperature is about to go above >103C. 
I have not tried yet if this happens if I preheat from the printer's own knob-interface.

**To Reproduce**
- Open Octoprint
- Select file for ABS print
- Manually set target temperature to 100C in the temperature pane.
- Wait a few minutes for the temperature to ramp up
- Start the print
- Wait for the bed temperature to increase further
- M112 issued
  
**Expected behavior**
No M112 to be fired
  
**Additional notes**
This seems to be happening a lot after a recent firmware update.
I can't be sure when exactly it started happening or if it happens every time. I just had it happening to me 5 times in a row, failing each time.
I then reset the printer, (reset button below the knob), connected in OctoPrint and started the print directly. No issues.

gudnimg, 2022-03-26:
Does this happen on 3.10.1 as well? You listed 3.8.1 in the ticket.

Andos, 2022-03-26:
My bad! Yes, the firmware version is indeed 3.10.1. Sorry for the confusion. I have updated the first post.

Andos, 2022-03-26:
I'm not entirely sure how to share the serial logs. Where can I find them?

gudnimg, 2022-03-26:
I'm not too familiar with Octoprint myself, perhaps @leptun could chime in, I know he has that setup.

Another way is to connect a computer the USB port on the Printer and use PuTTy. See the settings in this link https://github.com/prusa3d/Prusa-Firmware/issues/3321#issuecomment-997177337 This can be done without Octoprint, should be the same result.

Liborsaf, 2023-02-23:
Hello, did you somehow solved the issue? I currently having the same problem, thanks!

Prusa-Support, 2023-04-05:
Is this issue still relevant?
Would it persist on the [most recent firmware](https://github.com/prusa3d/Prusa-Firmware/releases) version?

Michele Moramarco
Prusa Research

Prusa-Support, 2023-04-25:
This issue may have been fixed over the more recent firmware releases, and we can't exclude an issue on Octoprint's side (please check with Octoprin's community).

This issue will be closed as stale.
You can leave further comments if the issue persists and you think it needs more investigation.

Michele Moramarco
Prusa Research

closed 2023-04-25 by Prusa-Support