prusa3d/Prusa-Firmware #3464: [BUG] FW doesn't build if `BABYSTEPPING` is disabled

gudnimg, 2022-04-05:
**Printer type** - Any
**Printer firmware version** -  master branch, 3.10.1 and earlier

**MMU upgrade** - N/A
**MMU upgrade firmware version** - N/A
**SD card or USB/Octoprint** - N/A

**Describe the bug**
If `BABYSTEPPING` is disabled in `Configuration_adv.h` then the firmware will fail to build. There is code in ultralcd.cpp which doesn't respect this setting. 
  
**To Reproduce**
Comment out `#define BABYSTEPPING` in `Configuration_adv.h` and try to build the firmware. I get the following result:

[image]

  
**Expected behavior**
  Firmware should build. But I'm not sure what the expected behaviour is in the LCD. Should the babystepping menu be disabled completely?

3d-gussner, 2023-11-07:
@gudnimg Still valid or can it be closed?

gudnimg, 2023-11-12:
It is still valid but this is very low priority (not sure if it's worth fixing at this point). Closing this issue for now.

closed 2023-11-12 by gudnimg