prusa3d/Prusa-Firmware #3468: [BUG]Filament Sensor N/A after update to FW 3.11

jimdenault, 2022-04-09:
**Printer type** - MK3S+
**Printer firmware version** -  3.11

**MMU upgrade** -n/a
**MMU upgrade firmware version** - n/a

**SD card or USB/Octoprint** SD
  Please let us know if you print via SD card or USB/Octoprint

**Describe the bug**
 After flashing the printer with 3.11 the Filament (IR) Sensor was shown as [N/A] and Filament Autoload was [N/A] in the Settings menu. Sensor did not appear under Support>Sensor Info. Self-Test showed wiring error with Filament Sensor. Hardware and wires were tested and good. Flashing the printer with Firmware 3.10.1 eliminated the problem.
  
**To Reproduce**
Flash my printer with firmware 3.11
  
**Expected behavior**
I expected that the properly connected and functioning IR sensor would be recognized by the printer self-test feature.
  
**G-code**
  Please attach a G-code. This will make it easier for us to replicate the error.

**Video**
  Please attach a video. It usually helps to solve the problem.

3d-gussner, 2022-04-11:
@jimdenault 
Thanks for reporting the issue.
Can you please verify few things as we have not changed any Filament Sensor related code between FW 3.10.1 and FW 3.11.0
To be able to troubleshoot your issue we need some data from you.

- Did you flash [FW3.11.0 for MK3S](https://github.com/prusa3d/Prusa-Firmware/releases/download/v3.11.0/prusa3d_fw_MK3S_MK3S+_3_11_0_4955.hex) on your **MK3S**, or maybe by accident the **MK3** firmware?

If you have flashed the correct firmware please continue with following steps: 
- Connect with via USB with Putty or Octoprint (Pronterface doesn't work) and turn on serial logging
  - Send `M115` to get the firmware version
  - Send `D3 Ax0d48 C1` to get the FSensor version stored in EEPROM
  - Re-detect filament sensor
    - Remove filament from sensor
    - On LCD goto `Settings -> HW Setup - Fsensor Detection` and follow the steps on screen
  - Send `D3 Ax0d48 C1` to get the FSensor version stored in EEPROM

Post the serial output here. **Remove the SN (Serial Number) before posting it.**
 
It should look like this (Ignore the SN update failed as this example has been done on a simulator)
```
=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2022.04.11 09:35:37 =~=~=~=~=~=~=~=~=~=~=~=
start
;S
SN update failed
echo: 3.11.0-4955
PowerUp
echo: Last Updated: Apr  7 2022 11:09:51 | Author: (none, default config)
Compiled: Apr  7 2022
echo: Free Memory: 2066  PlannerBufferBytes: 1760
echo:Hardcoded Default Settings Loaded
adc_init
Extruder fan type: NOCTUA
CrashDetect ENABLED!
FSensor ENABLED (sensor board revision: 0.4 or newer)
Sending 0xFF
echo:SD card ok
LCD status changed
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
MMU not responding - DISABLED
M115
FIRMWARE_NAME:Prusa-Firmware 3.11.0 based on Marlin FIRMWARE_URL:https://github.com/prusa3d/Prusa-Firmware PROTOCOL_VERSION:1.0 MACHINE_TYPE:Prusa i3 MK3S EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000
Cap:AUTOREPORT_TEMP:1
Cap:AUTOREPORT_FANS:1
Cap:AUTOREPORT_POSITION:1
Cap:EXTENDED_M20:1
ok
D3 Ax0d48 C1
D3 - Read/Write EEPROM
000d48  01
ok
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
Measured filament sensor high level: 4.50V
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
Measured filament sensor low level: 0.40V
echo:busy: paused for user
D3 Ax0d48 C1
D3 - Read/Write EEPROM
000d48  01
ok
```

jimdenault, 2022-04-11:
body{font-family:Helvetica,Arial;font-size:13px}Hi,I went back to my download history and saw that I had downloaded the MK3 version, not MK3S. I thought I was being very careful, too!Thanks for the quick response. So sorry for taking your time with this simple problem.Best,Jim  
        
     
     
        
            
                
                    _________________
                
            
        
        JIM DENAULT, ASCcinematographer+1 917 224 1810
        
            
                
                    
                        http://jimdenault.com

3d-gussner, 2022-04-11:
Glad we didn't break anything. Please close this issue.

closed 2022-04-11 by jimdenault