prusa3d/Prusa-Firmware #3470: [FEATURE REQUEST] LCD Brightness GCode support

PartemImperium, 2022-04-11:
If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - MK3S, MK3
**MMU Upgrade** - N/A

**Is your feature request related to a problem? Please describe.**
Back in 3.9.0 support was added to control the brightness of the LCD on MK3's. It would be nice if support could be added to control the brightness of the LCD through GCode. 
  
**Describe the solution you'd like**
 It looks like there is M256 in the marlin docs reserved for LCD brightness https://marlinfw.org/docs/gcode/M256.html.

3d-gussner, 2023-11-07:
@PartemImperium Thanks for the request. I can't promise if that will be implemented but assigned it to myself as a reminder.

3d-gussner, 2025-02-27:
Created https://github.com/prusa3d/Prusa-Firmware/pull/4851 it cost quite a bit of resources.