prusa3d/Prusa-Firmware #3483: [FEATURE REQUEST] Maintenance Reminders

matthewethomas, 2022-04-28:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

If it makes sense, enter what type of printer or upgrade the feature request applies to.
I have a MK3S, but this concept can apply to all

**Is your feature request related to a problem? Please describe.**
I just changed the nozzle on my printer after using it for about 6 months and could not believe the improvement in print quality. Then I google and find that I should be changing it out after so many print hours. Did you know the printer tracks print hours? I didn't. Digging deeper, and I realized I'm really bad about following [the regular maintenance schedule](https://help.prusa3d.com/article/i3-printers-regular-maintenance_2072). What I'd like is the ability to set maintenance reminders for the nozzle, lubrication, belts, etc. that triggers based on hitting a threshold of print hours.
  
**Describe the solution you'd like**
In the settings menu, I'd like the option to have various maintenance reminders based on a number of print hours. Some things I'd like to be reminded of would be:
- Check/Replace the nozzle 
- Clean/Lubricate the bearings
- Verify belts are tightened to spec
- Clean extruder gears

On power up, if maintenance reminders are enabled and I've hit the maintenance threshold, the printer should beep at me, and then I should see the maintenance reminder, and then be presented with two options:
- Remind me at the next scheduled maintenance time (acknowledging and dismissing the reminder)
- Remind me on next power up (snoozing the reminder)

Maintenance reminders should be enabled by default, but I should be able to turn them off individually in the settings menu

vadim-ryazanov, 2022-11-20:
At the very minimum I'd love to see plain hours in the LCD statistics menu. Instead of days and hours. Would make tracking operating time way easier in a farm setting.

Prusa-Support, 2022-11-29:
Thank you for sharing your valuable ideas.

We already plan to implement similar features. The new generation boards (like the Buddy 32-bit board) should directly support similar functionalities in the near future (see [Buddy FW 4.3.3](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.3.3), _Extended print statistics_). Later on, Prusa Connect should support similar features on other devices (like the MK3- series, based on Einsy 8-bit board).

Easy conversion of statistics data units sounds like a good idea too. Our developers will make their considerations about it.

Michele Moramarco
Prusa Research

vadim-ryazanov, 2022-12-02:
Thank you for your reply, Michele!

closed 2023-11-15 by github-actions[bot]