prusa3d/Prusa-Firmware #3489: Promote nozzle size option to main menu

ciaranmcnulty, 2022-05-04:
I have the new Revo RapidChange hot end and am consequently swapping nozzles a lot more than before.

It would be great to have the Nozzle diameter selector next to the Sheet selector in the main menu (rather than in Settings->HW Setup)

I guess if you allow in future more custom nozzle sizes (e.g. 0.35mm) it could have a similar logic to how the Sheet selector only appears if the user has configured >1 sheet in the deeper menu? 

Otherwise have an option in HW Setup like 'show on main screen'

Thanks for adding 0.8mm already 👍

3d-gussner, 2022-05-13:
@ciaranmcnulty Thanks for the feature request, but are you aware that the Nozzle diameter "only" warns or prevent you to print gcode sliced for a different nozzle size?
It doesn't adjust anything during printing.

The sheet menu on the other hand is gcode independent as it loads the z-offset Live-Z value depending of selected one.

ciaranmcnulty, 2022-05-13:
Yep, it's just a good practice to change it when you change the nozzle and it's a few menus deep

Mauxx91, 2022-12-10:
Yeah I would agree on this feature request, as I did set the warning to stop the print since it prevents my easily distracted mind to launch a print sliced for the wrong nozzle size. It would be convenient to have this settings on main menu first level as well as the sheet change, since with revo you're gonna change nozzle a lot.
Thanks!

Prusa-Support, 2023-01-10:
Thanks for your suggestion.

The feature would surely be handy if you swap the nozzle very frequently.
This would probably have more chances to happen with specific non-stock hotends, but you may still have a preferred nozzle.
If you only swap the nozzle for a few prints, only to revert back to the preferred nozzle rather soon, please consider not changing the _HW Setup_ settings but discarding the reminder for a few prints only.

Our developers will take your suggestion into consideration though.

Michele Moramarco
Prusa Research

closed 2023-11-15 by github-actions[bot]