prusa3d/Prusa-Firmware #3490: [BUG] Fix resume/pause for arc (G2/G3) commands

FormerLurker, 2022-05-04:
This relates to all printer types and all firmware up to the latest release for both SD and serial printing.

Pause/resume is broken for arc commands.  To reproduce, send a long arc command, pause in the middle of the arc, and resume.
  
Test gcode for the issue can be found [here - arcs(1).zip](https://github.com/prusa3d/Prusa-Firmware/files/8621142/arcs.1.zip)

Video of the issue, from @leptun: https://photos.app.goo.gl/duduGsUVGoHSdT7k8

leptun, 2022-05-04:
@FormerLurker 
> We might as well add support for the R form of G2/G3 while we're at it, since we'll basically have all we need to support this anyway. I'd suggest making it work as closely to the current marlin implementation as possible.

I agree, we should add also the R argument to G2

I didn't have time to read carefully the comment, but here is what I proposed to @wavexx yesterday (for reference):

leptun  1 day ago
> I think I know a way how to get rid of all the pause resume inconsistencies with both G1 and G2. But this will require that we don’t hardstop for pause. It needs to be a tiny bit softer (pause at the start of the next block). I’ll explain this in detail later on today.

leptun  1 day ago
> Pause and resume works like this at the moment:
It instantly halts the planner, it extracts the destination position of the current move (gcode_target), it saves the halt position and parks the hotend. On resume, the extruder is unparked to the saved halt position and the last gcode command is executed again, with the added note that gcode_target overwrites whatever the G1 calculates as the target position (needed for handling relative moves).
My idea is to change like this:
On pause (any event source) we set a flag for the planner. As soon as the current executing block is finished, the planner ISR checks for this flag and clears it after clearing the planner buffer (we basically signal the planner to abort on current_block end). This postponed halt has the advantage of always stopping between split segments.
Now, what advantage could this bring us? Well, we can store in the buffer just a single extra value (uint16_t). We'll call this segment index from now on. Whenever a split move is planned, this segment index is saved in the block. These index basically start from 0 and increment for each queued block (for that gcode instruction only!! It resets on the next instruction). When we soft halt the planner, we can retrieve also the index of the last segment that was printed. On resume, we return to the halt position as always and requeue the last gcode command (and execute it). Based on this last index, we can basically skip the segments which were already printed (something like this: for (int i = start_segment; i < total_segments; i++)).
This also means that we no longer need gcode_target. Instead we need gcode_start_position since on resume we need to make the parser think we are at the start of the G command, not at current_position.
This approach has the following advantages:
All G0-G3 commands will have the same save/resume mechanism
There will be no inaccuracies generated by interpolation between segments on resume. (applies to XYZ from G2/3 and Z from G0/1
This approach has the following disadvantages that need to be discussed as I don't have a perfect solution yet:
The stop is not instantaneous. This is problematic for power panic and crash detection. How would one handle this? Both need a hard_stop, but with hard_stop you break G2/3.
Needs quite a bit of testing. Also, what would the screen display while waiting for the current segment to finish? I wouldn't do an lcd update, but that would mean the lcd could be frozen for a bit while waiting for the soft_stop flag to get cleared (could be apparent on the 30mm split segments in the first layer)

@FormerLurker Any opinions about this approach?

FormerLurker, 2022-05-04:
I am starting to understand what is done during a pause!  I've been focusing on the power panic bit, since that's the difficult case.

In your approach above, wouldn't we still need to store the X, Y and Z position from the start of the G2/G3 command?  This is the bit that I'm not sure how to work around in my second suggestion above.

We could store the radius as well as the extrusion remaining (my first suggestion), stop without completing the current segment, complete the segment on resume, then re-issue the gcode and calculate the midpoint of the circle, though this requires the destination coordinates of the arc (X, Y, Z target).

I'm pretty convinced now that @wavexx's initial thoughts about storing all of this in the planner may well be required, unless we can come up with something very clever.  It wouldn't take much more EEPROM space for power panic in either case, but it will require more working ram, and some additional assignments when adding things to the planner.

I'm still holding out hope that we may be able to get away using far less memory.  G2/G3 commands with only 1 segment don't need anything special.  They will complete in the normal way, and should not be re-sent after resume.  G2/G3 commands with 2 segments would only need to store, at most, BLOCK_BUFFER_SIZE/2 starting positions, so there should be a way to reduce the memory requirements substantially.  Maybe we could create a circular buffer that is half the BLOCK_BUFFER_SIZE that holds X/Y/Z coordinates?  We'd need some way to dequeue these, possibly by adding some bit flag to the planner (Is New Arc Command)?  Then, during pause or power panic save, we'd just need to store the starting position of the the first arc in the planner.  We'd need to do something similar if we use radius, but we'd need to store the target position instead, but we could skip the Z coordinate in that case and would need to add extrusion remaining (I think).

leptun, 2022-08-15:
@FormerLurker I have a working prototype of my implementation here: https://github.com/leptun/Prusa-Firmware/tree/fix_multi_segment_pause_resume
If you have some time, please give it a try and see if you can find any issues with it. I've handled both stop and save to ram/from ram and power panic. In my quick testing it appears to be working, but there can obviously be edge cases that I didn't think of.

FormerLurker, 2022-08-15:
@leptun, Excellent!  I'll try it out ASAP.  Thank you for your time on this issue, it is much appreciated!

leptun, 2022-08-15:
I've pushed another commit. I haven't tested that one yet, so please be aware that might break everything

leptun, 2022-08-21:
@FormerLurker Aaaaand I was right. The last commit did indeed break everything. I pushed a fix for that.
Here are some stock printer builds of the branch: 
[FW3111-Build4987_83a3f59e+RESUME_DEBUG.zip](https://github.com/prusa3d/Prusa-Firmware/files/9388633/FW3111-Build4987_83a3f59e%2BRESUME_DEBUG.zip)

I also enabled temporarely `RESUME_DEBUG`. This should be helpful during testing.
About the branch where I developed this, these changes are based on top of #3552. As a result, you get quite a few more changes in the build. Those shouldn't affect the pause-resume, but you should be aware about those.
If you want to see a clean diff, use this: https://github.com/wavexx/Prusa-Firmware/compare/temp_model_check...leptun:Prusa-Firmware:fix_multi_segment_pause_resume?expand=1

PR #3566 (2022-08-24): Fix multi segment pause-resume
This PR fixes #3490.

This PR is based on top of #3552. As a result, there are a lot of unrelated commits. If you want to see a clean diff, see: https://github.com/wavexx/Prusa-Firmware/compare/temp_model_check...leptun:Prusa-Firmware:fix_multi_segment_pause_resume?expand=1
This PR will get merged to the MK3 branch after #3552 is merged some time next week.

Previously, the pause-resume worked like this: For each G0/1 command, along with each planner segment, the target destination of that command was storred so that in the event the move was interrupted, it coul [... cut]
closed 2022-08-24 by DRracer