prusa3d/Prusa-Firmware #3497: [BUG] Experimental feature "E-cool mode" has odd effects on extruder on MMU2S load sequence.

ggppjj, 2022-05-15:
**Printer type** - MK3S+
**Printer firmware version** -  3.11.0

**MMU upgrade** - MMU2S
**MMU upgrade firmware version** - 1.0.6

**SD card or USB/Octoprint**
  Octoprint

**Describe the bug**
  With E-cool mode enabled in the experimental features menu, the loading sequence causes the extruder to make a clicking sound.
  
**To Reproduce**
  With an MK3S+ MMU2S printer, enable the E-cool mode in the experimental features menu, then do a first-layer calibration on any filament, listen to the extruder during the load sequence.
  
**Expected behavior**
  No observable difference from E-cool mode being off.
  
**G-code**
  On-board first-layer calibration sufficient to reproduce issue.

**Video**
  [Video of issue on YouTube (apologies for the vertical video).](https://www.youtube.com/shorts/Sud9mxOqvwY)

5jvm0u4, 2022-09-20:
Oh, I thought that was intentional...
But yeah... it makes no sense to do that though

leptun, 2022-09-20:
Yeah... E-cool doesn't work great with the MMU since stealthchop doesn't work well with jerky motion. When the mmu is loading, it plans small moves, so there are small pauses between these moves. The new mmu implementation (3.13) should be better in this regard since the gears only start turning once the filament reaches the final loading state, so the gears shouldn't spin with no filament for more than 1-2 seconds per load. Besides this noise, I don't think there is any other side effect of using E-cool with the mmu

ggppjj, 2023-02-17:
Is there a PR I can track for this issue? Would it be acceptable for this issue to be added to the 3.13 milestone?

leptun, 2023-02-18:
Well, there is no PR for this issue, but we could add the issue to the 3.13 milestone. Here is a video of how the situation is now in the new mmu implementation (MK3 branch). While the issue is not gone, it is considerably better than before:

https://user-images.githubusercontent.com/17808203/219855193-40351dd0-6d69-436c-ab19-175d5c9f6e48.MOV

ggppjj, 2023-02-20:
That sneak peek is compelling, thanks for sharing! Looks like the firmware is on the right track to get this fixed. I may try to see if I can figure out enough of the firmware to try to have it automatically disable e-cool during MMU load/unload and submit a PR, but that's a long shot from my end. Something about the knocking strikes me as destructive in some way, even though it demonstrably is fine. Anyways, thanks again for the update!

gudnimg, 2023-03-24:
Hi 👋 I can give you feedback over the weekend regarding the new MMU FW. It would be best if you could create a draft PR, then I could share some feedback about specific lines of code in case I see some issues. I may also have some questions ;)

ggppjj, 2023-03-24:
Excellent, thank you! I should warn you in advance that I'm very much a script kiddie in this regard, my current level of experience is mainly c# related and even then there isn't much there to work off of. I don't know what I don't know, so hopefully what I am doing so far isn't a complete mess, haha. I'll finish the first pass, test the basics on my own printer just to make sure it doesn't completely fail, then submit the draft PR shortly.

ggppjj, 2023-03-24:
Draft ready! It built and my printer seems happy enough, please look it over at your leisure and let me know how it looks!