prusa3d/Prusa-Firmware #3498: [BUG]3.11 (M601) doesn't heat after resume

flyerdp, 2022-05-17:
I talked to support and they told me to report this as a *NEW* bug despite my explaining I saw it for 3.10
MK3S+
**Printer firmware version** -  3.11.0.4955

**MMU upgrade** - No MMU
**MMU upgrade firmware version** - N/A

**SD card or USB/Octoprint**
I tried this with BOTH octoprint and SD card, no change

**Describe the bug**
As far as I can tell this is the for the most part the same as the 3.10 bug #3148 
I downloaded an STL  that instructs to place a pause at a specific layer to insert some magnets.  At the pause I insert the magnets and have chosen resume from BOTH octoprint and the printers interface for each method of print.  In BOTH cases after I chose to resume the words "Pause Printing" were on the bottom of the display still despite the print head moving.  I noticed after a couple of minutes nothing was extruding from the print head.  It was not until the second attempt that I realized the nozzle temp had cooled significantly which is probably why no extrusion occurred.
  
**To Reproduce**
Insert a pause on a layer,  for my print it was at 2.15mm height.  Resume after the print being paused after at least 1 minute
  
**Expected behavior**
 I expected the print to resume and extrude filament to complete the print.
  
**G-code**
[dishwasher_sign_revised_12_0.15mm_PLA_MK3S_1h38m.zip](https://github.com/prusa3d/Prusa-Firmware/files/8704821/dishwasher_sign_revised_12_0.15mm_PLA_MK3S_1h38m.zip)

**Video**
  Please attach a video. It usually helps to solve the problem.

Prusa-Support, 2022-05-17:
Hello! Thank you for the report. Have you attempted the workaround proposed by @3d-gussner in #3148? 

"_I also tried to reproduce the issue in the MK404 simulator and everything went fine. Tried my own M601 gcode as well yours.
Also paused and resumed the printer manually in some combination of printer and Octoprint.
To use FW3.10.0 correctly with an USB host system like Octoprint you need to add in GCODE scripts

M601 in After print job is paused
M602 in Before print job is resumed
Without these I could reproduce your issues with "blank" screen.

Modified TestInsert gcode to trigger at 0.4 a pause
Copied TestInsert gcode to Octoprint
Started TestInsert gcode in Octoprint
Resuming via LCD generated a "blank" screen for few seconds and continued to print without heating.
Started another M601 test gcode from SD card with Octoprint connected

The printer stopped at the correct layer
Resuming the print from LCD resumed correctly waiting to heat up
At this moment the printer doesn't know what the host system is capable. The GCODE scripts M601 M602 is a workaround to establish kind of "two-way" communication for pause/resume.

There may be some printer/host negotiation in the future what both (printer/host) are capable of and this workaround isn't needed anymore, but for the meanwhile that is the "only" option to use M601 with USB hosts like Octoprint and FW3.10.0

Other USB host systems may work fine._"

An user also noticed it could possibly stem from a combination of 3.10.x firmware and 2.3.3 or older Prusa Slicer. Would it be solved by 3.11.0 and-or newer Prusa Slicer 2.4.2? 

Let me know, if the proposed solution doesn't work we'll review it in even more detail. 

Alessandro Pantaleo
Prusa Research

flyerdp, 2022-05-18:
I am still fairly new to 3D printing. Unfortunately the instructions you
wrote here and that were in the 3148 issue thread were not clear to me at
first.  After some research on what the goal was to resolve this I added
M602 in the before job resumes octoprint script section as that is what I
deciphered from the 3148 conversations and it worked now.  I now see you
asked me to add M601 in the before print section but that did not seem to
be necessary.  I am using the latest prusaslicer and that is what it said
it was doing when I went to a specific layer and chose the add pause (M601)
so perhaps that is not necessary anymore.

I thought I had used the SD card to print when this was failing but I don't
believe I did now.  I have it on SD card and I put it in the printer but I
am so used to sending to octoprint via API that I think I just did it the
second time it failed before making the bug report.    In hindsight I have
been having to turn off the RPI port on the printer to use the SD card and
I did not do that which is what makes me think I sent it to octoprint after
all.

Thanks,
-=David

gudnimg, 2022-05-18:
FYI I believe this Pull Request will solve your problem https://github.com/prusa3d/Prusa-Firmware/pull/3481 though I don't have Octoprint to confirm the fix.

The issue appears if you:
1. Stopped the print with M601 (This turns off the extruder and it will cool down)
2. And then resumed the print while printing via USB. (here the bug is the firmware does not turn on the extruder to heat back up).

This can be resolved manually in gcode with M602 as that is the same behaviour when using an SD card.

HolgerHU, 2022-05-18:
The same bug also with my 3 MK3S/S+ printers. Thanks for this post. Even with the work around #3418, the hotend cools down permanently. But there may be printers where it works as desired. A speciality? Firmware 3.9.3 is the only one that works as expected on my printers.
I therefore use a g-code script that causes a pause in octoprint or Repetier-Server. The connected printer and its firmware are then irrelevant.

flyerdp, 2022-05-19:
As stated above I am running the newest prusaslicer which says it is inserting M601 on a pause at a layer.

I connected to the octoprint script section and in the "Before Print Job is Resumed" script block I added M602.  See the attached image.

This resolved the issue while printing through octoprint and unpausing via LCD knob and octoprint
[image]
.

HolgerHU, 2022-05-19:
Thanks for the tip, but unfortunately that doesn't work with the Repetier-Server. Octoprint may behave a little differently. I will try it.

Prusa-Support, 2022-05-27:
Hello! Sorry for the late reply here from my side, thank you for your feedback. Good to hear this did the trick for Octoprint so far, thank you for the feedback regrading Repetier-Server, I brought it to our devs attention for them to evaluate and review this aspect. 

Alessandro Pantaleo
Prusa Research

HolgerHU, 2022-05-27:
Thx @Prusa Support for the consideration.
Octoprint should handle the required "action-command" correctly from version 1.3.9 onwards. Repetier-Server currently does not process the "action commands" paused and resumed as required for the M601. Therefore instead of M601 at Repetier-Server I use the special server command ;@pause in the GCode file, together with some GCodes for the X, Y and Z movements in the event-dependent GCodes for manual control of the printer during the pause and resume. Not as elegant as using the M601, but works for me.
[RepRap Action Commands](https://reprap.org/wiki/G-code#Action_commands)

Holger

kai-osthoff, 2022-06-06:
I can confirm the behaviour at:
Octoprint 1.8.1, Python 3.7.3, OctoPi 0.18.0, Prusa MK3S Firmware 3.11.0

Same issue. After Pausing via LCD or Octoprint it doesn't heat up the nozzle again.

Added M602 at "resume after pausing" in Octoprint fixes that issue, but the "Blank LCD screen" is still there for about two or three seconds.

Kai

Prusa-Support, 2022-07-12:
The "blank screen" was observed by our developer-collaborator too but doesn't seem concerning - it is not concerning as long as the screen restores itself when refreshes a few seconds later.

About Repetier-Server or any other kind of serial communication, these are "hosts" which would override the "firmware logic", while the firmware would only execute commands.
Even in this case, the user should find a suitable configuration to allow the host (Octorpint, Repetier-Server, or similar..) to send the codes `M601` after print pause and `M602` before resuming.

The problem seems to be a matter of configuring the "host" as emerged in issue #3148.
- This [comment](https://github.com/prusa3d/Prusa-Firmware/issues/3148#issuecomment-843442610) explains the solution.
- This [comment](https://github.com/prusa3d/Prusa-Firmware/issues/3148#issuecomment-855201971) explains the problem and gives examples of "more elaborate integrations" of the pause process. Another user in the same thread also found useful the integration of a "dwell" via `G4` command to ensure the temperatures are reached
- The video in this [comment](https://github.com/prusa3d/Prusa-Firmware/issues/3148#issuecomment-966463702) demonstrates how to improve the pause configuration in Octoprint, and shows how it 

The reason why this issue was first observed on fw 3.10 is yet to be clarified but fw 3.11 may actually have simply inherited this.
After double-checking this and the older issues, this actually seems to be a duplicate of issue #3148.
If you agree, please close this issue and feel free to contribute with your feedback and thoughts on the already existing issue.

Michele Moramarco
Prusa Research

closed 2023-11-15 by github-actions[bot]

andrew-manger, 2025-08-10:
This error still exists