prusa3d/Prusa-Firmware #3499: distinguish between "c" filament changes and "p" pauses when reporting "time until" in the display

krhackbarth, 2022-05-17:
was Issue #8206.

I have prints where I pause the print to insert an RFID tag and then stop the print later for a color change.  Right now, the display shows "time until" each item with a "c".   I'd like more information by indicating time until pause with a "p" and time until filament change with a "c".  You could even rotate through all three (including time remaining in the print "r").  I hear this has to first be supported in the firmware.

3d-gussner, 2022-05-17:
I think that the functionality like it is implemented is working fine and additional `p` would
- use more resources, which is very limited on the Atmel 2560. 
- also would need different Slicers to support it (more human resources are needed, and ??? if approved)
- and may confuse users

Imagine  hh:mm`c` ==> time to `C`hange filament or time for Intera`c`tion with printer

arekm, 2022-08-27:
Possible workaround - add ```M117 Last operation was XXX``` (like pause or color change) to pause and color change g-codes in slicer. But no idea how long M117 message is visible on lcd.

gudnimg, 2022-08-27:
> But no idea how long M117 message is visible on lcd.

The message should stay on the screen until the print is resumed. Though this only valid on the `MK3` branch (FW 3.12)

arekm, 2022-08-27:
Ok, so this workaround will not work then.

Prusa-Support, 2022-09-17:
Thank you for participating in our community.

Please consider closing this issue if your all questions have been answered.

Michele Moramarco
Prusa Research

krhackbarth, 2022-09-17:
sounds like it can't be done.

closed 2022-09-17 by krhackbarth