prusa3d/Prusa-Firmware #3503: [BUG] 3.11.0 Printer requests resend of very old line during crash recovery

bdaroz, 2022-05-22:
**Printer type**
MK3S+ w/ MMU2S

**Printer firmware version** 
3.11.0 / 1.0.6

**SD card or USB/Octoprint**
Octoprint

**Describe the bug**
On a paused (M601) and since resumed print if a CRASH is detected and attempted to recover the printer sometimes will request a resend of a very old line.
  
**To Reproduce**
Not 100% reliable. On one print this happens about 1 in 3 times.
  
**Expected behavior**
The printer should not ask to resend an ancient line.
  
**G-code (Octoprint terminal)**
```
Send: N128823 G1 X137.461 Y154.511 E.10675*102
Recv: ok
Send: N128824 G1 X140.25 Y154.464 E.09441*91
Recv: ok
Send: N128825 G1 X141.905 Y154.486 E.05604*99
Recv: ok
Send: N128826 G1 X144.432 Y154.597 E.08561*96
Recv: echo:enqueing "CRASH_DETECTEDY"
Recv: LCD status changed
Recv: tmc2130_home_enter(axes_mask=0x01)
Recv: T:210.0 /210.0 B:60.1 /60.0 T0:210.0 /210.0 @:29 B@:2 P:0.0 A:42.8
Recv: echo:busy: processing
Recv: T:209.9 /210.0 B:60.2 /60.0 T0:209.9 /210.0 @:31 B@:0 P:0.0 A:42.6
Recv: echo:busy: processing
Recv:   0 step=45 mscnt= 731
Recv: tmc2130_goto_step 0 61 2 1000
Recv: tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x01
Recv: tmc2130_home_enter(axes_mask=0x02)
Recv: T:210.0 /210.0 B:60.1 /60.0 T0:210.0 /210.0 @:29 B@:11 P:0.0 A:42.6
Recv: echo:busy: processing
Recv: T:210.0 /210.0 B:60.2 /60.0 T0:210.0 /210.0 @:29 B@:0 P:0.0 A:42.5
Recv: echo:busy: processing
Recv:   0 step= 2 mscnt=  32
Recv: tmc2130_goto_step 1 17 2 1000
Recv: tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x02
Recv: echo:enqueing "CRASH_RECOVER"
Recv: ok
Send: N128827 G1 X146.181 Y154.718 E.05934*106
Recv: T:210.2 /210.0 B:60.2 /60.0 T0:210.2 /210.0 @:26 B@:0 P:0.0 A:42.7
Recv: echo:busy: processing
Recv: T:210.2 /210.0 B:60.1 /60.0 T0:210.2 /210.0 @:27 B@:14 P:0.0 A:42.7
Recv: echo:busy: processing
Recv: T:210.5 /210.0 B:60.0 /60.0 T0:210.5 /210.0 @:21 B@:25 P:0.0 A:42.5
Recv: echo:busy: processing
Recv: T:210.5 /210.0 B:60.0 /60.0 T0:210.5 /210.0 @:22 B@:24 P:0.0 A:42.7
Recv: Resend: 107982
Should resend line 107982 but no sufficient history is available, can't resend
Changing monitoring state from "Printing" to "Error"
Send: M112
Send: N128828 M112*24
Send: N128829 M104 T0 S0*25
Send: N128830 M104 T1 S0*16
Send: N128831 M104 T2 S0*18
Send: N128832 M104 T3 S0*16
Send: N128833 M104 T4 S0*22
Send: N128834 M140 S0*81
Changing monitoring state from "Error" to "Offline after error"
Connection closed, closing down monitor
Closing down send loop
```

Prusa-Support, 2022-05-27:
Hello! Thank you for the report, I brought this aspect to our devs attention for them to evaluate and attempt to reproduce the issue. I'll let you know on this thread for any updates on the situation. 

Alessandro Pantaleo
Prusa Research

wavexx, 2022-08-20:
Looks like this is indeed a bug. In absolute terms resend shouldn't ask for a line which is >100 from the last number sent (this is 16 entries from the planner buffer + cmdqueue which depends on the line size).

The skip here is about 20k. I cannot reproduce it locally though.

Do you have a little more details in how this happens?

Crash detected while the printer is moving back from a pause/filament change (and before it starts to print)?

bdaroz, 2022-08-20:
Apologies - it's been a bit since this has been happening and I've avoided the issue by simply not using OctoPrint and instead printing from the SD Card. If memory serves...

These prints were ~2hrs or so, with a pause about 85% of the way in (pause to embed something in the print). The print would resume (after adjusting the hotend temp and waiting first -- the issue that resume w/ OctoPrint never resumed the temp - but that's another issue and mentioned here only for completeness) and continue printing for some time - generally a few minutes, but I do not recall a specific point that reliability triggered the crash.

At some not-too-distant point a Y-Axis crash would be detected, with no clear reason why (the print area is at least 1-2cm from the edge of the normal printable area in all dimensions), the printer would re-home each axis and then request the ancient history from OctoPrint which would (rightly?) freak and kill the print job.

I'm sorry you can't reproduce it locally, I can't reliably do so either or I would have absolutely attached a test case. I hope this helps in some way.

wavexx, 2022-08-20:
I think the description is good enough. Do you have the g-code that you were using maybe? I'm not saying it's relevant (it shouldn't), but sometimes it does help.

bdaroz, 2022-08-20:
Unfortunately not that I can share publicly. :(

closed 2023-11-15 by github-actions[bot]