prusa3d/Prusa-Firmware #3507: [BUG] Hotend temperature reported with invalid name in serial communication

crysxd, 2022-05-24:
**Printer type** - MK3S
**Printer firmware version** -  3.11.0

**MMU upgrade** - Unknown
**MMU upgrade firmware version** - Unknown

**SD card or USB/Octoprint**
OctoPrint

**Describe the bug**
With OctoPrint 1.8.0 a new feature was introduced to display additional temperatures besides hotend and bed. Prusa MK3 have in multiple instances started to show additional temperatures with a name following a pattern `O*T` (leading `O` before the T). Those are send once by the printer instead of the `T` temperature (see logs below) and seem to be a random combination.

The below logs are collected over a day and filtered for lines containing `OOOT`, meaning almost all temperature lines correctly use only `T` for the hotend and below are exceptions.

```
2022-05-23 13:17:41,229 - Recv: FIRMWARE_NAME:Prusa-Firmware 3.11.0 based on Marlin FIRMWARE_URL:https://github.com/prusa3d/Prusa-Firmware PROTOCOL_VERSION:1.0 MACHINE_TYPE:Prusa i3 MK3S EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000
2022-05-22 17:30:38,575 - Recv: OOOOOOOOOOOOOOT:199.8 /200.0 B:60.0 /60.0 T0:199.8 /200.0 @:64 B@:38 P:0.0 A:35.6
2022-05-22 17:55:35,591 - Recv: OOOOOOOOOOOOOOT:199.9 /200.0 B:60.0 /60.0 T0:199.9 /200.0 @:67 B@:37 P:0.0 A:35.4
2022-05-22 18:05:53,558 - Recv: OOOOOOOOOT:199.9 /200.0 B:60.0 /60.0 T0:199.9 /200.0 @:74 B@:34 P:0.0 A:35.4
2022-05-22 18:08:28,078 - Recv: OOOOOOOOOOOOOOOOOOOOOOOT:200.0 /200.0 B:60.0 /60.0 T0:200.0 /200.0 @:69 B@:27 P:0.0 A:35.6
2022-05-22 18:22:04,988 - Recv: OOOOOOOOOOOOOOOT:200.0 /200.0 B:60.2 /60.0 T0:200.0 /200.0 @:61 B@:17 P:0.0 A:35.5
2022-05-22 18:26:31,935 - Recv: OOOOOOOOOOOOOOOOOOOOOOOOOOT:200.0 /200.0 B:60.0 /60.0 T0:200.0 /200.0 @:76 B@:29 P:0.0 A:35.5
2022-05-22 18:45:01,672 - Recv: OOOOOOOOOOOT:200.0 /200.0 B:60.2 /60.0 T0:200.0 /200.0 @:75 B@:0 P:0.0 A:35.6
2022-05-22 19:03:57,759 - Recv: OOOOOOOOOOOOOOOOOOOOOOOT:200.1 /200.0 B:60.1 /60.0 T0:200.1 /200.0 @:75 B@:24 P:0.0 A:35.6
2022-05-22 19:07:02,381 - Recv: OOOOOOOOOOOOOOOOOOOT:199.9 /200.0 B:60.2 /60.0 T0:199.9 /200.0 @:78 B@:12 P:0.0 A:35.6
2022-05-22 19:23:29,827 - Recv: OOOOOOOOOOOOOOOOT:200.7 /200.0 B:60.0 /60.0 T0:200.7 /200.0 @:69 B@:52 P:0.0 A:35.7
2022-05-22 19:36:38,675 - Recv: OOOOOOT:200.5 /200.0 B:60.2 /60.0 
```

Disclaimer: I do not own the printers in question but the additional temperature readings were reported to me as a bug in my app OctoApp, after investigation the root cause turned out to be the above issue
  
**To Reproduce**
Not known
  
**Expected behavior**
Hotend temperatures are always indicated by `T` instead of `OOOO...T`
  
**G-code**
Will deliver soon

**Video**
/

crysxd, 2022-05-24:
As a small addition, the leading `O` also seem to affect other lines, potentially causing other issues as well

```
2022-05-22 11:30:03,141 - Recv: OOOOOOO succeeded.
2022-05-22 11:33:21,242 - Recv: OOOOOOOOOOOOOOOOOOOOOOOOOOOOOO succeeded.
2022-05-22 11:35:45,923 - Recv: OOOOOOOOOOOOOecho:busy: processing
2022-05-22 11:35:46,136 - Recv: OOOOOOOOOOOT:200.1 /200.0 B:60.3 /60.0 T0:200.1 /200.0 @:84 B@:0 P:0.0 A:35.3
2022-05-22 11:35:46,247 - Recv: OOOOOO succeeded.
2022-05-22 11:39:10,853 - Recv: OOOOOOOOOOOOOOOOOOOOT:199.8 /200.0 B:60.2 /60.0 T0:199.8 /200.0 @:75 B@:12 P:0.0 A:35.5
2022-05-22 11:39:11,046 - Recv: OOOOOOOOOO succeeded.
2022-05-22 11:41:36,001 - Recv: OOOOOOOOOOOOOOOOOOOOOOOOOOOOOO succeeded.
2022-05-22 11:45:14,124 - Recv: OOOOOOOOOOOOOOOOOOOOOOOT:200.1 /200.0 B:60.3 /60.0 T0:200.1 /200.0 @:69 B@:11 P:0.0 A:35.5
2022-05-22 11:45:14,243 - Recv: OOOOOOO succeeded.
2022-05-22 11:47:41,164 - Recv: OOOOOOOOOOOOOOOOOOOOOOOOOOOOO succeeded.
2022-05-22 11:51:12,656 - Recv: OOOOOOOOOOOOOOOOOOOOOOOOOOOOOO succeeded.
2022-05-22 11:53:40,360 - Recv: OOOOOOOOOOOOOOOOOOOOOOOOOOOOO succeeded.
2022-05-22 11:57:12,252 - Recv: OOOOOOOOOOOOOOOOOOOOOOOOOOOOOO succeeded.
2022-05-22 12:00:00,448 - Recv: OOOOOOOOOOOOOOOOOOOOOOOOOOOOOO succeeded.
2022-05-22 12:03:31,882 - Recv: OOOOOOOOOOOOOOOOOT:199.5 /200.0 B:60.2 /60.0 T0:199.5 /200.0 @:75 B@:23 P:0.0 A:35.6
2022-05-22 12:03:32,144 - Recv: OOOOOOOOOOOOO succeeded.
2022-05-22 12:06:17,338 - Recv: OOOOOOOecho:busy: processing
2022-05-22 12:06:17,776 - Recv: OOOOOOOOOOOOOOOOOOOOOOO succeeded.
2022-05-22 12:09:58,016 - Recv: OOOOOOOOOOOOOOOOOOOOOOOOOOOOOO succeeded.
2022-05-22 12:12:51,664 - Recv: OOOOOOOOOOOOOOOOOOOOOOOOOOOOOO succeeded.
2022-05-22 12:16:32,429 - Recv: OOOOOOOOOOOOOOOOOOOOOOOOOOOOOO succeeded.
2022-05-22 12:18:44,088 - Recv: OOOOOOOOOOOOOOOOOOOOOOOOOOOOOO succeeded.
2022-05-22 12:20:44,352 - Recv: OOOOOOOOOOOOOOOOOOOOOOOOOOOOOO succeeded.
2022-05-22 12:23:33,286 - Recv: OOOOOOOOOOOOOOOOOOOOOOOOOOecho:busy: processing
2022-05-22 12:23:33,359 - Recv: OOOO succeeded.
2022-05-22 12:25:50,249 - Recv: OOOOOOOOOOOOOOOecho:busy: processing
2022-05-22 12:25:50,405 - Recv: OOOOOOOOT:200.0 /200.0 B:60.2 /60.0 T0:200.0 /200.0 @:72 B@:6 P:0.0 A:35.4
2022-05-22 12:25:50,524 - Recv: OOOOOOO succeeded.
2022-05-22 12:29:39,161 - Recv: OOOOOOOOOOOOOT:200.0 /200.0 B:60.2 /60.0 T0:200.0 /200.0 @:68 B@:19 P:0.0 A:35.5
2022-05-22 12:29:39,472 - Recv: OOOOOOOOOOOOOOOOO succeeded.
2022-05-22 12:32:07,520 - Recv: OOOOOOOOOOOOOOOOOOOOOOOOOOecho:busy: processing
2022-05-22 12:32:07,585 - Recv: OOOO succeeded.
2022-05-22 12:35:56,427 - Recv: OOOOOOOOOOOOOOOOOOOOOT:200.4 /200.0 B:60.2 /60.0 T0:200.4 /200.0 @:63 B@:14 P:0.0 A:35.4
2022-05-22 12:35:56,550 - Recv: OOOOOOOecho:busy: processing
2022-05-22 12:38:32,699 - Recv: OOOOOOOOOOOOOOOOOOOOOOOOOOOOOO succeeded.
2022-05-22 12:42:24,850 - Recv: OOOOOOOOOOOOOOOOOOOOOOOOOOecho:busy: processing
2022-05-22 12:42:24,907 - Recv: OOOO succeeded.
2022-05-22 12:45:00,495 - Recv: OOOOT:199.8 /200.0 B:60.0 /60.0 T0:199.8 /200.0 @:74 B@:38 P:0.0 A:35.3
2022-05-22 12:45:00,995 - Recv: OOOOOOOOOOOOOOOOOOOOOOOOOO succeeded.
2022-05-22 12:48:48,448 - Recv: OOOOOOOOOOOOOOOOOOOOOOOOOOOOOO succeeded.
2022-05-22 12:51:20,571 - Recv: OOOOOOOOOOOecho:busy: processing
2022-05-22 12:51:20,923 - Recv: OOOOOOOOOOOOOOOOOOO succeeded.
2022-05-22 12:55:11,342 - Recv: OOOOOOOOOOOOOOOOOOOOOOOOOOOOOO succeeded.
2022-05-22 12:57:42,507 - Recv: OOOOOOOOOOOOOOOOOOOOOOOOOOOOOO succeeded.
2022-05-22 13:01:21,391 - Recv: OOOOOOOOOOOOOOOOOOOOOOOecho:busy: processing
2022-05-22 13:01:21,510 - Recv: OOOOOOO succeeded.
2022-05-22 13:03:49,996 - Recv: OOOOOOOOOOOOOOOOOOOecho:busy: processing
2022-05-22 13:03:50,192 - Recv: OOOOOOOOOOO succeeded.
2022-05-22 13:07:35,790 - Recv: OOOOOOOOOOOOOOOOOOOOOOOOOOOOOO succeeded.
2022-05-22 13:10:05,542 - Recv: OOOOOOOOOOOOOOOOOT:200.0 /200.0 B:60.3 /60.0 T0:200.0 /200.0 @:70 B@:5 P:0.0 A:35.5
2022-05-22 13:10:05,656 - Recv: OOOOOOecho:busy: processing
2022-05-22 13:10:05,787 - Recv: OOOOOOO succeeded.
2022-05-22 13:13:47,781 - Recv: OOOOOOOOOOOOOOOOOOOOOOOOOOOOOO succeeded.
2022-05-22 13:16:16,087 - Recv: OOOOOOOOOOOOOOOOOOOOOOOOOecho:busy: processing
2022-05-22 13:16:16,168 - Recv: OOOOO succeeded.
2022-05-22 13:20:13,567 - Recv: OOOOOOOOOOOOOOOOOOOOT:200.2 /200.0 B:60.2 /60.0 T0:200.2 /200.0 @:72 B@:8 P:0.0 A:35.6
2022-05-22 13:20:13,751 - Recv: OOOOOOOOOO succeeded.
2022-05-22 13:22:42,974 - Recv: OOOOOOOOOOOOOOOOOOOOOOecho:busy: processing
2022-05-22 13:22:43,126 - Recv: OOOOOOOO succeeded.
2022-05-22 13:26:40,979 - Recv: OOOOOOOOOOOOOOOOOOOOOOOOOOOOT:200.0 /200.0 B:60.0 /60.0 T0:200.0 /200.0 @:76 B@:31 P:0.0 A:35.6
2022-05-22 13:29:10,026 - Recv: OOOOOOOOOOOOOOOOOOOecho:busy: processing
```

crysxd, 2022-05-24:
Thanks for the quick reply! This is a bit weird to say the least, but great to know where it comes from.

crysxd, 2022-05-24:
Actually, after thinking a bit more about this issue, I still believe this should be treated as a bug. Issue #2326 is about a separate line `OOOOOOOO succeeded` which is absolutely fine. This ticket is about `OOOOO` showing up in the temperature reports, which should be a standardised reply. Polluting this reports with debug information for proprietary hardware breaks interoperability between Prusa printers and other software like OctoPrint.

Prusa-Support, 2022-05-30:
Hello! Thank you for the additional feedback, I brought this to our devs attention to further review it. 

Alessandro Pantaleo
Prusa Research

gudnimg, 2023-01-10:
Adding to this issue:

With the MMU rewrite in upcoming v3.13 the “O”/“o” log is removed.

gudnimg, 2023-06-22:
Adding this to the 3.13 Milestone as a reminder for us to close this issue once FW 3.13 is released 👍

gudnimg, 2023-07-31:
Closing as this is now resolved with FW 3.13.0 and MMU FW 3.0.0 :)

closed 2023-07-31 by gudnimg