prusa3d/Prusa-Firmware #3510: [QUESTION]Additional temperature readings

crysxd, 2022-05-26:
**What is your question?**

In the terminal logs I can see additional temperature values reported for `E`, `W`, `P` and `A` next to `T` (hotend) and `B` (bed). I did not find any documentation on what those temperatures are?

Example:
```
Recv: T:199.5 /200.0 B:59.8 /60.0 T0:199.5 /200.0 @:76 B@:46 P:0.0 A:35.6
Recv: T:199.3 /200.0 B:59.7 /60.0 T0:199.3 /200.0 @:78 B@:61 P:0.0 A:35.6
Recv: T:199.4 /200.0 B:59.7 /60.0 T0:199.4 /200.0 @:76 B@:60 P:0.0 A:35.6
Recv: T:199.5 /200.0 B:59.8 /60.0 T0:199.5 /200.0 @:75 B@:53 P:0.0 A:35.6
```

gudnimg, 2022-05-26:
Does this help?
[image]

crysxd, 2022-05-28:
That helps a lot already, thanks! Do you also happen to know what `E` and `W` is? They don't seem to always show up.

leptun, 2022-05-28:
`E` and `W` appear while waiting for a heater to reach target temperature. `E:` says the active extruder number (should say 0). 'W:' is a countdown for after the target temperature is reached. The current temperature has to be at the target temperature for 3 seconds (cofigurable at compile with TEMP_RESIDENCY_TIME). If the current temperature is not at the target temperature, then `W:?` is shown instead.

crysxd, 2022-05-28:
So `E` and `W` aren't temperatures at all. Thanks! for the info!

closed 2022-05-28 by crysxd