prusa3d/Prusa-Firmware #3511: [FEATURE REQUEST] Report print status for prusa connect or other loggers

jokohoko, 2022-06-08:
At the moment prusaConnect works great if you print trough the interface. But if you start a print from the SD card it cannot show any progress or print time in the interface. 
Likewise i wanted to program a monitor that keeps track of printer usage in our farm at the university to detect issues or to know how bussy they are. But there does not seem to be any Gcode to provide insight in the print currently printing. It could be enough to periodically print out the time remaining information or the filename of the active print. But preferably this should need a separate gcode.

If i missed anything i would like a reference to the documentation. But i did not find any insight into this information.

And later on this would also improve the interface of prusaconnect since it now shows ERROR if someone starts a print from the SD card.
[image]

jokohoko, 2022-06-09:
I stand corrected. On some files it does show the information...
Can anyone here tell me what gcode is used then to provide this information?

I will be logging communication between the two this weekend to find out.
So this request can probably be closed. But it would be nice to have official documented support for this.

Prusa-Support, 2022-06-15:
Hello! Thank you for the report and sorry for the late response. A possible suspect can be that some of the GCodes that didn't show that info could have been not sliced with Prusa Slicer, or with an older version of Prusa Slicer, or it could be that the function "supports remaining times" could have been deactivated for these prints. Reference screenshots below. 

Alessandro Pantaleo
Prusa Research

[image]
---
[image]

jokohoko, 2022-06-17:
There is a Gcode for this it turns out. M27. So the bug is more on the prusaConnect side where it is not always shown or requested correctly.

closed 2022-06-17 by jokohoko

Prusa-Support, 2022-06-22:
Hello! Thank you for your response and for the report on this aspect - I'll move the thread to the Prusa Connect section, it can be good for our devs to notice and evaluate it. 

Alessandro Pantaleo
Prusa Research