prusa3d/Prusa-Firmware #3512: [BUG] MK2.5S + 3.11.0 == immediate FIRMWARE CRASH (Watchdog timeout)

fornellas, 2022-06-11:
**Printer type** - MK2.5S
**Printer firmware version** - prusa3d_fw_MK25S_3_11_0_4955_RAMBo13a

**MMU upgrade** - N/A
**MMU upgrade firmware version** - N/A

**SD card or USB/Octoprint**
  Octoprint

**Describe the bug**
  When firmware 3.11.0 is flashed, as soon as the printer boots, I get the message:

```
FIRMWARE CRASH!
Crash reason:
Watchdog timeout
```
 Reverting to `prusa3d_fw_MK25S_3_9_3_3556_RAMBo13a_en-cz` gets the printer back to working state.
  
**To Reproduce**

- Get an MK2.5S.
- Go from firmware 3.9.3 to 3.11.0.
  
**Expected behavior**
  Firmware upgrade to work.
  
**G-code**
  N/A.

**Video**
https://photos.app.goo.gl/uZ3UpfysMVAu1XZBA
[image]

fornellas, 2022-06-11:
PS: this seems to be affecting more people: https://forum.prusa3d.com/forum/original-prusa-i3-mk2-5s-mk2-5-hardware-firmware-and-software-help/mk2-5s-3-10-1-firmware-crash/.

Prusa-Support, 2022-06-13:
Hello! Thank you for the report and sorry for the issue. We were aware of this for 3.10.1, we'll look into it in more detail for the 3.11.0 to have it fixed as soon as possible! 

Alessandro Pantaleo
Prusa Research

wavexx, 2022-06-14:
@fornellas this happens all the time right? Is there any output coming from the serial when the crash happens?

wavexx, 2022-06-20:
Thanks a lot. If it emits some output during start, it's easier since you need to send D23 to enable crash dumps on the serial. However, I suspect this happens before the serial is ready. Given the 2.5 doesn't have an internal xflash, we might need your help to flash some firmware that initializes the serial earlier so that we can get the dump.

fornellas, 2022-06-21:
NP, I'll try the serial output tonight.
Question: IIRC those AVRs only have state at its EEPROM, so, assuming you can't repro this bug with another machine, I'd guess the bug is likely triggered by some either bad state at the EEPROM / bad handling of valid EEPROM state. Would it be helpful for me to share a dump of my EEPROM?

wavexx, 2022-06-21:
It *could* be. Yes, eeprom dump would be helpful.

However it could also could get stuck inside a TWI receive which never times out when checking for the filament sensor. This should have been fixed in the current sources (3.11.1 branch if you know how to rebuild from source), so it's also something you can try to narrow down the problem.

wavexx, 2022-06-21:
Another test you could do, to verify the above, is try to unload the filament first (or keep it loaded) before flashing the new version. This changes the timeout behavior of filament sensor detection logic, so you might be able to process further. If that's the case, we know it's that piece of code that gets stuck.

rtaylorgarlock, 2022-08-24:
I'm on 3.11 w/ mk2.5s and am currently stuck with this issue, though it didn't start immediately after a firmware upgrade. I've had many hours of successful printing since then. I'll try a clean flash. Anything else I should do to avoid sensor issues?

wavexx, 2022-08-24:
So far we partially debugged this on a MK3S and the FW to seems to get stuck inside the spi_txrx call. The reason is most likely an interrupt performing something on the spi bus right in the middle of the transfer, and again most likely this is related to the temperature isr, but we're still investigating exactly what is going on and get a board that does this to reproduce it. The timing here is critical, since often just rebuilding the code fixes the issue by shuffling some instructions around :(

@rtaylorgarlock does your mk2.5s stay alive long enough with FW 3.11 to perform a crash dump over the serial, or does it keep rebooting? If we can get the dump it would help us confirm this is the same issue. Did you notice anything that could have happened before it started to crash?

Meanwhile in the current MK3 branch we overhauled completely the temperature regulation in an attempt to solve this and other issues (see #3552). Knowing if the bug is still present here would be tremendously useful. If you want to help us test and need help rebuilding the source let us know.

wjcampbe, 2022-08-27:
I have just had this issue after upgrading from 3.11.0 RC1 to 3,11,0 full while troubleshooting a layer shift (turned out it was a model error). I did fw upgrade then removed SD to load the repaired and resliced model. Error appeared on SD insertion.

rtaylorgarlock, 2022-09-16:
> So far we partially debugged this on a MK3S and the FW to seems to get stuck inside the spi_txrx call. The reason is most likely an interrupt performing something on the spi bus right in the middle [...]

@wavexx, my unit did stay alive long enough to perform a serial dump, although I don't know how to do that currently. Are instructions available? Can't be that hard! 😊
I've had a few successful prints on older firmware, though I'm betting I can reproduce the issue on current firmware.

wavexx, 2022-09-18:
> though I'm betting I can reproduce the issue on current firmware.

That would actually be much easier to debug, since the changes between 3.11 and the current FW are quite large at this point.

In any case, on a MK2.5 you need to add "D23" somewhere to the start gcode or send it before starting a print. You need to print with a serial logger attached (octoprint, printcore on linux or putty). You don't actually need to print via usb, you just need to have serial logging. When a crash happens, the dump is immediately streamed on the serial.

There are some instructions here: https://help.prusa3d.com/article/dump-memory-mk3-s-and-mk2-5-s_245237 however if you're on windows do *not* use pronterface, as we discovered it just truncates the serial output, corrupting the dump.

Tanemahuta, 2022-10-09:
Bump: Any news on the issue?

leptun, 2022-10-09:
3.11.0 has been... special. There have been numerous unexplainable issues with that firmware that simply went away in the meantime. I suggest that anyone who had an issue like this should switch to [3.12.0-beta1](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.12.0-BETA1). This version switched to a newer avr-gcc compiler (fixed some code generation bugs in the compiler we already encountered a few releases ago) and also has the rewritten thermal control (safer code, might fix some of the unexplainable bugs). Debugging 3.11.0 makes little sense now considering we can't reproduce the same bugs at all on newer versions.

Tanemahuta, 2022-10-09:
Thank you for the clarification.

phptanker, 2022-10-11:
Hi, just got this issue after printing daily for about a month, then powering off the printer for a few days. 

The printer booted, beeped loudly and looked like it was being flashed (progress bar over the display). Now I get the error on each boot.

leptun, 2022-10-12:
> (progress bar over the display)

Do you have a MK2.5/s? If yes, then a crash will cause this sort of bars because of a bug in the bootloader (gets stuck in the bootloader on crash).

kchauvin, 2022-10-26:
I have a MK2.5S and got this crash after printing for a week with 3.11.  I just flashed the beta 3.12 and I'm still getting this error on boot.  My UART logs are below:

`echo: 3.12.0-BETA1-5536
echo: Last Updated: Sep 29 2022 20:36:59 | Author: (none, default config)
Compiled: Sep 29 2022
echo: Free Memory: 2851  PlannerBufferBytes: 1792
echo:Stored settings retrieved
adc_init
FSensor ENABLED
echo:SD init fail
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
`

Prusa-Support, 2023-01-11:
Hello @kchauvin, thank you for reporting this and sorry for the late response from my side. Can you please attempt the final 3.12.0 version, to see if it would still give the error? https://github.com/prusa3d/Prusa-Firmware/releases

With 3.9.3 it was working, but was it possible to print via SD as well? Or only via USB? If still not possible to print via SD in spite of not having that error "SD init fail" on older FW version, it might point to an issue with either SD card or the SD card reader - some more info to narrow things down on this guide: https://help.prusa3d.com/article/sd-card-not-working_2095

Alessandro Pantaleo
Prusa Research

Prusa-Support, 2023-01-12:
Hello, thank you for your update! Good to see that factory reset and then flashing 3.12.0 seems to have worked, let's keep an eye on the situation of course, but it seems like good progress! 🙂 

Alessandro Pantaleo
Prusa Research

fornellas, 2023-01-27:
Getting back to this @wavexx  / @Prusa-Support, sorry for the delay.

I've just tested current latest and this is what I get at serial output (along side with a similar crash message on the panel):

```
start
echo: 3.12.1-5686
echo: Last Updated: Jan 20 2023 10:31:39 | Author: (none, default config)
Compiled: Jan 20 2023
echo: Free Memory: 2858  PlannerBufferBytes: 1792
echo:Stored settings retrieved
adc_init
FSensor ENABLED
echo:SD init fail
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
```

I was forced to revert to 3.9.3.

> Another test you could do, to verify the above, is try to unload the filament first (or keep it loaded) before flashing the new version. This changes the timeout behavior of filament sensor [...]

The previous error was with filament loaded. Same issue with filament unloaded.

> There are some instructions here: https://help.prusa3d.com/article/dump-memory-mk3-s-and-mk2-5-s_245237 however if you're on windows do not use pronterface, as we discovered it just truncates the serial output, corrupting the dump.

Sadly, it seems this option is not available on 3.9.3 which I'm stuck with.

I managed to do an EEPROM dump with:

```
avrdude -v -q -p m2560 -c stk500v2 -P /dev/ttyACM0 -D -b 115200 -U eeprom:r:/tmp/eeprom.hex:i
```

[eeprom.hex.gz](https://github.com/prusa3d/Prusa-Firmware/files/10522724/eeprom.hex.gz)

Hopefully it could be useful for debugging.

---

What I just ended up trying out is the [factory reset procedure](https://help.prusa3d.com/article/full-system-refresh-original-prusa-i3_133258). I trtied these 3 with no success:

- Shipping prep
- Service prep
- Statistics

But finally, "All data" make it work. Sadly, it means I completely lost all the statistics I've accumulated over the years...

leptun, 2023-01-28:
Unfortunately the eeprom dump is unusable. The bootloader that is shipped with MK2 and MK3 printers has a bug which renders the eeprom dumping and writing useless. If you look at the hex dump, you will see how the dump restarts from the beginning at address 0x800. See https://github.com/prusa3d/stk500v2-prusa/pull/2 for more details.

It's unfortunate that the bug was solved by a full factory reset. It would have been good to know exactly which setting caused the firmware to crash. But in any case, it's good that the issue was solved in the end.

Due to the corrupted dump, the only recoverable statistic is EEPROM_FILAMENTUSED. EEPROM_TOTALTIME is unusable unfortunately. You should be able to write it to the printer with this gcode command: `D3 Ax0fed C4 X3ac70100`. Make sure it was written successfully (check the response from the printer and check also the menu after restarting the printer).

leptun, 2023-01-28:
@fornellas Please see the updated command. I made a typo in the initial post

fornellas, 2023-01-28:
@leptun thanks for the details. Not sure I understand how bootloader would influence EEPROM dumping with avrdude (in my mind those wore independent things), but the value 0x3ac70100 (which I bet you got from address 0x7F9 from the dump) is 1165.38m, which is exactly what I had before, and `D3 Ax0ff1 C4 X3ac70100` worked like a charm, TY. I'll use `D3` to restore some other stats.

Though sadly, as I played around with restoring the EEPROM, it seems the corruption bug ended up zeroing the total print time, and that's lostt forever at this point, but meh, I'll take what I can. Thanks!

---

@leptun  for debugging the root cause, I believe you still have a shot. Before I knew the EEPROM dump / restore was half broken, I played around with it. I've downgraded to 3.9.3 and restored the EEPROM dump with:

```
avrdude -v -q -p m2560 -c stk500v2 -P /dev/ttyACM0 -D -b 115200 -U eeprom:w:/tmp/eeprom.hex:i
```

and... the crash bug is back! So I'm guessing if you get a vanilla MK2.5S with 3.9.3 and restore the dump (as half broken as it is), you'd likely be able to trigger the same bug, and cut a fix (though, it is also posible some other junk at the half broken restore trtiggers the same watchdog bug...).

leptun, 2023-01-28:
Since the crash is still happening, the issue should not be closed.

> Not sure I understand how bootloader would influence EEPROM dumping with avrdude (in my mind those wore independent things)

Avrdude talks directly with the bootloader. It doesn't talk with the firmware when reading/writing the flash/eeprom. Since the bootloader is the one that has the eeprom read bug, avrdude can't be used for dumping the eeprom (but it should work for writing a valid hex file back). If you need to dump the eeprom, use `D3`, or better yet use the dump_eeprom and update_eeprom tools from https://github.com/prusa3d/Prusa-Firmware/tree/MK3/tools if you can.

closed 2023-11-15 by github-actions[bot]

numo68, 2023-11-26:
I have a similar issue with 3.13.2; in my case 3.12.2 works fine. See #4513

Prusa-Support, 2023-12-23:
Firmware crashes may have multiple reasons so we should focus on the reproducibility of the problem, and logs can help too.
The firmware crash that you experienced with the most recent firmware is probably not related to the old ones reported in this thread so thanks for opening a new issue.
This one will remain closed as obsolete.

Michele Moramarco
Prusa Research