prusa3d/Prusa-Firmware #3518: [Community made] <German translation: Troubleshooting -> FINDA setup and troubleshooting

blumauer, 2022-06-28:
There may be a translation mistake on the Point "Faulty sensor" ... "the best way... insert a **piece of filament**"
in German: "Am besten... indem Sie ein **Stück Draht** in den Selektor einführen"
a **"Stück Draht**" is a piece of metal - **a cable** - may this can make issues due to any magnetic properties.
Greetings, Blumauer

Prusa-Support, 2022-07-14:
Thank you for your feedback.

Because 'Prusa Research' is the direct maintainer of the German translations, I will report your suggestion to the competent department. 

Michele Moramarco
Prusa Research

3d-gussner, 2023-11-24:
Closing as https://help.prusa3d.com/de/article/finda-einrichtung-und-fehlerbehebung_2250 has been updated

closed 2023-11-24 by 3d-gussner