prusa3d/Prusa-Firmware #3525: `avr-gcc: error: ...avr6.xn: No such file or directory` when compiling on Windows

jeffamelang, 2022-07-02:
I followed the instructions to build with Arduino on Windows and attempted to run `Sketch->Verify/Compile` _before making any modifications_, and received this error:

```
Archiving built core (caching) in: C:\Users\<username>\AppData\Local\Temp\arduino_cache_702913\core\core_PrusaResearch_avr_prusa_einsy_rambo_34f3e465d1daf5b0fb33eac52c962d7b.a
avr-gcc: error: <username>\AppData\Local\Arduino15\packages\PrusaResearch\hardware\avr\1.0.4/ldscripts/avr6.xn: No such file or directory

exit status 1
Error compiling for board PrusaResearch Einsy RAMBo.
```

I made sure to use the `1.8.5` version of the Arduino IDE, as the instructions specifically said. 

I see that there was [a commit](https://github.com/prusa3d/Prusa-Firmware/commit/98fa74a0d7e9ae91bb583f51b424d68fab8b4d96) to "Change to Arduino IDE 1.8.19 and Arduino boards 1.0.5", do the instructions need to be updated?

jeffamelang, 2022-07-02:
Ah, I believe this actually has to do with there being a space in my Windows username.  Sigh, I strongly dislike doing development on Windows. Does anyone know of a workaround for this, other than making a new Windows user with a name that doesn't have a space?

leptun, 2022-07-02:
What code are you building? Recently, we added support for arduino 1.8.19 in the MK3 branch, but the readme wasn’t updated yet. You may try that arduino version with the latest board version. Keep in mind that the MK3 branch is the development branch, so be prepared to encounter some bugs. The latest arduino might work even with some of the prior releases, but there were some bugs we also fixed when we switched to the new arduino.

jeffamelang, 2022-07-02:
I had literally followed the instructions on the README without deviation, so I did not change branches and the default branch appears to be `MK3`. 

I've checked out branch `MK3_3.11.1` and tried to compile and have the same error. I strongly suspect that this has to do with having a space in the Windows username, as my username is "Jeff Amelang" and the error is 

```
avr-gcc: error: Amelang\AppData\Local\Arduino15\packages\PrusaResearch\hardware\avr\1.0.4/ldscripts/avr6.xn: No such file or directory
```

Note the `error: Amelang` part that omits the first part of the path that includes my first name. 

Perhaps we can add quotes around paths in whatever build instructions the Arduino IDE is using? Do you know what file contains the build instructions, that I could mess with? I'm not familiar with this type of Arduino development.

leptun, 2022-07-02:
Try doing this:
Go to `"C:\Users\Jeff Amelang\AppData\Local\Arduino15\packages\PrusaResearch\hardware\avr\1.0.4\boards.txt"`
And change
```
prusa_einsy_rambo.compiler.c.elf.extra_flags=-Wl,-u,vfprintf -lprintf_flt -lm -Wl,--script={runtime.platform.path}/ldscripts/avr6.xn
```
to
```
prusa_einsy_rambo.compiler.c.elf.extra_flags=-Wl,-u,vfprintf -lprintf_flt -lm -Wl,--script=./ldscripts/avr6.xn
```
The problem is that the path contains spaces. This would be fine if we were able to add quotes around each dir name in the path, but that is not possible. The only quotations that can be added are around the entire `{runtime.platform.path}`, which has the unintended side effect of treating `\` as an escape character. There really isn't anything that can be done about `runtime.platform.path` since that's how arduino gives it to us on Windows. Other paths provided by arduino are properly defined and contain `\\` instead of `\`, so these can be added in quotes without side effects.
Using a relative path instead of the absolute path provided by arduino is a hack and it might not work on other platforms, so this needs to be tested more.

closed 2023-11-15 by github-actions[bot]