prusa3d/Prusa-Firmware #3532: M702 Unload raises extruder too much

timur-tabi, 2022-07-04:
MK3 upgraded to MK3s
3.11.0-4955

When I ask my printer to unload the filament, first it raises the extruder.  IMHO, it raises the extruder too much, perhaps about 4-5cm.  This often puts the extruder close to the of top the Z-axis, making it harder to load new filament.

I'm pretty sure that this is a change during the last year or so.  When I first got my MK3, it only raised the extruder about 2cm, which is much more reasonable.

wavexx, 2022-07-09:
Yes, we indeed increased the unload raising distance to make unloading easier when Z is low since it's more convenient to clean the nozzle in that case.

I fail to see how raising close to the top makes it any harder though?

timur-tabi, 2022-07-09:
1) You are raising the Z even if Z isn't already low.  If the extruder is already near the top of the printer, you still try to raising it 4-5cm.  If you're saying that it's only supposed to raise the extruder that much only when "Z is low", then there is a bug.

2) My printer is stored inside an enclosure, and trying to load filament when the extruder is that high is difficult.

wavexx, 2022-07-09:
Makes sense. We can definitely improve this to raise only a minimal height, unless below a minimum amount (say Z<5cm). I though this was already the case, but it looks like not all conditions are covered in the same way.

timur-tabi, 2022-12-13:
Any chance that this bug will be fixed soon?  I was hoping to see a fix in 3.12

gudnimg, 2022-12-20:
In 3.13 there is now an option to use the Z parameter with M702. For example M702 Z10. Any Z-lift done by M702 is also reverted when the unload is finished. 

This won’t make it into 3.12, there are simply too many changes.

timur-tabi, 2022-12-22:
How will that work with the control panel on the printer itself?

Prusa-Support, 2023-01-10:
Thanks for your interest.
I guess that once the the code `M702` is enabled, it can be dynamically implemented via _custom g-code (start, end, color change, pause print...)_ in Prusa Slicer.

Michele Moramarco
Prusa Research

timur-tabi, 2023-01-10:
Why would I use the slicer to unload filament?  I need this feature in the control panel of the printer itself.

thatdecade, 2023-01-10:
I too see this issue on 3.11.0-4955.  Works as @wavexx describes, the z-axis doesn't move during unload when z position is >5cm, unless I power down my machine between jobs.

To duplicate the issue, I can print any object taller than 5cm. Then power cycle the machine and select unload from control panel.  The z-axis raises regardless of current height.

To keep this feature, the following should be considered:

- Machine thinks z position = 0 on power up.
- If the previous print ended within 5cm of the top, the head will crash during unload (probably ok).
- If the printer is in an enclosure, a z position higher than 10cm is hard to reach with your hand.
- After unload performs a z-movement, there is no way to move the z position back down (hold button for live adjust z is disabled). Instead I must cancel the unload, manually move axis down, and attempt unload again.

timur-tabi, 2023-03-30:
So you're saying that I could issue the M702 command every time the printer is turned on, and then the firmware will use that value whenever I unload from the control panel?

I don't want the firmware to raise the nozzle at all when unloading the printer.  But I send "M702 0", won't that just immediately unload the filament?

Prusa-Support, 2023-04-21:
Not necessarily - that's not what I meant.
The parameter could be dynamically modified via custom g-code (start, end, color change, pause print...) in Prusa Slicer but that was just an idea.

Most importantly, the Z parameter for M702 is not implemented yet (possibly in future 3.13 firmware version).
Please wait for it to be implemented. Until then, I'm afraid all we can do is guesswork.

Michele Moramarco
Prusa Research

Prusa-Support, 2023-08-19:
With the latest release 3.13.0, the firmware now supports additional parameters for `M701`/`M702` codes:
- Z [Z move]
- L [extrusion length]
- T [extruder number] (to be used in combination on multi-extruder printers)
- P [mmu slot index] (to be used in combination on multi-extruder printers)
- S [save filament name] (for internal purposes)
- W [preheat value] (for internal purposes)

The parameter `Z` can be used to define a custom relative Z parking position.

This should enable the firmware to set a parametric "raise Z" value which may vary dynamically (I'm not aware of currently used parameters in FW 3.13.0).
The `Z` can't be defined via LCD panel but allows custom applications (via [insert custom g-code](https://help.prusa3d.com/article/insert-pause-or-custom-g-code-at-layer_120490), [code substitutions](https://help.prusa3d.com/article/g-code-substitutions_301694), serial line...)

Michele Moramarco
Prusa Research

timur-tabi, 2023-08-19:
Is there any way to change the default Z value, so that unloading via the control panel automatically uses the new default?  I'm guessing the answer is no.  

For example, if I do "M702 Z0" from Octoprint, and then next time choose "Unload" from the control panel, will the firmware remember the Z value from the last time M702 was executed?

Also, should I open another bug to get the control panel to use the Z option, or should we keep this bug open instead?

Prusa-Support, 2023-08-22:
Unfortunately not, it is not possible to alter the default values without building a custom firmware version, but the current implementation will open up new possibilities for future development.
For now, advanced users can send custom codes via serial line (e.g. via Octoprint) or use any of the options mentioned above to set custom codes in the print file already planned beforehand (e.g. at a certain layer height).

The default Z movement should be 50 mm
The default loading distance should be 95 mm.
The default unloading distance should be 80 mm.

When the Z and L parameters are not specified, the firmware will use the default values above.

In order to avoid confusion and for better visibility of your request, yes, I recommend closing this issue and opening a new specific one about changing these values via the printer's UI.

Please allow me to close this one already but feel free to submit further feedback below.

Michele Moramarco
Prusa Research

closed 2023-08-22 by Prusa-Support

ttabi, 2023-08-22:
https://github.com/prusa3d/Prusa-Firmware/issues/4344