prusa3d/Prusa-Firmware #3533: PLEASE ADD EXTRUDER CLICKING DETECTION!!!

TudorAndrei93, 2022-07-05:
There is nothing more frustrating that coming to the printer and see it printing in mid air because of a clog. Actually this is the most frequent problem users have, myself included. 

Can you please implement something like Crash detection, but for E axis/ extruder motor? 
It would be extremely useful.

Thank you a lot!!!!!

leptun, 2022-07-05:
Unfortunately, the crash detection (stallguard) only works when the stepper is moving at mid to high speeds. The extruder is waaaaaay too slow to be able to use that feature. The only moment where _maybe_ it could be used is while loading filament. But for the MK3 at least that also doesn't make much sense either.
Prusa did however announce in the XL video that they'll implement jam detection in the new hotend in a different way: https://youtu.be/9peACH52KTo?t=85

TudorAndrei93, 2022-07-05:
Thank you for the comment.
I know crash detection only works at mid to high speeds. 
Maybe an idea would be for example: the printer to make a safety check each layer change or every 2-3 minutes. Safety checks may consist of rapidly moving extruder maybe 4-5mm up and down with a mid speed.  I would gladly increase my print times by a few % to gain that consistent printing. 

Another fix may be for example to measure the current to stepper. I don't know if the current hardware can do that, but i am sure an in-line sensor can be added. 

Anyways, i think this is the biggest problem right now with MK3S, and i am sure a solution can be found, software and/or hardware.

Prusa-Support, 2022-07-16:
Thank you for your feedback.

Unfortunately, this is not possible with the current MK3- series hardware. 

This topic has been discussed multiple times in the past. This issue should be closed as a duplicate of #476.
Please check the [conclusion](https://github.com/prusa3d/Prusa-Firmware/issues/476#issuecomment-1126120764) of the issue and the other mentioned/duplicate issues if you'd like.

Nonetheless, there are various reasons for the extruder to start clicking, and various ways to prevent that.
I hope this [comment (#1757)](https://github.com/prusa3d/Prusa-Firmware/issues/1757#issuecomment-1133245662) helps.

This issue will be closed now. Please comment below if you think this issue deserves further studies in a separate issue, or feel free to participate in the existing conversations.

Michele Moramarco
Prusa Research

closed 2022-07-16 by Prusa-Support