prusa3d/Prusa-Firmware #3545: MMU Single Mode - Ask for extruder number for any color change

DeanDavis, 2022-07-13:
**Is your feature request related to a problem? Please describe.**
I print models all the time where I do a color change at a specific layer and maybe another change at another layer.
In single mode I set my color changes at the layer in PrusaSlicer as normal.
When I start that print the LCD asks me for the extruder to use to start the print but when it comes to the color change it expects to use the same extruder for ALL the layers and have me swap the filaments in the same extruder.
Let's say I have red in 3 and white already loaded in 2 when I start the print. I start the print on 3 because I want a red print to start. Now when the color change comes I want to print in white. It's already loaded in 2 I just want to switch to 2, have some purge come out and continue. But the printer wants me to load white into 3.

"Well why not just set up your print like that in the slicer?"
I might print this same object two days later and two days later I may have switched all my filaments for various reasons and when I come back to this print red is in 1 and white is in 5. I don't want to go back to my slicer and re-slice this whole thing.

I just want to start the print, be prompted for the extruder to start with, in this case 1 then when the color change comes be prompted for the extruder to switch to, in this case 5.

**Describe the solution you'd like**
On a sliced print using MMU Single Mode the printer should ask for the extruder number to start the print with.
Then when the printer encounters a M600 color change it unloads the current extruder and prompts for the extruder to use for the next color.
That color loads, extrudes some purge, asks if the color has changed and if "YES" then continues the print.

rtyr, 2022-07-13:
> Describe the solution you'd like
> On a sliced print using MMU Single Mode the printer should ask for the extruder number to start the print with.
> Then when the printer encounters a M600 color change [...]

This is feature request for the firmware. I transfered your issue to prusa3d/Prusa-Firmware.

gudnimg, 2022-07-13:
This looks like a duplicate of https://github.com/prusa3d/Prusa-Firmware/issues/1363 its currently on track to being included in firmware v3.12 :)

rtyr, 2022-07-14:
Indeed, I believe this can be closed then. I also added comment there, regarding M600 on regular multi-material print (where changing tools via M600 should not be possible).

closed 2022-07-14 by rtyr

DeanDavis, 2022-08-06:
This is not a duplicate but it is on the wrong repository. This is an issue with PrusaSlicer and not the firmware. I’ll post on Prusa Slicer repository.