prusa3d/Prusa-Firmware #3549: [BUG][Low priority] Lack of M84 in g-code files causes firmware crash

arekm, 2022-07-27:
**Printer type** - MK3S+
**Printer firmware version** -  3.11.0 and 3.11.1-RC1 using official hex releases

**SD card or USB/Octoprint**
Printing from SD card directly.

**Describe the bug**
Printing from g-code file (on sd card) that doesn't have M84 command causes firmware crash after processing
all existing g-code commands in file.

I've verified that all g-codes are processed by adding bunch of "M117 test" commands at the end of gcode
file to get "test "text on LCD and I got it on LCD.

**To Reproduce**
* decompress and run attached gcode file from sd card on the printer
* wait for it to finish and then observe crash 
 
**Expected behavior**
No crash.

**G-code**
This one reliably crashes for me:

[test-gcode-only-moves.gcode.gz](https://github.com/prusa3d/Prusa-Firmware/files/9198145/test-gcode-only-moves.gcode.gz)

Much less reliable but still sometimes even 3 line g-code file would also crash:
````
G90
G28 W
G1 Z50 F720 
````

**Crash dumps**
Nine crash dumps dumped by D21:
[crash-dumps.txt.gz](https://github.com/prusa3d/Prusa-Firmware/files/9198337/crash-dumps.txt.gz)

**Video**
https://www.youtube.com/watch?v=FN9xwIY07h4

leptun, 2022-07-27:
Just quickly looking at the crash log headers, the crashes were caused by stack overflow.
This is not a low priority issue and we've been investigating crashes for a while now. I'm tagging this for 3.12 and making it high priority.
@wavexx fyi

leptun, 2022-07-27:
@arekm What version was used with the crash files you provided? You mentioned it happens on two official versions, but we need to know the exact release of the dump, which sadly is not included in the crash log...

leptun, 2022-07-27:
@arekm Can you also provide a list with all the files that were on the SD card? An image of the card would be best if that is possible.

gudnimg, 2022-07-27:
@leptun I can confirm the crash on my MK3S+. (Am using `mk3-mmu` branch though). I have some spare time atm to look at this.
[image]

leptun, 2022-07-27:
I saw these messages on the stack as well in the crashes. It’s just a matter of finding where the reccursion happens in the EOF section.

gudnimg, 2022-07-27:
I can confirm this is a recursion issue. Trying to track it down as we speak :)

wavexx, 2022-07-29:
@arekm are you able to build PR #3551 and test it? Besides fixing this crash, I wonder if this has any impact on #3541

arekm, 2022-07-29:
I'll test after weekend (don't have access to the printer now). #3541 - no way to test since no way to reproduce failing build

arekm, 2022-07-31:
MK3 branch + mentioned fix doesn't crash on my test file but...

There seem to be another problem with it. After print finishes M27 command should return "Not SD printing" (octoprint sends M27 to monitor printing from SD) but that doesn't happen.

That's how it looks like with clean MK3 branch firmware:

````
2022-07-31 23:13:03,392 - Send: M27
2022-07-31 23:13:03,752 - Recv: T:25.3 /0.0 B:26.0 /0.0 T0:25.3 /0.0 @:0 B@:0 P:0.0 A:31.4
2022-07-31 23:13:04,145 - Recv: NORMAL MODE: Percent done: 100; print time remaining in mins: 0; Change in mins: -1
2022-07-31 23:13:04,156 - Recv: SILENT MODE: Percent done: 1; print time remaining in mins: 5; Change in mins: -1
2022-07-31 23:13:04,161 - Recv: NORMAL MODE: Percent done: 100; print time remaining in mins: 0; Change in mins: -1
2022-07-31 23:13:04,170 - Recv: SILENT MODE: Percent done: 100; print time remaining in mins: 0; Change in mins: -1
2022-07-31 23:13:05,587 - Recv: LA10C: Linear Advance mode: UNKNOWN
2022-07-31 23:13:05,591 - Recv: Done printing file
2022-07-31 23:13:05,597 - Changing monitoring state from "Printing from SD" to "Finishing"
2022-07-31 23:13:05,659 - Recv: echo:0 hours 0 minutes
2022-07-31 23:13:05,661 - Recv: LCD status changed
2022-07-31 23:13:05,709 - Recv: Not SD printing
````

and that on MK3 + 3551 PR:

````
2022-07-31 23:00:26,648 - Send: M27
2022-07-31 23:00:28,078 - Recv: NORMAL MODE: Percent done: 100; print time remaining in mins: 0; Change in mins: -1
2022-07-31 23:00:28,087 - Recv: SILENT MODE: Percent done: 1; print time remaining in mins: 5; Change in mins: -1
2022-07-31 23:00:28,096 - Recv: NORMAL MODE: Percent done: 100; print time remaining in mins: 0; Change in mins: -1
2022-07-31 23:00:28,100 - Recv: SILENT MODE: Percent done: 100; print time remaining in mins: 0; Change in mins: -1
2022-07-31 23:00:28,104 - Recv: Done printing file
2022-07-31 23:00:28,106 - Changing monitoring state from "Printing from SD" to "Finishing"
2022-07-31 23:00:28,256 - Recv: echo:0 hours 0 minutes
2022-07-31 23:00:28,258 - Recv: LCD status changed
2022-07-31 23:00:28,318 - Recv: T:26.1 /0.0 B:26.4 /0.0 T0:26.1 /0.0 @:0 B@:0 P:0.0 A:32.1
2022-07-31 23:00:29,546 - Recv: LA10C: Linear Advance mode: UNKNOWN
2022-07-31 23:00:30,323 - Recv: T:26.2 /0.0 B:26.5 /0.0 T0:26.2 /0.0 @:0 B@:0 P:0.0 A:31.7
2022-07-31 23:00:32,327 - Recv: T:26.1 /0.0 B:26.6 /0.0 T0:26.1 /0.0 @:0 B@:0 P:0.0 A:31.7
[all further messages are temperature data like earlier line]
````

No answer for M27. Octoprint then gets stuck in "Finishing" state waiting for ever for answer.

wavexx, 2022-08-06:
@arekm I was able to reproduce this, and I believe that is probably fixed as a result of [4f22de2](https://github.com/prusa3d/Prusa-Firmware/pull/3551/commits/4f22de2333bcd9cfd603c144dd037358fc3c9d5f). Can you try again that PR?

arekm, 2022-08-07:
Seems to be working. No crash and doesn't get stuck on M27 here.

PR #3551 (2022-08-19): Prevent re-entry in EOF command processing
cmdqueue will run commands when EOF is reached without returning to the
main loop, which is already incorrect.

However, since it needs to ensure the queue is empty, an st_synchronize
call can result in a re-entrant call to get_command, which will
reprocess EOF again. Even if we removed st_synchronize, another command
could be picked by an unsuspecting manage_inactivity() somewhere else.

Short-circuit EOF processing by clearing card.sdprinting immediately.
This is safe to do, since it's already done by printingHasFinished()
called a few lines down. W [... cut]
commit c0b5fea5 (2022-08-19): Prevent re-entry in EOF command processing

cmdqueue will run commands when EOF is reached without returning to the
main loop, which is already incorrect.

However, since it needs to ensure the queue is empty, an st_synchronize
call can result in a re-entrant call to get_command, which will
reprocess EOF again. Even if we removed st_synchronize, another command
could be picked by an unsuspecting manage_inactivity() somewhere else.

Short-circuit EOF processing by closing the file early and checking for
the file state early in get_command.

This should fix #3549
commit be3523b0 (2022-09-16): Prevent re-entry in EOF command processing

cmdqueue will run commands when EOF is reached without returning to the
main loop, which is already incorrect.

However, since it needs to ensure the queue is empty, an st_synchronize
call can result in a re-entrant call to get_command, which will
reprocess EOF again. Even if we removed st_synchronize, another command
could be picked by an unsuspecting manage_inactivity() somewhere else.

Short-circuit EOF processing by closing the file early and checking for
the file state early in get_command.

This should fix #3549
closed 2022-08-19 by DRracer