prusa3d/Prusa-Firmware #3554: [FEATURE REQUEST] Possibility to continue print halted for running out of filament after complications that lead to exiting replacing filament routine

pedjas, 2022-08-05:
**Printer type** - MK3S

**Is your feature request related to a problem? Please describe.**
The other day I had printer run out of filament and stopped. For some reason it could not load replacement. I had to fiddle with it including moving head, opening it to check for clodge, and basically exiting filament replacement procedure so it was impossible to continue print. Few hours of print that was just about to be finished went into waste basket.
  
**Describe the solution you'd like**
It would be good if, in occasion of stopping for causes like running out of filament, that could be continued, printer could remember its state in manner that it can restore state and continue printing even after had had to be moved or even printer turned off. 

I guess if print is done from SD card in moment of error, printer could be able to save current state and allow user to have new menu option to continue that interrupted print. Option could reset printer and then set remembered state and allow print to continue wherever it stopped. 

Idea is that as long as user do not move halted print from the printer bed that print could be continued if it was run from SD card.

Prusa-Support, 2022-08-11:
Hello! Thank you for the request, I passed this on to our devs for them to further review and evaluate the possibility of this addition. We'll update here in case of more info from our side along the way. 

Alessandro Pantaleo
Prusa Research

wavexx, 2022-08-20:
Would allowing to re-home XY have been sufficient in your case? To resolve clogs I also had to sometimes move the extruder, and with #2161 I could probably have saved some prints.

Allowing to transition from a "filament change/runout" to pause, then having both the above, would close the gap IMHO.

pedjas, 2022-08-20:
> Would allowing to re-home XY have been sufficient in your case? To resolve clogs I also had to sometimes move the extruder, and with #2161 I could probably have saved some prints.
> 
> Allowing to transition from a "filament change/runout" to pause, then having both the above, would close the gap IMHO.

I guess, if print is from SD card, printer knows exactly all that is needed to continue printing from the very point where it was interrupted regardless on anything that happens in meanwhile. If ti can recover from complete power loss, this not should be an issue.

wavexx, 2022-08-21:
True. Unfortunately the recovery mechanisms currently doesn't work when doing some service operations, filament change and pause included :(, mostly because these operations are multi-step and cannot really be repeated in full so there's extra work to be done to support them. During recovery it's also different if we need to save a pause, or a filament change (should the filament change will attempt to continue an interrupted feed?).

But I digress. Transitioning to pause allows a lot more stuff which is helpful to clear an issue. You have access to the entire Tune menu, which is my main point.

closed 2023-11-15 by github-actions[bot]