prusa3d/Prusa-Firmware #3555: [BUG] G28 Null Read when not supplying XYZ as parameters

toombaumarkt, 2022-08-05:
I'm currently working on a project regarding fuzzing a selection of the most widespread GCode parsing software.

When I started working on the Prusa-Firmware I immediatly stumbled upon this piece of code:

https://github.com/prusa3d/Prusa-Firmware/blob/9633f34e89494a980a716ddf64050621037dd4de/Firmware/Marlin_main.cpp#L4809

To me it seems like this line should only be executed, when the function call above returns true. Otherwise the strchr_pointer in cmdqueue.h is pointing to NULL, thus resulting in a read on address NULL + 1 in code_value_long().

And running a simple GCode file containing a single line of:
G28 X Z

does indeed lead to a SEGV on address NULL + 1 reported by UndefinedBehaviourSanitizer when testing with my modified build specifically made to fuzz the GCode parsing process of this firmware. 

To clarify a bit what I mean by "modified build": 
To be able to fuzz the GCode parsing section ( namely get_command() in cmdqueue.cpp and process_command() in Marlin_main.cpp ) I injected the fuzzed input into the cmdbuffer (cmdqueue.h) by making get_command() read from a chararray rather than a serial connection. After that I call process_command() to fully parse the current command. Apart from removing code belonging to heater management and other things I obviously don't need when not running the code on a printer, I did not modify any of the parsing related sections. Though I did remove every instance of PROGMEM, but this isn't the problem here, is it? 

Butl this piece of code still does seem a bit weird to me, seeing that everywhere else there's always an if statement checking whether strchr_pointer is not null, before calling any of the code_value() functions in cmdqueue.h. But even after looking at the code for ages, I can't tell, where I could have made a mistake, so i decided to ask here in the hope an expert comes to the rescue :D.

Thank you!

wavexx, 2022-08-06:
You are absolutely right. I sadly expect the MK3 FW to be chock-full of these cases due to the incredibly naïve g-code parser.

So in that sense it's absolutely great that you're reporting the results of the fuzzing. I'm not sure if you're aware of MK404 (https://github.com/vintagepc/MK404/), but I expect you can use it as a soft target to fuzz the entire firmware as opposed to a subset of the gcode parser.

toombaumarkt, 2022-08-07:
Alright, I will certainly share my findings when I find more of these cases. The MK404 project sounds very interesting, definitely looking into that! Thanks!

PR #3557 (2022-08-19): G28: check for X/Y/Z parameters before calling code_value_long
Fixes #3555, thanks to @toombaumarkt
commit 06d19b47 (2022-08-19): G28: check for X/Y/Z parameters before calling code_value_long

Fixes #3555, thanks to @toombaumarkt
commit 1b90394a (2022-09-16): G28: check for X/Y/Z parameters before calling code_value_long

Fixes #3555, thanks to @toombaumarkt
closed 2022-08-19 by DRracer