prusa3d/Prusa-Firmware #3556: [Community made] Add Afrikaans Language Translation

GLDRitchie, 2022-08-06:
Prusa Research will NOT follow up these issues!
The maintainers of the "Community made" feature should/will react.

Please, before you create a new "Community made" ticket, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Which Community made feature do you want to address?**
Language Translations

**What is your request/question/suggestion?**
Can an Afrikaans Language translation for the MK3S+ be added?
I am willing to develop and maintain the translation

Prusa-Support, 2022-08-08:
Hello! Thank you for your request. Unfortunately, I wouldn't be able to determine this possibility from here, but in general we welcome everyone to join us and localize the printer to their native language  - can you please send us the request through this form? https://docs.google.com/forms/d/e/1FAIpQLSc2tj4qQ2PXp7-ewg9L5u6AAI_2pmodyORAv-KwkeHv0XOepA/viewform. Once we process your request and prepare certain things, we will get back to you with further instructions.

Full info on community translation on this article: https://help.prusa3d.com/article/community-translations_282960

Alessandro Pantaleo
Prusa Research

3d-gussner, 2022-08-12:
@GLDRitchie Did you fill in the form? Please consider to close this issue.

closed 2023-10-25 by 3d-gussner