prusa3d/Prusa-Firmware #3558: [BUG] Stuck at language selection when selecting German

BeeCave, 2022-08-07:
**Printer type** - [MK3S+]
**Printer firmware version** -  [3.11.0, 3.11.1-RC1]

**USB/Octoprint**
  
**Since 3.11.0 and 3.11.1-RC1 stuck at language selection**
  Hi all,
since Firmware 3.11.0 and 3.11.1-RC1 i'm stuck in the language selection menu when choosing the german language (Deutsch). Selecting the English works. Selecting the German language in 3.10.1 works flawlessly.
  
I'm using https://plugins.octoprint.org/plugins/firmwareupdater/ to flash new firmware.

Regards
Markus

**To Reproduce**
  Flash 3.11.0 or 3.11.1-RC1 and select "German - Deutsch"
  
**Expected behavior**
  Selecting German language works and doesn't end in a loop.

arekm, 2022-08-07:
How do you update with plugin? Two avrdude commands are needed to flash properly on prusa.

That plugin doesn't do that correctly:

https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/issues/309

leptun, 2022-08-07:
You must use PrusaSlicer to update the firmware if you are using languages. The problem with firmwareupdater is that it uses regular avrdude which doesn't upload also the language data. As a result, when you update to 3.11.1-RC1 with firmwareupdater, the language data is still from the old firmware, hence why it doesn't work there, but works once you revert back to 3.11.0

BeeCave, 2022-08-07:
Thanks for the hint. I'm flashing with PrusaSLicer next time. Hopefully the plugin will get an update in the future.

3d-gussner, 2022-08-09:
@BeeCave Please consider to close this issue.

3d-gussner, 2022-08-12:
@BeeCave Can you confirm that the flashing with Prusa Slicer solved the issue? If so please close this issue.

BeeCave, 2022-08-13:
@3d-gussner Yes, flashing with Prusa Slicer solved the issue.

closed 2022-08-13 by BeeCave

arekm, 2022-08-17:
@leptun Are broken USB to serial converters still used in MK3S+? Tried to use avrdude without workaround for ';' character on my 2 months old mk3s+ and it failed, so looks like it.

Anyway for brave people:
https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/pull/325

leptun, 2022-08-17:
@arekm it is not “broken”. In the prusa 32u2 usb to serial firmware, “;” is used as an escape character for commands addressed to the 32u2 chip. This has been the standard since the MK2 and it hasn’t changed since the release of the MK3 (except some minor command expansion recently). So the “;” workaround is still needed.

What were the arguments that you provided avrdude and what avrdude version did you use?

arekm, 2022-08-17:
Comment in code says "*infamous ';' bug*" so thought it is bug.

My goal was to get stock 'avrdude' working somehow. By intializing mk3 hardware for external flashing (via char entry_magic_receive[] = "w25x20cl_enter\n") first and then calling stock avrdude.

That unfortunately won't work if ";" workaround is needed as stock avrdude doesn't have it.

```
(stock)avrdude -v -q -p m2560 -c arduino -P /dev/ttyACM0 -D -U flash:w:/tmp/tmp5ipknjf_:i -u
```
(/tmp/tmp5ipknjf_ contains only second section of firmware, that with languages)

leptun, 2022-08-17:
Try using “wiring” as the programmer type in avrdude. This is what I’m using with stock avrdude and I’ve had no upload issues (somehow)

arekm, 2022-08-17:
"-c wiring" for base firmware works well on stock avrdude for me, too but not for languages.

Just tested -c wiring for languages part. So using -c wiring with *languages part of hex only* (== second firmware section after first :00000001FF marker in hex file) and with w25x20cl_enter command.

Sadly that way it overwrites base firmware with languages data (while we want to write that into external flash). Overwriting base firmware makes printer go out of service obviously (until base firmware reflash).

(btw. PrusaSlicer uses "-c arduino" for languages)

So I guess there is no way to use stock avrdude for flashing languages.

leptun, 2022-08-18:
> so printer goes into proper mode

It may be in the correct mode, but it most likely will not be after avrdude is called. The problem is that when avrdude is called, the serial port is opened again and that triggers a reset of the printer, rendering your previous w25x20cl_enter attempts useless.

>but "wiring" mode doesn't work for external flash / languages.
(btw. PrusaSlicer uses "-c arduino" for languages)

Yeah, I think that's correct. The "arduino" protocol is simpler, so that one was used instead to save on flash.

> I guess there is no way to use stock avrdude for flashing languages.

It might be possible to use it, but it probably won't be easy. Just for reference, this is the code in prusaslicer that does this (in the patched avrdude): https://github.com/prusa3d/PrusaSlicer/blob/991f2bdbd18cbc18912147d7886bb1e1e3412742/src/avrdude/arduino.c#L84

> Which part of firmware actually handles "w25x20cl_enter" command? Can't find it in prusa-firmware and also in published
on github mk3-32u2-firmware. Some closed bootloader variant in prusas?

For the Prusa-Firmware, it is https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/optiboot_xflash.cpp. The slicer part was linked above.
The 32u2 that is on github is outdated. There is another repo, but it is not public yet for some reason. The 32u2 firmware has nothing to do with language flashing so you can just ignore it. The 2560 bootloader IS special, but not for flashing (it's only responsible for the language select, and that part is public)

arekm, 2022-08-18:
> The problem is that when avrdude is called, the serial port is opened again and that triggers a reset of the printer, rendering your previous w25x20cl_enter attempts useless.

In my testing serial was opened only once (by avrdude) because I used

> Just for reference, this is the code in prusaslicer that does this (in the patched avrdude): https://github.com/prusa3d/PrusaSlicer/blob/991f2bdbd18cbc18912147d7886bb1e1e3412742/src/avrdude/arduino.c#L84

exactly this approach - patched **wiring code** for avrdude to do external flash init. To see if wiring would work. But it didn't (log pasted earlier).

Well, we will need to live with a patched avrdude ;(