prusa3d/Prusa-Firmware #3559: [FEATURE REQUEST] Print multiple G-Code objects in one print

jwillmer, 2022-08-10:
I have multiple G-Code files on my printer SD card. Most are small parts like a shopping coin, filament swatch etc. It would be cool if I could select multiple for one print process and the firmware merges the G-Code so that it prints them simultaneously. A first step could be to print the same file multiple times by adding a multiplier factor.

wavexx, 2022-08-20:
It's not something that the firmware should do IMHO (at least not without a lot of assistance from the slicer first).

Even assuming you have all same-layer-height/same-filament/same-profile files, one of the dumbest problems from the FW point of view is that the g-code file doesn't contain boundary information of each object in order to lay them out. The purge line, for example, is something that would need to be stripped to avoid overlap, but I can currently only think of hacks in order to do so.

Prusa-Support, 2022-09-07:
I apologize. After hearing from the internal developers I must say that what I stated above is not very accurate.

They are working and testing secondary applications for the "bounding box".

Printing multiple gcodes at once, or simply would shifting a whole print, would actually require active slicing capabilities ("lot of assistance from a slicing engine") on the printer' side, or via an external board/server.
This is extremely unlikely to happen with the current hardware and state-of-the-art technology.

This issue can remain open for now though, as inspiration for future developments.

Michele Moramarco
Prusa Research

closed 2023-11-08 by 3d-gussner