prusa3d/Prusa-Firmware #3561: [FEATURE REQUEST] Do not enter SD card menu if it is empty

The3Designer, 2022-08-17:
**Printer type** - MK3S (this is the printer I own, but it might also be applicable to other models)
**MMU Upgrade** - n.a.

**Is your feature request related to a problem? Please describe.**
When an SD card is inserted into the 3D printer when it is powered on, the SD card menu is automatically opened. This occurs regardless of the contents of the SD card. Consequently, the SD card menu is also automatically opened/entered when there are no files on the SD card, which requires the user to manually go back to the previous menu.
  
**Describe the solution you'd like**
I'd like the 3D printer (firmware) to check if there are any folders or files on the SD card, and if there are none (i.e., the SD card is empty), then the SD card menu should not automatically be opened/entered upon inserting the SD card into the 3D printer.

wavexx, 2022-08-20:
I like the idea, but it could equally be confusing if you plug the card without knowing that's empty and keep wondering why you get no SD menu?

The3Designer, 2022-08-20:
@wavexx I don't know how likely that scenario is. If you want to print something, you've put a file on the SD card and you know that the SD card will not be empty when inserting it into the 3D printer. Likewise, if you've printed something and cleared the SD card on the PC (which is what I do after pretty much every print), then you know that the SD card will be empty when inserting it into the 3D printer again.

If this change/feature request is relatively simple to implement and doesn't require too much additional bytes in the code, then I'd love for it to be implemented. It definitely isn't a critical bug or anything, so it shouldn't get priority status ;)

wavexx, 2022-08-20:
Ok but roundtrip question: if you just cleared the SD card why would you insert into the printer again knowing there's nothing to print?

The3Designer, 2022-08-20:
Because I like to keep the SD card in the printer, even if it is empty. This way, I always know where it is, and I won't use it for a different device (i.e., I like to keep the SD card that came with my Prusa reserved for the printer).

Maybe this is an edge case, and I'm the only one (or one of few) doing this, but that's the origin/reason for this feature request :)

arekm, 2022-08-20:
Special cases are bad and confusing. I would be really surprised to see no reaction on inserted card with proposed behaviour. Is the card faulty or empty?

If changing behaviour then stop listing content at all on insertion and only say on LCD that "Sd card was inserted" (for short period of time) or some icon indicating sd card existence.

wavexx, 2022-08-20:
Hah, I get it now it's just for storage (makes sense). I'm still concerned though that when you inserted an empty sd card and got no reaction, it would be quite confusing.

The3Designer, 2022-08-20:
I don't understand why it could be confusing when an empty SD card is inserted to the 3D printer and there is no response given. You know as a user when the SD card is empty, right? The opposite is definitely true: if you've put a file on the SD card to print, you'll know it isn't empty when inserting the SD card into the 3D printer.

But like I said before, this isn't a crucial problem, so if others think this will result in confusing behaviour, then I can close the issue.

wavexx, 2022-08-20:
Just speaking from personal experience, I normally have a stash of SD cards next to the printer. You just plug one and nothing happens: my thinking starts with "is the card dead" or "is the reader faulty?".

The3Designer, 2022-08-20:
Haha okay, but why would you "just plug one in"? Maybe I'm too organized and/or my setup is simple with one 3D printer with a single SD card that I reserve for it, but for such a scenario, this feature request makes sense.

Regardless, from this discussion it seems like there is no demand for the proposed feature request.

The3Designer, 2022-08-21:
Perhaps a solution to the possible confusion we discussed here: an SD card indicator icon on the LCD display (bottom right):

[image]

When a user inserts an empty SD card into the powered on 3D printer, this icon will be displayed in the bottom right. If the SD card is not empty, the SD card menu will automatically be opened/entered. If the user then goes back to this main menu, the indicator icon will still be displayed.

@wavexx and @arekm what do you think about this? Could this help remove the possible confusion on inserting an empty SD card?

leptun, 2022-08-21:
The last line can't contain any extra character. The status line is already too small and all texts are made so they fit those 20 characters.

The3Designer, 2022-08-21:
Would either of these work?

[image]
Move the right column of data one "pixel block" to the left, which makes room for the SD card indicator icon on the top right.

[image]
Place the SD card icon (indicator) in front of the print percentage text (stock "pixel block" alignment).

leptun, 2022-08-21:
The first one can't work because of the MMU stuff (somewhere in the middle of the screen, second or third line (don't remember exactly).
The second one works better, but what about when SD printing or USB printing where the symbol gets replaced by text?

The3Designer, 2022-08-21:
> The second one works better, but what about when SD printing or USB printing where the symbol gets replaced by text?

Hmmm yeah, that wouldn't be desired behaviour either... At this moment, I can't come up with a possible solution.
Given the lack of demand for the originally proposed behaviour, and the challenges faced with an SD card indicator icon, I'll close this issue.

closed 2022-08-21 by The3Designer