prusa3d/Prusa-Firmware #3563: Mesh bed levelling not working when Hotend Fan is running on MK3+

mukenukem, 2022-08-20:
As soon as the Hotend fan is running I get permanent "Some problem encountered, Z-Leveling enforced". This happens when 3 or 5 calibration attempts are selected in the calibration menu. I am quite sure this is caused by the Hotend fan running on PWM since the fan speed is higher during startup. When the Hotend temperature is below 50° and the fan stops Mesh bed Leveling runs quickly and without stalls. As soon as the Hotend fan is running the MBL stalls on some points and forces Z-Levelling.

I am quite sure this is caused by either some noise on the power supply of the Super Pinda caused by the PWM of the fan (since the fan runs on 5V and the Superpinda probably too) for some crosstalk from the Einsy Rambo to the Hotend.

Solution: Either stop or run the Hotend fan at full speed (cannot verify if it works) during probing of the calibration points.

wavexx, 2022-08-20:
@leptun have you ever noticed this with the delta?

leptun, 2022-08-20:
I've not noticed this on any MK3 printer yet. The only things I noticed are as follows:
- Some of the newer einsy boards have that noise on the 5V line (something something different buck converter IC), and that definitely doesn't help with the reliability of the pinda. The fan is just adding to the issue.
- The super pinda seems to trigger slower rechecks more often for me, but that might just be because of the higher hysteresis of the sensor. On my Klipper printer it didn't cause me any issues.
- The MK2.5 printers generate A LOT of rechecks. It's borderline stupid. I still don't know what causes these since it happened both with a minirambo and an einsy (both with the bed at 12V). It might be related to EMI form the bed, but I'm not certain of this. It could also be because of the flimsier frame or different magnet positioning.
- There is also the delta PSU vs Meanwell PSU 24V rail noise, but I don't think this is significant in this case since I don't get rechecks on a MK3S with Pinda V2 and delta PSU.
- The Delta fan might be the reason for this since it draws power from the same 5V rail as the superPinda. I don't think it's noise on the wires that causes this, but rather just a noisy 5V supply caused by the delta fan PWM in combination with some other issue.

Now, I have a MK3S+ with a delta fan, superpinda and a delta PSU and I haven't noticed rechecks with this one, so I'm still not convinced it's because of these 3 parts. I still suspect this might be related to the 5V regulator issue, but I can't say for certain until more testing is done internally.

mukenukem, 2022-08-20:
I am using the delta PSU. I have a couple of MK3+ and 2.5+ in the office and can cross check next week. I know for sure that from a certain firmware version after the introduction of the 7x7 pattern the probing started to randomly stall/get slower. What I learned today on my latest mk3+ that when no fan is running the mesh bed leveling runs superfast, with the fan it stalls (it also stalls if the Hotend is not heated and cooling down). I will probably swap the Delta fan with a Noctua and hard-wire it to 5V (or use a 12V fan) till this issue is resolved.
Is there a reason why the fan is driven with PWM ? Certain desired RPM ? Is the Noctua driven with full power ?

leptun, 2022-08-20:
Delta is driven with pwm, noctua is full power. You can disable the pwm in the experimental settings menu. Go to HW setup and long press. That should make the menu appear at the end. Go there and toggle altfan check or something like that.

Just an idea, this might also be because of vibrations. Any chance this might be the case?

wavexx, 2022-08-20:
I have some very nasty part fans I'm too lazy to replace, but despite the vibrations I can still pass MBL without a single recheck. I would tend to think about PWM noise here.

mukenukem, 2022-08-21:
Today I verified that indeed the hotend fan PWM is causing the troubles. Generating probably too much noise on the 5V line. I will have a look at the voltage with an oscilloscope next week. If I disable the fan detection and the delta fan is running full speed the MBL finishes quickly without a single stall. Same with cold hotend and no fan running. With fan detection and PWM I cannot finish MBL. It often stalls and enforces Z-levelling. As workaround for me I will disable fan detection and probably put a Noctua in. Next week I will cross check the other MK3 and MK2.5 in the office. I think I noticed some stalling on the MK2.5. All of them have the SuperPinda and I think also a Noctua fan.
For future firmware releases I would recommend disabling the PWM at least during MBL. Then you have the accuracy when you need it and for crash detection running on PWM should be no problem.

wavexx, 2022-08-21:
@mukenukem can you share your S/N of the printer and Einsy version? I'm curious if this is specific of some batches.

mukenukem, 2022-08-21:
Well, the printer is built from genuine pare parts I had laying around (thought I store some spare parts but nothing needed to be replaced) and ordered from Prusa, just the Einsy Rambo was bought in the wild because it was at that moment not available. Silkscreen reads EINSY RAMBo 1.1b and looks exactly like the previous boards I got from Prusa with the kits. Also the soldering is flawless. The board included some heatsinks for the stepper drivers. So I should have bought a genuine board... damn. I will crosscheck the issue with our 2 MK3+ in the office, those are genuine. Just have to connect a different fan there. This will probably take a couple of days because I am on the street the next week. I will update asap.

mukenukem, 2022-08-24:
I crosschecked this issue today with a genuine Prusa supplied Einsy Rambo and could not duplicate the issue. So probably my Einsy is crap. I already ordered one from Prusa (no need for issues with the mainboard, better have this vital part original) and will check with this one.
Probably still a good idea to disable PWM when bed levelling for enhanced accuracy.

wavexx, 2022-08-24:
It would still be very interesting if you could look at that with the scope if you can (and if you can read anything on the regulator of your 1.1b that would be interesting too). I don't have a delta fan at hand, but I'll try to get my hands on one and try to run the noctua with pwm as well.

mukenukem, 2022-08-24:
Yes will do as soon as my new board arrives. I have 3 MK3+ available so I have a decent sample size ;)

thew1se, 2022-10-03:
Just commenting that I have recently received an new Einsy board for my MK3+ for a separate problem, but am now seeing this same issue since replacing the board (I have just noticed this after about 5 prints and never had this problem prior).  The repeat calibration attempts appear random.  If i do 3x3 it sometimes gets through to the print.  If I do 7x7, no chance.   If I can help by providing some information, let me know very specifically what to give you.

thew1se, 2022-10-03:
@leptun yes I have a Delta hotend fan.   I did as you requested and everything functioned as it should with no repeat calibrations during mesh bed levelling.... Even with 7x7 with 5 checks.  Thank you!     Let me know if you need Einsy board info to help determine which versions are affected, (and where to find this information).

leptun, 2022-10-03:
If you could say which einsy revision you have (written on the pcb), that would be great. I’ll try and see if I can somehow make the printer disable the fan pwm every time the pinda is being used.

thew1se, 2022-10-03:
Where on the PCB is it written? can you send a screenshot or let me know between which connectors?     Also, I was writing with chat support on Prusa3d and they said that Altfan detection might need to be set to off by default.

leptun, 2022-10-03:
It is to the left of the USB connector: 
[image]

That doesn’t sound right. Altfan detection should be enabled by default. The altfan override should only be used if the user has a custom fan which doesn’t need PWM regulation.

thew1se, 2022-10-03:
Ok i will let you know in a few hours (print in progress).

thew1se, 2022-10-03:
Looks like 1.21, but the second decimal is not crisp. 

[image]

leptun, 2022-10-03:
Einsy Rambo 1.2f

thew1se, 2022-10-03:
Need anything else?

leptun, 2022-10-03:
Nothing right now. Thanks for the info provided. I’ll see what I can find out about this board revision and what changed.

thew1se, 2022-10-03:
Is a side effect of turning off AltFan that it will always run at 100%?  seems to be running fast now.

leptun, 2022-10-03:
Yes. Turning that setting off will make it so it no longer checks for the fan type on startup and it defaults to 100% speed instead of ~50% that is when using a delta fan with the altfan detection enabled.

thew1se, 2022-10-12:
Hi was this issue addressed in Firmware 3.12.0-BETA1?

thew1se, 2022-10-12:
It happens mostly when it goes to the 2nd layer.  The preset settings in prusa slicer for PC are 110C for first layer and 115C afterwards.  Sometimes it gets to 115C and sometimes not.  I have fans in my case which usually keeps the temp at about 31C, but I turn these off for this portion of the printing because of this issue.   Even with them off, I sometimes get thermal runaway, so assumed it was the hotend fan causing too much cooling in the area.

thew1se, 2022-10-12:
Bed

wavexx, 2022-10-12:
Can you open a new issue for this and attach a video of it? I doubt the airflow of the hotend fan can influence the bed either.

thew1se, 2022-10-12:
Next time I print PC Blend I will.  Printing PLA in progress atm.

Prusa-Support, 2022-10-18:
Hello. Please notice that 100% fan speed can actually be a cause of _thermal runaway_ errors sometimes.
In some cases, the printer could struggle to recover from a sudden temperature drop if the print fan is not enabled gradually.
This could be a matter of print settings, as explained in https://github.com/prusa3d/PrusaSlicer/issues/6819#issuecomment-901700225, and concern [prusa3d/PrusaSlicer](https://github.com/prusa3d/PrusaSlicer) repository.
In rare cases, deformed or mishandled extruder assembly parts could cause similar issues too.

Moreover, high-temperature materials in general tend to naturally cool and set down fast enough even if not aided by the print fan. For this reason, these usually don't require a powerful print fan which may indeed cause warping and poor layers blending.

Please @thew1se consider reaching our [customer support](https://help.prusa3d.com/article/customer-support_2287) before opening a new issue related to the _thermal runaway_ errors.

Michele Moramarco
Prusa Research

goyetus, 2023-04-22:
I have three Mk3s+ printers all with delta printers and firmware 3.11.0.
 
Yesterday my delta fan died.   I tryed 5 original prusa delta fans in the printer.   ALL FAILED with the SUPERPINDA. 
When delta fan is running,  the probe fails.

You have the exact problem in this github from  prusa:
https://github.com/prusa3d/Prusa-Firmware/issues/3563

-----------------------------
VIDEO 0  -  Mk3s1 and Mk3s2  working perfect
-----------------------------
My mk3s+ 1  and Mk3s+2  working perfect with Delta fan
Firmware 3.11.0
https://drive.google.com/file/d/125S7-9ukBKt-T2uvpneaLfKm3fTNpL6B/view?usp=sharing

Videos about my faulty MK3s+
It seems to be a bad conversion in the 5v rail  as seen in the github.
-----------------------------
VIDEO 1
-----------------------------
- Hotend: Ambient
- Bed:  Ambient
- Fan no spinning  (below 50º)
NO FAILING
https://drive.google.com/file/d/120pyaO5EuHfUA0aDxXVQ2JPuVQc8doQc/view?usp=sharing

-----------------------------
VIDEO 2
-----------------------------
- Hotend 100
- Bed 60
- No fan attached
NO FAILING

https://drive.google.com/file/d/11tbd-1Bb5Sf-rRyuREfuRSuzjGzIEFgH/view?usp=sharing

-----------------------------
VIDEO 3
-----------------------------
- Hotend below 60
- FAN SPINNING
- Bed 60

"""FAILING""""
https://drive.google.com/file/d/1219NJC1mqG7LU-k2UWocwxzjZOJ6iCy7/view?usp=sharing

mukenukem, 2023-04-22:
Maybe the Delta fan draws more current than the Noctua (and what the step down can cope with) which causes excessive ripple on the 5V rail..... some fans draw current in nasty pulses.... maybe it would be an option to disable the fan during the probing. I think the hotend will stay cool enough if you have the fan running at full speed when travelling between the probing points ?

goyetus, 2023-04-22:
> Maybe the Delta fan draws more current than the Noctua (and what the step down can cope with) which causes excessive ripple on the 5V rail..... some fans draw current in nasty pulses.... maybe it [...]

I would love to STOP the fan while probing,  but is not possible.   
When heater reach 51º  ,  fan start spinning. 

I tested also 5 delta fans in this printer.   Yesterday was working with a DELTA FAN without any problem,  today,  no delta fan works anymore with the probe.....

It seems something bad with the motherboard..... the printer was perfect about 8 months!!!

mukenukem, 2023-04-22:
Maybe some additional filtering on the 5V? I swapped the Deltas with Noctua and this seems to have at least my problem solved.

goyetus, 2023-04-22:
> Maybe some additional filtering on the 5V? I swapped the Deltas with Noctua and this seems to have at least my problem solved.

I dont know.... not all the Board are with this issue.....  
Meanwhile.....Noctua at full speed = 4500 rpm  
At normal speed is 4300 in Prusa.

Delta at 50%  is about the same,  4500 rpm....  but can go 100%  to 8000 rpm  and double CFM noctua does.....

hope this get fixed.....  I purchased all "delta" last year....

mukenukem, 2023-04-22:
Afaik does Noctua implement a smoother commutation which should make it quieter. And probably less noisy on the supply voltage. So my guess is more ripple with the Delta fan which affects the SuperPinda...

leptun, 2023-05-03:
For me the Delta fan didn't seem to cause any issues. I started getting the slow reprobing only when the SuperPinda wasn't perfectly perpendicular to the bed. Can you check if maybe this is the case also for you? After I lowered one of the Z motors by a few mm, every point would be probed slowly.

thew1se, 2023-05-03:
Do you have the same einsy board mentioned above?

leptun, 2023-05-03:
I'm using Delta PSU + Einsy 1.2f + superpinda + delta fan + R7 extruder plastics

thew1se, 2023-09-24:
This issue is still present.   Is there any chance this will get fixed?  If not, I would like a new einsy board.

Prusa-Support, 2023-10-31:
Unfortunately, as explained in https://github.com/prusa3d/Prusa-Firmware/issues/3933#issuecomment-1599661735, we can't guarantee that a new board will solve the problem because there may be multiple causes of the problem, but we are working on it.

In the meantime, please make sure Altfan is ON or perform a factory reset to recalibrate the PWM, or consider using a Noctua fan as it seemed to avoid the problem for some users active on the other thread.

Michele Moramarco
Prusa Research

goyetus, 2024-02-07:
Ok,  you close The previuous issue as Unresolved, affecting all the Fans I purchased from Prusa.
Is going prusa to refund me all the spare parts  I purchased in Fans?   About 6 of them.   

I think I have been patient enough.

Prusa-Support, 2024-02-13:
The issue https://github.com/prusa3d/Prusa-Firmware/issues/1281 should be closed as obsolete and it is most likely unrelated, but if you are talking about https://github.com/prusa3d/Prusa-Firmware/issues/3933, that one was closed accidentally, and reopened.

About refunds, these inquiries must be discussed directly with Customer Supports in the dedicated official support channel.

Michele Moramarco
Prusa Research

Prusa-Support, 2024-07-02:
New firmware [FW 3.14.0](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.14.0) is out, and it improves the MBL in several ways.
This specific issue is hardly reproducible, however, our developers are positive that the new MBL process can address several weak points of the previous behavior, which are potentially closely related to this issue.

Please give it a try and share feedback.

Michele Moramarco
Prusa Research

benncathloy, 2024-08-12:
The issue described in bug report #3563 pertains to the MK3+ 3D printer, where mesh bed leveling fails to function correctly when the hotend fan is operating. This problem can lead to inaccuracies in bed leveling, affecting print quality and adhesion. The fan's operation seems to interfere with the sensor's ability to detect the bed surface accurately, causing inconsistent measurements. Users experiencing [Scope Leveling Kits](https://medium.com/@fadimahar02/best-scope-leveling-kits-950ec93ac2b6) this issue may need to troubleshoot by checking fan wiring, adjusting fan speed settings, or seeking firmware updates that address this specific bug. Addressing this problem is crucial for maintaining the precision and reliability of 3D prints.

thew1se, 2024-12-10:
Still an issue which has not been addressed.

closed 2025-02-17 by github-actions[bot]