prusa3d/Prusa-Firmware #3567: [BUG] Serial interface not responding on J19 // External SPI

SaguaroMaster, 2022-08-21:
MK3S
Ffirmware -  3.11
USB/Octoprint

I got the "External SPI Flash unavailable" warning, and I can't print from SD card and can't switch languages. Now my issue is that along with the SPI bus to the LCD, the serial port, which can be used to connect to OctoPrint/PrusaLink stopped working too for no reason it seems. When I try to connect a serial monitor to it, it only echoes back anything that I send it and ONLY at 9600 instead of 115200 baud.
Here is what I have tried so far:
-Different serial to usb adapters
-Different cables
-Made sure GND is connected
-Turned the RPi Port on/off.
-Tested J19 RX/TX pins for shorts
-Flashed older firmware
-Refalshed current firmware

When I first got the SPI flash unavailable warning I was trying out different Y axis motors, so I must have accidentally done something to the hardware of the board, altrough I cant seem to find anything broken on it.
All other functions work properly, and printing trough USB works too.

leptun, 2022-08-21:
Are you able to switch TMC modes? (as in switch from normal mode and stealth mode). If it does not switch, then you've probably busted the 3.3V regulator or the level shifter IC for the SD card and all devices connected on the SPI bus can no longer talk to the atmega2560.

SaguaroMaster, 2022-08-21:
I have just checked, mode changing works as expected.

leptun, 2022-08-21:
Can you check if U11 outputs 3.3V?

SaguaroMaster, 2022-08-21:
I have a print running, will check it and reply within 20 minutes

SaguaroMaster, 2022-08-21:
I think you have found the problem, U11 is probably toast as it has a tiny burn mark in its corner and on the output i measured around 0.6 volts. Is there another 3.3V source on the board from where I could bridge that voltage there? Or any other possible solutions?

leptun, 2022-08-21:
U11 or U12 are most likely damaged, U9 and U14 _could_ be damaged. When I had this happen, the LDO output was shorted to GND by the LDO.
In my case, I broke the LDO by dropping a screw on the MMU electronics while powered, sending a 24V spike on the 5V rail. The LDO that is used is only rated for something like 6V max, so it died first (and somehow it was the only damaged component).
This is what the fix ended up looking like. I replaced the U11 with an 3.3V LDO stolen from an arduino UNO. 
[image]

I really don't recommend this fix however. It was painful to solder such tiny components.

I confirmed it was U11 that was damaged by injecting 3.3V from an arduino into the 3.3V rail and seeing which component heated up: https://photos.app.goo.gl/8QwVe19eesr8yEGQ8

Can you post a picture with the pins you connected on J19? There is a 24V pin on that header and you might have accidentally used it.

SaguaroMaster, 2022-08-21:
I have just broke off one of the 3 caps near U11 and injected 3.3V using one of the pads with a lab bench PSU. When i supplied that voltage and turned on the printer the warning went away. The board dre 0.17W / around 50mA, i think that is acceptable. I have a 5V to 3.3V small buck converter lying around, im thinking a will use that as a replacement, do you think it won't cause more problems?

SaguaroMaster, 2022-08-21:
Okay, so i have wired up a buck converter, which gets 5V from a fan header. As there is a common GND, only 1 wire is enough to supply 3.3V. I dont know how long will the converter or my soldering last, but so far it works ok.

[image]

Now the warning is gone, language switch and printing from SD works again. The serial header still does not seem to work, but hey at least the other thing got fixed.

I'm still open for ideas regarding the serial interface, as that would be the most important part for me

Thank you @leptun !

leptun, 2022-08-21:
Considering the only way to kill that LDO is by injecting high voltage into the 5V rail, I'd be skeptical of anything that you plugged into the J19 header, including the UART1 pins of the atmega2560 (the ones on the J19 header).

SaguaroMaster, 2022-08-21:
I have only plugged in GND, TX and RX from the RPi. Once or twice I accidentally plugged GND into the RX pin, but that should not have caused this.

Prusa-Support, 2022-09-07:
Thanks for reporting.

This seems to be the case of a hardware fault, which would hardly find a solution on the firmware side.
@SaguaroMaster could this issue be closed?

Michele Moramarco
Prusa Research

closed 2022-09-08 by SaguaroMaster