prusa3d/Prusa-Firmware #3568: [BUG] MKS3+ stuck at sd card read (probably - see description)

arekm, 2022-08-21:
**Printer type** - MK3S+
**Printer firmware version** -  own build of MK3 branch as of 3ebd8b88a4d1f6879979d7afddd67d430416e6d8

**SD card or USB/Octoprint**
Printing from SD card.

**Describe the bug**

Printer was printing for few hours and then it stopped. Pause & resume didn't help (procedure finished
but printer was back in a stuck state).

Note - I'm using sdwire hardware for sdcard which could be causing trouble, so I'm not sure if this is really a printer firmware bug. sdwire is basically a switch which can disconnect sdcard lines from the printer (but it shouldn't do that unless told to do so which wasn't the case here).

So far happened two times with different micro sd cards and different sd card adapters. And I'm using sdwire for a weeks already. The second time it was stuck just at the end of intro line (note: mine intro line is a bit longer than default).

Questions:
- printer was not reporting any sd reading errors, nothing on LCD, nothing on serial. But does it report sd card problems if these occur or just silently tries to deal with these?

- is it possible to pause a print, remove sd card and insert it again (or maybe even new card with the same file) and resume a print ? (didn't try that but I'm wondering for next time when this problem happens)

Dump and firmware hex attached.

```
2022-08-20 19:32:14,649 - Send: M27
2022-08-20 19:32:16,138 - Recv: fixed-length-upper-bridge_0.6n_0.3mm_PETG_MK3S_7h14m
2022-08-20 19:32:16,144 - Recv: SD printing byte 3970560/8303993
2022-08-20 19:32:16,145 - Recv: 03:42
2022-08-20 19:31:27,102 - Recv: T:251.5 /250.0 B:89.7 /90.0 T0:251.5 /250.0 @:38 B@:105 P:0.0 A:44.4
2022-08-20 19:31:27,341 - Send: M27
2022-08-20 19:31:27,660 - Recv: fixed-length-upper-bridge_0.6n_0.3mm_PETG_MK3S_7h14m
2022-08-20 19:31:27,664 - Recv: SD printing byte 3964777/8303993
2022-08-20 19:31:27,666 - Recv: 03:41
2022-08-20 19:31:27,668 - Recv: ok
2022-08-20 19:31:28,344 - Send: M27
2022-08-20 19:31:28,512 - Recv: fixed-length-upper-bridge_0.6n_0.3mm_PETG_MK3S_7h14m
2022-08-20 19:31:28,514 - Recv: SD printing byte 3965526/8303993
2022-08-20 19:31:28,517 - Recv: 03:41
2022-08-20 19:31:28,518 - Recv: ok
2022-08-20 19:31:29,109 - Recv: T:250.2 /250.0 B:89.7 /90.0 T0:250.2 /250.0 @:52 B@:90 P:0.0 A:44.3
2022-08-20 19:31:29,347 - Send: M27
2022-08-20 19:31:29,826 - Recv: fixed-length-upper-bridge_0.6n_0.3mm_PETG_MK3S_7h14m
2022-08-20 19:31:29,830 - Recv: SD printing byte 3966495/8303993
2022-08-20 19:31:29,832 - Recv: 03:41
2022-08-20 19:31:29,833 - Recv: ok
2022-08-20 19:31:30,223 - Recv: NORMAL MODE: Percent done: 50; print time remaining in mins: 216; Change in mins: -1
2022-08-20 19:31:30,229 - Send: M114
2022-08-20 19:31:30,231 - Recv: SILENT MODE: Percent done: 50; print time remaining in mins: 222; Change in mins: -1
2022-08-20 19:31:30,535 - Recv: X:136.50 Y:154.47 Z:19.10 E:278.18 Count X: 156.82 Y:155.02 Z:19.41 E:276.66
2022-08-20 19:31:30,544 - Recv: ok
2022-08-20 19:31:30,546 - Send: M27
2022-08-20 19:31:30,683 - Recv: fixed-length-upper-bridge_0.6n_0.3mm_PETG_MK3S_7h14m
2022-08-20 19:31:30,684 - Recv: SD printing byte 3967195/8303993
2022-08-20 19:31:30,687 - Recv: 03:41
2022-08-20 19:31:30,688 - Recv: ok
2022-08-20 19:31:31,112 - Recv: T:253.2 /250.0 B:89.7 /90.0 T0:253.2 /250.0 @:16 B@:76 P:0.0 A:44.6
2022-08-20 19:31:31,353 - Send: M27
2022-08-20 19:31:31,678 - Recv: fixed-length-upper-bridge_0.6n_0.3mm_PETG_MK3S_7h14m
2022-08-20 19:31:31,682 - Recv: SD printing byte 3967834/8303993
2022-08-20 19:31:31,683 - Recv: 03:41
2022-08-20 19:31:31,684 - Recv: ok
2022-08-20 19:31:32,357 - Send: M27
2022-08-20 19:31:32,505 - Recv: fixed-length-upper-bridge_0.6n_0.3mm_PETG_MK3S_7h14m
2022-08-20 19:31:32,506 - Recv: SD printing byte 3968471/8303993
2022-08-20 19:31:32,509 - Recv: 03:41
2022-08-20 19:31:32,510 - Recv: ok
2022-08-20 19:31:33,119 - Recv: T:247.4 /250.0 B:90.4 /90.0 T0:247.4 /250.0 @:80 B@:0 P:0.0 A:44.5
2022-08-20 19:31:33,360 - Send: M27
2022-08-20 19:31:33,599 - Recv: fixed-length-upper-bridge_0.6n_0.3mm_PETG_MK3S_7h14m
2022-08-20 19:31:33,603 - Recv: SD printing byte 3969386/8303993
2022-08-20 19:31:33,604 - Recv: 03:41
[... 102 lines cut ...]
2022-08-20 19:32:28,278 - Recv: 03:42
2022-08-20 19:32:28,279 - Recv: ok
2022-08-20 19:32:28,283 - Send: M27
2022-08-20 19:32:29,774 - Recv: T:253.4 /250.0 B:90.8 /90.0 T0:253.4 /250.0 @:13 B@:0 P:0.0 A:44.7
2022-08-20 19:32:29,785 - Recv: fixed-length-upper-bridge_0.6n_0.3mm_PETG_MK3S_7h14m
2022-08-20 19:32:29,787 - Recv: SD printing byte 3970560/8303993
2022-08-20 19:32:29,788 - Recv: 03:42
2022-08-20 19:32:29,789 - Recv: ok
2022-08-20 19:32:29,791 - Send: M27
2022-08-20 19:32:31,290 - Recv: fixed-length-upper-bridge_0.6n_0.3mm_PETG_MK3S_7h14m
2022-08-20 19:32:31,295 - Recv: SD printing byte 3970560/8303993
2022-08-20 19:32:31,296 - Recv: 03:42
2022-08-20 19:32:31,297 - Recv: ok
```

I was using own firmware build from MK3 branch with latest 3ebd8b88a4d1f6879979d7afddd67d430416e6d8 commit.
[Firmware-intl.hex.txt](https://github.com/prusa3d/Prusa-Firmware/files/9389609/Firmware-intl.hex.txt)

Dump, as seen on video I've tried to do dump. That's what was stored:
[dump.txt](https://github.com/prusa3d/Prusa-Firmware/files/9389620/dump.txt)

**Video**
https://www.youtube.com/watch?v=qWh7y4qZMLA&feature=youtu.be
Tried to pause and resume at beginning.
Notice how slowly it reacts to knob.

arekm, 2022-08-25:
Maybe the problem is with sd card slot?

https://www.printables.com/model/15388-prusa-mk3-sd-card-slot-fix-not-reading-the-card-at.

Answer for my first question in initial issue comment would shed some light here...

(note, problem didn't repeat so far, yet)

Prusa-Support, 2022-09-26:
Hello! Thank you for your message, sorry for the late response. Normally, if card was removed (or not reading correctly, "as if it was removed"), there should be a message indicating this (https://cdn.help.prusa3d.com/wp-content/uploads/cardremoved-600x320.jpg). I tested the pausing-removing card-reinserting, it seemed that resuming the print still worked regularly though I wouldn't be able to guarantee it fully for all cases. 

Will note this for our devs, in case of more info from our side, we'll expand on it on this thread. 

Alessandro Pantaleo
Prusa Research

closed 2023-11-16 by github-actions[bot]