prusa3d/Prusa-Firmware #3572: [BUG]<menu language loop firmware 3.11>

A11lessandro2, 2022-08-26:
**Printer type** - MK3S+ mmu2s
**Printer firmware version** -  3.11.0

**MMU upgrade**  -  MMU2S
**MMU upgrade firmware version**  - 1.0.6

**SD card or USB/Octoprint** -   pc usb 

  **Describe the bug**
During the installation of firmware 3.11.0 using windows 10 and prusaslicer 2.42, a firmware loading error comes out,
the printer starts up anyway and exits the language selection menu. Among all the various languages of choice you can choose only the English language, all the other languages if selected send the loop the menu of the choice of languages.

**To Reproduce**
  avrdude-slic3r -v -p atmega2560 -c wiring -P COM6 -b 115200 -D -U flash:w:0:C:\Users\Dell_1\Desktop\Firmware agosto 2022\prusa3d_fw_3_11_0_MK3S_1_0_6_MMU2S\prusa3d_fw_MK3S_MK3S+_3_11_0_4955.hex:i

avrdude-slic3r: Version 6.3-20160220-prusa3d, compiled on Dec 21 2021 at 17:14:48
                Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
                Copyright (c) 2007-2014 Joerg Wunsch

                Using Port                    : COM6
                Using Programmer              : wiring
                Overriding Baud Rate          : 115200
                AVR Part                      : ATmega2560
                Chip Erase delay              : 9000 us
                PAGEL                         : PD7
                BS2                           : PA0
                RESET disposition             : dedicated
                RETRY pulse                   : SCK
                serial program mode           : yes
                parallel program mode         : yes
                Timeout                       : 200
                StabDelay                     : 100
                CmdexeDelay                   : 25
                SyncLoops                     : 32
                ByteDelay                     : 0
                PollIndex                     : 3
                PollValue                     : 0x53
                Memory Detail                 :

                                         Block Poll               Page                       Polled
                  Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                  ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                  eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
                  flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
                  lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
                  signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

                Programmer Type : Wiring
                Description     : Wiring
                Programmer Model: AVRISP
                Hardware Version: 15
                Firmware Version Master : 2.10
                Vtarget         : 0,0 V
                SCK period      : 0,1 us

avrdude-slic3r: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0,00s

avrdude-slic3r: Device signature = 0x1e9801 (probably m2560)
avrdude-slic3r: safemode: hfuse reads as D0
avrdude-slic3r: safemode: efuse reads as FD
avrdude-slic3r: reading input file "C:\Users\Dell_1\Desktop\Firmware agosto 2022\prusa3d_fw_3_11_0_MK3S_1_0_6_MMU2S\prusa3d_fw_MK3S_MK3S+_3_11_0_4955.hex"
avrdude-slic3r: writing flash (250500 bytes):

Writing | ################################################## | 100% 44,21s

avrdude-slic3r: 250500 bytes of flash written
avrdude-slic3r: verifying flash memory against C:\Users\Dell_1\Desktop\Firmware agosto 2022\prusa3d_fw_3_11_0_MK3S_1_0_6_MMU2S\prusa3d_fw_MK3S_MK3S+_3_11_0_4955.hex:
avrdude-slic3r: load data flash data from input file C:\Users\Dell_1\Desktop\Firmware agosto 2022\prusa3d_fw_3_11_0_MK3S_1_0_6_MMU2S\prusa3d_fw_MK3S_MK3S+_3_11_0_4955.hex:
avrdude-slic3r: input file C:\Users\Dell_1\Desktop\Firmware agosto 2022\prusa3d_fw_3_11_0_MK3S_1_0_6_MMU2S\prusa3d_fw_MK3S_MK3S+_3_11_0_4955.hex contains 250500 bytes
avrdude-slic3r: reading on-chip flash data:

Reading | ################################################## | 100% 32,08s

avrdude-slic3r: verifying ...
avrdude-slic3r: 250500 bytes of flash verified

avrdude-slic3r: safemode: hfuse reads as D0
avrdude-slic3r: safemode: efuse reads as FD
avrdude-slic3r: safemode: Fuses OK (E:FD, H:D0, L:FF)

avrdude-slic3r done.  Thank you.

avrdude-slic3r -v -p atmega2560 -c arduino -P COM6 -b 115200 -D -u -U flash:w:1:C:\Users\Dell_1\Desktop\Firmware agosto 2022\prusa3d_fw_3_11_0_MK3S_1_0_6_MMU2S\prusa3d_fw_MK3S_MK3S+_3_11_0_4955.hex:i

avrdude-slic3r: Version 6.3-20160220-prusa3d, compiled on Dec 21 2021 at 17:14:48
                Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
                Copyright (c) 2007-2014 Joerg Wunsch

                Using Port                    : COM6
                Using Programmer              : arduino
                Overriding Baud Rate          : 115200
avrdude-slic3r: prusa_init_external_flash(): MK3 printer did not boot up on time or serial communication failed
avrdude-slic3r: arduino_open(): Failed to initialize MK3 external flash programming mode
avrdude-slic3r: Could not open port: COM6

avrdude-slic3r done.  Thank you.

  
**Expected behavior**
 
Using windows 10 with this firmware 3.11.0 I always get the same error, a few months ago I updated the printer
   to the previous version 3.10.1 always using windows 10, and everything was fine, firmware installed correctly.
   However, I have already solved the problem by loading the 3.11.0 firmware with Linux Ubuntu. Using Ubuntu the firmware was installed correctly, no errors.

Prusa-Support, 2022-09-07:
Thank you for reporting and thank you for the log.
The log may not be enough to reproduce the error but could be useful for understanding.

Often, when the firmware installation fails because "COM port could not be opened", there is a concern about permissions (e.g. admin privileges denied) or about the communication (e.g. the port or the device are kept busy by other programs running in the background) limiting the functionality.
In some cases, the drivers may have not been installed (Prusa Slicer not installed via drivers pack) or may have not been installed correctly (e.g. antivirus programs restricting the installation).

In other words, there are chances that the cause of the issue is on the computer side. In fact, firmware and board worked fine when the firmware was installed from a different computer.

At the moment, we are not able to reproduce this issue.
Our developers may need more precise feedback and background information to be able to reproduce this issue.

Michele Moramarco
Prusa Research

Deloix, 2022-09-09:
I had this problem too, on Windows 10. I was updating after years from 3.4.1 to 3.11. Always disconnect error after verify of firmware. What helped was run slicer as Administrator, what I found after hour here, you should add this to update guide or somewhere on update window!!!

leptun, 2022-09-11:
@Deloix Are you using an account that has administrator privilages or is it a regular user account? It might be possible we've never noticed this since most of the developers use accounts with admin privilages on their machines.

Deloix, 2022-09-11:
@leptun I have default instalation of Win10, it means you are administrator, but standard programs runs in user mode for security reasons, if program runs in Admin mode itself it will ask(if you did not turn off UAC) and it have little shield on icon. Or you can run anything in admin mode by right click and select run as admin... So module which doing update should ask for admin privs before installation, this should be handled on developer side. Every program runs in user mode by default, when it not need acces folders outside ussers folders. It will ask when it need acces outside (you can disable asking in UAC), that is why you have right click menu "Run as administrator" when you trust program and you dont want be bother by asking or you need bypass some security behavior. And I think this will probably similar for COM/USB ports when doing something like disconnect and reconnect during firmavre installation, windows. Windows in user mode will evaluate it as suspicious and will not reconnect the device without user action. So maybe it would help to either run the installation directly with admin rights, or during the installation ask the user to reconnect and click again and at this point wait until this happens. But maybe I am wrong, I am not USB/COM port expert 😁

arekm, 2022-09-11:
That most likely wasn't "admin" issue since first avrdude-slic3r worked. Which means access to COM was fully granted. And "prusa_init_external_flash()" is called after COM opening.

Prusa-Support, 2022-09-22:
So likely it was not an "admin issue" but it would be worth a giving a try "running the program as administrator" anyway.

Sometimes, something may go wrong during the '_dpinstx_' drivers installation, which, you may want to try installing manually (as per [this article](https://help.prusa3d.com/article/firmware-updating-issues-mk2s-mk3s-mmu2s_2225)), possibly running the  specific drivers installation "as administrator".

This is not necessary the most of time, in fact the same program and printer could work flawlessly on a different computer.
In case you realize that "running the program as administrator" is always necessary on your computer, you can find an option in [right click] _properties > compatibility > run this program as an administrator_, which will supposedly remain ticked for all the subsequent launches of the program (maybe unless a new shortcut is created, or until the next program update).

Other than this, thank you all for sharing your experience and our developers will look into this.
Clear instructions for the reproducibility of the problem though, would be a big help.

Michele Moramarco
Prusa Research

Chsldz, 2022-12-30:
Hey guys,
so i had the same issue. On the first flash the avrdude-slic3r worked. At the second time i couldn't open the port and therefore i couldn't change the language.
Starting the slicer as admin worked! The flash was successfull and i finally could change the language!
So try running as admin and I should hopefully work.

closed 2023-11-16 by github-actions[bot]