prusa3d/Prusa-Firmware #3580: [BUG] SpoolJoin does not reset to Filament 1 on next print job

jtcmedia, 2022-08-30:
**Printer type** - MK3S
**Printer firmware version** -  3.11.0

**MMU upgrade** - MMU2S
**MMU upgrade firmware version** - 1.0.6

**USB/Octoprint**

**Describe the bug**
  SpoolJoin does not reset back to Filament 1 when another print job is executed.
  
**To Reproduce**
  Filament 1, 2, and 3 are loaded with filament. A print job runs, uses up Filament 1 and 2, and finishes with Filament in 3. On the next print job, you want to do the same, and re-load Filament 1 and 2. You choose Filament 1 using a plugin like Filament Manager (single mode) yet the printer will start at 3, regardless, unless you reset it.
  
**Expected behavior**
  SpoolJoin resets back to 1 on next print job OR give the User a choice to reset it.

Prusa-Support, 2022-09-21:
Thank you for reaching us.

Not sure about the functionality and compatibility of a plugin like 'Filament Manager'.

The SpoolJoin function can be used when printing in Single Mode.
Based on the normal use (via LCD panel) of the MK3S printer upgraded with MMU2S, you have two options to print in Single Mode.
a. Slice with "MMU2S Single" printer (preset in Prusa Slicer). The printer will automatically ask what filament to "load to nozzle" at the beginning of a print.
b. Print an "MK3S standalone" .gcode file, but you need to select the option "load to nozzle" from the LCD menu before the print. There you will have an option to load from a specific filament position.

In conditions of normal use as per the manual, it will be possible to select the filament position before the start of the print, otherwise, the preloaded filament will start the print (**not a bug**).

Please let me know if you have further doubts, or please consider closing this issue.

Michele Moramarco
Prusa Research

Prusa-Support, 2022-10-11:
I hope my previous comment helps to clarify.
With that, I would close this issue.

Michele Moramarco
Prusa Research

closed 2022-10-11 by Prusa-Support