prusa3d/Prusa-Firmware #3581: [BUG] 3.12 - During G28 with 7x7 mesh, Nozzle icon disappears while calibration points left are 10 or more

gudnimg, 2022-08-31:
**Printer type** - All
**Printer firmware version** -  3.12 (unreleased)

**MMU upgrade** - N/A - MMU can be left disconnected
**MMU upgrade firmware version** - N/A

**SD card or USB/Octoprint**
 N/A - SD card can be left disconnected

**Describe the bug**

While the calibration points left are double digits (larger than 9) the nozzle icon is overwritten with a space.

This for-loop seems to be responsible. It is called during mesh bed leveling and at various other scenarios such as at printer's boot-up. https://github.com/prusa3d/Prusa-Firmware/blob/66b97683bbb838661e5e8a2a3deda7cc7e8eb9d8/Firmware/ultralcd.cpp#L674

I cannot explain why this wasn't an issue before. Looking at the Git history, the issue first starts here in my fork: https://github.com/gudnimg/Prusa-Firmware/commit/b64377b9de17103ad970369fed8e967d872602d2 but this does not make sense since even if I take these changes out, the issue is still there. We likely need to look more deeply into this.

[image]

  
**To Reproduce**
Make sure the Mesh Bed Level settings are set at 7x7 mesh. (Problem does not appear with 3x3 mesh)

1. Open Serial stream
2. Send G28 command to the printer
  
**Expected behavior**
Nozzle icon should be present on the LCD at all times.
  
**G-code**
N/A

**Video**
N/A

gudnimg, 2022-09-10:
Interestingly, I'm not seeing the issue during First Layer Calibration 🤔

gudnimg, 2022-09-10:
I stopped seeing the issue on the MMU branch. Super weird, I haven't looked into this yet, but I think the underlying issue is probably still there we just don't see it visually for some reason.

leptun, 2022-09-11:
I still see this issue. MK3_MMU2 branch, MK3S build (en-only, but it doesn't matter).

Instructions to trigger:
- Reset printer. Make sure only the "Prusa i3 MK3S OK." message is visible.
- Set MBL to do 7x7
- Run Mesh bed leveling from the Calibration menu.

The thing is that the startup message is never "finished". By this I mean lcd_status_message gets initialized to WELCOME_MSG, but WELCOME_MSG is not 20 characters long, so the rest of the status line is filled with NULL bytes instead of spaces. lcd_finishstatus would need to be called for that. Later on, the following snippet catches these NULL bytes and prints spaces in their place (render function):
```
    // Fill the rest of line to have nice and clean output
    for (uint8_t fillspace = 0; fillspace < LCD_WIDTH; fillspace++)
        if ((lcd_status_message[fillspace] <= 31))
            lcd_print(' ');
```
The problem with that snippet is that it runs even if a special status is being displayed, such as the MBL status message. In that scenario, the MBL stuff is printed (entire line) and then the spaces missing in lcd_status_message are appended. One solution would be to "finish" the status line on startup, or better yet to use the corrent printf function instead of the snippet above to print a left justified 20 character string where the rest is padded with spaces: "%-20.20S" (for progmem) or "%-20.20s" (for ram)

gudnimg, 2022-09-17:
For all occurences of `"Prusa i3 MK3S OK."`, `lcd_finishstatus()` is called correctly via `lcd_updatestatus()`

PR #3600 (2022-09-21): Fix initialization of LCD status line message
Fixes #3581 

Change in memory:
Flash: -48 bytes
SRAM: -1 byte
commit f1e127ac (2022-09-21): Initialise status line message in setup()

Fixes #3581
commit 336c41ff (2022-09-21): Initialise status line message in setup()

Fixes #3581
closed 2022-09-21 by wavexx