prusa3d/Prusa-Firmware #3590: [BUG]<bed and nozzle don't heat up when print is started>

MeatTrident, 2022-09-11:
**Printer type** - MK3S+
**Printer firmware version** - 3.11.0-4955

**MMU upgrade** - MMU2S
**MMU upgrade firmware version** - 1.0.6

**SD card or USB/Octoprint**
  Please let us know if you print via SD card or USB/Octoprint
SD card

**Describe the bug**
  A clear and concise description of what the bug is.
When I try to start a print the target temperature is 0 and the print doesn't start  

**To Reproduce**
  Please describe steps to reproduce the behavior.
I installed the MMU2s and tried to start a print  

**Expected behavior**
  A clear and concise description of what you expected to happen.
I expected the bed and nozzle to heat up en the print to start  

**G-code**
  Please attach a G-code. This will make it easier for us to replicate the error.
It happend with any G-code

**Video**
  Please attach a video. It usually helps to solve the problem.

https://user-images.githubusercontent.com/113301122/189537227-ebb79aa6-28e1-4bc3-bcfd-47f08eb2859e.mp4

arekm, 2022-09-11:
Please attach any (tested!) gcode file just to be sure.

Prusa-Support, 2022-09-13:
Hello! In addition to sending us the GCode(s) you tested having this issue, can you also please check if downgrading temporarily to 3.10.1 and/or temporarily disconnecting the MMU2S (by disconnecting the power and signal cable, from the MMU2S side) would "resolve" the issue? That way we can further narrow this down and review in more detail. Thank you! 

Alessandro Pantaleo
Prusa Research

closed 2023-11-16 by github-actions[bot]