prusa3d/Prusa-Firmware #3595: [BUG]<Firmware crash at loading filament and print file Firmware 3.11>

fogarasia, 2022-09-15:
**Printer type - MK3S
**Printer firmware version - 3.11

SD card 

Describe the bug
  During filament automatic loading, I have chosen a file to print from the SD card, and pressing print the printer suddenly restarted.
  
To Reproduce
Unload the filament and auto-load the filament. Go to the main menu and then choose a file to print, press print, and wahtch.

arekm, 2022-09-15:
Are you capable of building and trying own firmware image? 

If yes then try firmware from MK3 branch as it has https://github.com/prusa3d/Prusa-Firmware/pull/2274 merged.

(on my MK3 build it doesn't crash with above reproduction method)

fogarasia, 2022-09-19:
Unfortunately not capable of this.

Prusa-Support, 2022-09-19:
Hello! Thank you for your message, sorry for the late response. So did this happen only with autoload switched on, and starting a print, but wouldn't occur with autoload switched off? And would it not occur with previous firmware, if we were to downgrade it temporarily, such as 3.10.1? Would self test also pass regularly if we were to do it now? >menu >calibration >self test. Feel free to send me a video of the reboot, it can be useful to better report it to devs in case. 

Let me know, and in case of anything to further review, of course feel free to ask. 

Alessandro Pantaleo
Prusa Research

moskyt, 2022-09-20:
Hello! I also wanted to report this issue but found it here already. It is very easy to reproduce (it happened twice to me today: as the LCD shows normal menu during filament change it does not remind you that you should not do anything and patiently wait for the filament autoload) and is not very helpful to suggest downgrading the firmware, shoot a video of reboot (it's just a reboot! nothing interesting going on) or running a self test. 

I definitely don't want to be uncivil or impolite, but suggesting to your paying customers that they should spend hours doing (probably meaningless) tests instead of fixing the firmware is not very ... customer-oriented. Probably the best solution would be to block the UX during the filament change, it is just very easy to forget that you need to wait if you are in a hurry.

fogarasia, 2022-09-20:
I have sent the Debug info via mail. And @moskyt is right, possibly blocking the UX is a solution.

Prusa-Support, 2022-09-25:
Hello, thank you for the responses. It will be fixed on this PR, though at this moment it's not merged yet: https://github.com/prusa3d/Prusa-Firmware/pull/3564. In essence, we won't allow the user to select a SD file if autoload is still taking place. 

Alessandro Pantaleo
Prusa Research

gplv2, 2022-11-09:
I can confirm this is reproducible .  It crashes the firmware every time.

arekm, 2022-11-09:
Try newer firmware, beta or alpha. https://github.com/prusa3d/Prusa-Firmware/releases

gplv2, 2022-12-03:
Unless I looked over it, none of the release notes on any those firmwares list this bug as fixed though.  Just upgrade and try (beta/alpha releases) isn't really a proper strategy I support.

closed 2023-07-26 by 3d-gussner