prusa3d/Prusa-Firmware #3604: [BUG] 3.11.0 regression: Chronic underextrusion, extrude commands fail to extrude

TheloTheGreat, 2022-09-18:
**Printer type** - MK3S+
**Printer firmware version** -  Glitchy on 3.11.0, healthy on 3.10.1

**SD card or USB/Octoprint**
  Printed via Octoprint

**Describe the bug**
  Since upgrading from firmware 3.10.1 to firmware 3.11.0, all prints fail with severe underextrusion after about two minutes. (It seems to work fine for the first couple of minutes.)

After canceling the print, manual filament extrude commands fail to do anything, both through the LCD commands "Move Axis > Extruder" and through the Octoprint UI.

Resetting the printer makes the extruder commands work again... for a while.

Reverting the firmware to 3.10.1 makes everything work fine again. (So it was not any mechanical issue or clog, it was purely the firmware refusing to extrude.)
  
**To Reproduce**
  - Update to firmware 3.11.0. To flash this firmware, I used PrusaSlicer, running on a Raspberry Pi that's connected via USB to the printer.
  - Preheat at extruding temperature, and try to operate the extruder via Move Axis > Extruder. (It should work.)
  - Start a print that takes more than two minutes to complete
  - Observe severe underextrusion after a few minutes
  - Cancel the print
  - Preheat the filament to extrude temperature (I used 215 C for the bundled Prusament PLA),  then try to manually extrude or retract the filament. Observe that the extruder gears don't rotate at all, and they don't even try to rotate - there's no clicking sounds or anything.
  - Reset the printer
  - Preheat at extruding temperature, and try to operate the extruder via Move Axis > Extruder. (It should work again.)
  
**Expected behavior**
  The extruder should attempt to extrude as expected at all times.
  
**Picture of under-extrusion**
  Here's an example where the print stopped really extruding after a few minutes, during the first layer: [First layer underextrusion](https://thelo.ca/underextrusion.jpg)

gudnimg, 2022-09-18:
Do you have logs from the printer when this happens? Are there any error messages in it?

TheloTheGreat, 2022-09-18:
I didn't keep logs, sorry. The LCD screen didn't show any visible errors, for what that's worth.

TheloTheGreat, 2022-09-19:
At the point where the extruder stops working, it stops rotating entirely. The bondtech gears don't even make an effort to rotate, whether there is filament loaded or not. No clicking sounds or audible motor whirr.

TheloTheGreat, 2022-09-19:
As I said before, this is not a mechanical issue, it's a firmware issue. It happens only on firmware 3.11.0, and never on firmware 3.10.1. The issue immediately started when I flashed 3.11.0, and immediately stopped when I reverted to 3.10.1. I have already triple-checked all of these mechanical components beforehand, including the bondtech gears, they are not involved.

TheloTheGreat, 2022-09-23:
Alright, I have some time to test. I've reflashed 3.11, and the issue happened again.

I noticed that when the printer is in the bad state of failing to obey extrude commands, the LCD screen for Settings ­- Move axis - Extruder says: "Extruder  +nan". When it's still in a good state (after startup), it says something like "Extruder: +139" instead. Maybe that's a clue?

I'm not sure if this is the log file you want, but the Octoprint "Terminal output" looks like this, when it's in the bad state and I send a failed extrude command:

Recv: T:210.2 /210.0 B:42.5 /0.0 T0:210.2 /210.0 @:28 B@:0 P:0.0 A:31.0
Recv: T:210.7 /210.0 B:42.5 /0.0 T0:210.7 /210.0 @:18 B@:0 P:0.0 A:31.9
Send: G91
Recv: ok
Send: M83
Recv: ok
Send: G1 E5 F300
Recv: ok
Send: M82
Recv: ok
Send: G90
Recv: ok
Recv: T:210.1 /210.0 B:42.6 /0.0 T0:210.1 /210.0 @:27 B@:0 P:0.0 A:31.4
Recv: T:210.7 /210.0 B:42.6 /0.0 T0:210.7 /210.0 @:15 B@:0 P:0.0 A:31.3

TheloTheGreat, 2022-09-23:
Trying again, this time the LCD screen shows "Extruder: +inf" during the bad state, rather than "+nan".

Pushing on the filament from above does not let it move as it would do with steppers disabled, the extruder's motors do keep it in place.

leptun, 2022-09-23:
The broken extruder coordinate is a very important clue. While in the bad state, try creating a memory dump: https://help.prusa3d.com/article/dump-memory-mk3-s-and-mk2-5-s_245237

Do NOT use pronterface for the dump data (instructions need updating). I think octoprint should be fine for dumping the crash log data from xflash.

TheloTheGreat, 2022-09-23:
New info: I just flashed a slightly updated version of the firmware: prusa3d_fw_MK3S_MK3S+_3_11_1-RC1_4987.hex , rather than prusa3d_fw_MK3S_MK3S+_3_11_0_4955.hex .

I then started my usual test print to reproduce the issue. Usually the issue happens within 2-3 minutes of printing, but... so far it's holding steady after 15 minutes of printing, surprisingly. Maybe there was a fix to this issue sometime between the 3.11.0 and 3.11.1RC1 releases?

I'll keep you updated.

leptun, 2022-09-23:
No. There was no fix for such an issue. Your problem is yet another unexplained bug that happens only on certain boards. Not all boards fail the same way and the issue almost always goes away with the tiniest change in the code. We've been unable to cause a bug like this yet.
Please try flashing the version that caused the issue and do the memory dump. We also need the exact version (and commit number if possible).

TheloTheGreat, 2022-09-23:
Here is a memory dump of a new occurrence of the glitch, as pasted from my Octoprint terminal, from the glitchy firmware:
prusa3d_fw_MK3S_MK3S+_3_11_0_4955.hex

This time, the extruder showed "-inf" as its value during the bad state.
I've left some of the log leading up to the memory dump, for context.
[serial-output-2022-09-23-extruder-fail.txt](https://github.com/prusa3d/Prusa-Firmware/files/9636788/serial-output-2022-09-23-extruder-fail.txt)

The newer firmware continued the test print fine for several hours before I cancelled it. So we have something like this:

Healthy: prusa3d_fw_MK3S_MK3S+_3_10_1_4697.hex
Glitchy: prusa3d_fw_MK3S_MK3S+_3_11_0_4955.hex
Healthy?: prusa3d_fw_MK3S_MK3S+_3_11_1-RC1_4987.hex

leptun, 2022-09-24:
Does this glitch only happen while USB printing? Is this issue specific to a certain gcode file? If it is possible, please upload the gcode file that you were using at the moment.

Here is some stuff that looks weird in the memory dump:
- destination[3] is -inf. Stuff around in memory looks fine.
- adv_comp is NaN. This one is suspicious
- gcode_target[3] - expected because of current_position[3]
- position[3] is a weird value
- position_float[3] is -inf
- current_position[3] is -inf

Please also try printing the same gcode file, but with linear advance disabled. To do that, remove any reference of M900 from the gcode file and reset the printer for good measure before printing from USB again.

TheloTheGreat, 2022-09-24:
Here is the last gcode file I was using, though the issue happened with some other files as well:
[Barre tension trouee v6_0.1mm_PLA_MK3S_4h27m.gcode.txt](https://github.com/prusa3d/Prusa-Firmware/files/9639221/Barre.tension.trouee.v6_0.1mm_PLA_MK3S_4h27m.gcode.txt)

TheloTheGreat, 2022-09-25:
I have just modified the gcode to remove all M900 uses, then printed that file. The glitch happened again (Extruder: +inf), here's the serial output and memory dump from that attempt:

[serial-output-2022-09-25-extruder-fail.txt](https://github.com/prusa3d/Prusa-Firmware/files/9641172/serial-output-2022-09-25-extruder-fail.txt)
[Barre tension trouee v6_0.1mm_PLA_MK3S_4h27m-no-linear-advance.gcode.txt](https://github.com/prusa3d/Prusa-Firmware/files/9641176/Barre.tension.trouee.v6_0.1mm_PLA_MK3S_4h27m-no-linear-advance.gcode.txt)

TheloTheGreat, 2022-09-25:
New info: Attempting to reprint the same gcode file from the SD card directly, with the USB cable physically disconnected, does *not* trigger the glitch so far.

wavexx, 2022-09-25:
@TheloTheGreat would you mind trying to print exactly the same file via serial using a beta version of the upcoming 3.12 version and report the behavior?

This is the build at commit 6bee4fec8dadf515c1ceccb8165704ae679de5fe for the MK3S+: [Firmware-MK3S-3.12-6bee4f.zip](https://github.com/prusa3d/Prusa-Firmware/files/9641305/Firmware-MK3S-3.12-6bee4f.zip)

TheloTheGreat, 2022-09-25:
Alright, I have just done yet another test print with each of the four firmwares mentioned above, including this new one. The results are:

Healthy: prusa3d_fw_MK3S_MK3S+_3_10_1_4697.hex
Glitchy: prusa3d_fw_MK3S_MK3S+_3_11_0_4955.hex
Healthy: prusa3d_fw_MK3S_MK3S+_3_11_1-RC1_4987.hex
Healthy: Firmware-MK3S-3.12-6bee4f.hex

arekm, 2022-09-25:
Your bugreport is another indication of prusa3d_fw_MK3S_MK3S+_3_11_0_4955.hex firmware behaving "weird" (in my case it was https://github.com/prusa3d/Prusa-Firmware/issues/3541#issuecomment-1180509587)

leptun, 2022-09-26:
@wavexx Could there be some issue with the cmdqueue? I don't get how the coordinates could get to inf unless:
- Something overwrote the memory (very slim chances of not getting a Inf, NaN being the one many times more likey)
- Some bug with the LA math. Still unlikely since disabling it didn't help
- Some parsing error in the cmdqueue because of an incomplete string or something like that
- Some isr changing current_position since we never guard reading this variable.

I'm still extremely surprised how many tiny glitches there are in 3.11.0. It's not like we will see the issues again anytime soon, especially now that we have the updatd compiler, but it would still be great if we idenitified if this is a bug with the compiler (seen before) or a bug in our code.

wavexx, 2022-09-26:
That's super hard to tell. 3.11 has a ton of issues, several due to issues in locking we fixed, odd isr behavior and a non-zero amount likely introduced by the LTO pass which is why we scrapped gcc 5.4. Debugging that is a sadly dead end unless it's happening right during setup().

In general I highly doubt any part of the planner will write garbage into position/destination unless we're handling power-panic. I suspect corruption, but why exactly is again super-hard to tell using this version. Could be related to cmdqueue, but also not: the UART isr might just be shifting the time to the interrupted instruction somewhere else. You can also corrupt the cmdqueue by writing a single early null anywhere in the buffer resulting in truncated instructions, so you might be seeing a cmdqueue issue due to a bad write completely unrelated to it. I would be cautious about pointing to planner/queue mostly because it's an easy target for random behavior.

It's a completely different story if that still happens with gcc 7.3 though.

leptun, 2022-09-27:
The provided `Barre.tension.trouee.v6_0.1mm_PLA_MK3S_4h27m-no-linear-advance.gcode` prints just fine on my printer with `prusa3d_fw_MK3S_MK3S+_3_11_0_4955.hex` and octoprint

closed 2023-11-16 by github-actions[bot]