prusa3d/Prusa-Firmware #3606: [BUG]Firmware compilation error

oxcarPC, 2022-09-18:
hi guys. I'm trying to build firmware from cloned repository using build.sh script but i'm getting the following error 

temperature.cpp:2340:1: error: body of constexpr function 'constexpr float temp_model::iir_mul(float, float, float, float)' not a return-statement
 }
 ^

I'm using linux debian bullseye 64bit.
What can I do?

leptun, 2022-09-18:
Which avr-gcc version are you using?
Also, try using PF-build.sh as that is the prefered method for builting unless you need something special.

oxcarPC, 2022-09-18:
if I use PF-build.sh the compilation finishes ok but not whit build.sh, I think I'm missing someting whit avg-gcc. How can I get it?

oxcarPC, 2022-09-18:
I just installed gcc-avr on my system. This is the output of avr-gcc -v :

Using built-in specs.
Reading specs from /usr/lib/gcc/avr/5.4.0/device-specs/specs-avr2
COLLECT_GCC=avr-gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/avr/5.4.0/lto-wrapper
Target: avr
Configured with: ../gcc/configure -v --enable-languages=c,c++ --prefix=/usr/lib --infodir=/usr/share/info --mandir=/usr/share/man --bindir=/usr/bin --libexecdir=/usr/lib --libdir=/usr/lib --enable-shared --with-system-zlib --enable-long-long --enable-nls --without-included-gettext --disable-libssp --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=avr CFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-avr-Q3AFj7/gcc-avr-5.4.0+Atmel3.6.2=. -fstack-protector-strong -Wformat ' CPPFLAGS='-Wdate-time -D_FORTIFY_SOURCE=2' CXXFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-avr-Q3AFj7/gcc-avr-5.4.0+Atmel3.6.2=. -fstack-protector-strong -Wformat ' FCFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-avr-Q3AFj7/gcc-avr-5.4.0+Atmel3.6.2=. -fstack-protector-strong' FFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-avr-Q3AFj7/gcc-avr-5.4.0+Atmel3.6.2=. -fstack-protector-strong' GCJFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-avr-Q3AFj7/gcc-avr-5.4.0+Atmel3.6.2=. -fstack-protector-strong' LDFLAGS=-Wl,-z,relro OBJCFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-avr-Q3AFj7/gcc-avr-5.4.0+Atmel3.6.2=. -fstack-protector-strong -Wformat ' OBJCXXFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-avr-Q3AFj7/gcc-avr-5.4.0+Atmel3.6.2=. -fstack-protector-strong -Wformat '
Thread model: single
gcc version 5.4.0 (GCC)

leptun, 2022-09-18:
You should just get the gcc-avr package of your distro. I think you get that error because the MK3 branch now requires avr-gcc 7.3.0. Make sure that is the version that is used when compiling with build.sh. From your log it seems you have 5.4.0.
If PF-build.sh works for you, why use build.sh? We are planning to get rid of that script anyway for 3.13 and switch to cmake fully.

oxcarPC, 2022-09-18:
I use build.sh is because I have a custom Configuration_prusa.h in a Cloned prusa made of my own.
I will try to insall avr-gcc 7.3.0 version.

leptun, 2022-09-18:
You can create a branch for your custom modifications and add your Configuration_prusa.h to the variants folder (with another name). PF-build.sh should pick up on it and show it as an option

oxcarPC, 2022-09-18:
Thanks a lot! It works.

closed 2022-09-18 by oxcarPC