prusa3d/Prusa-Firmware #3610: [ENHANCEMENT]<Allow for steel sheet profile change and babystep info gathering by gCode>

dawidpieper, 2022-09-19:
Enter what type of printer or upgrade the enhancement applies to.
**Printer type** - [-]
**MMU Upgrade** - [-]

**Is your enhancement related to a problem? Please describe.**
The only way blind users can use Prusa printers, as there's no screenreader and its implementation is not possible due to firmware limitations, is via Octoprint, Pronterface or other external software.
Almost any operation is possible. The biggest problem I encounter is the inability to use the steel sheets profiles and check the value of the Live Adjust Z parameter. It can be set with ```G83``` but there is no way to read the current or use profiles.
  
**Describe the solution you'd like**
* When ```G83``` called with no parameters, return information about current babystep and selected profile
* Add option to change profile without overriding its value

If this issue gets accepted, I will be happy to work on this. In fact, I made some ugly and temporary solutions for personal use, so I have a start base.

In hope that you will look favorably on it,
Kind regards,
Dawid Pieper

Prusa-Support, 2022-09-24:
Hello! Thank you for your message. After checking with our devs, unfortunately, they informed me that this specific way, with GCode possibly manipulating sheet profiles, won't be implemented. Closing this specific issue as not planned, but in case of more ideas for enhancements do not hesitate to open new threads with more ideas. 

Alessandro Pantaleo
Prusa Research

closed 2022-09-24 by Prusa-Support

wavexx, 2022-09-30:
@dawidpieper We have a PR related to this: #2201 
It looks like that should do everything you need, and if not that would be the perfect time to have some feedback.