prusa3d/Prusa-Firmware #3611: Please add a an option to keep the extruder and bed hot during a pause.

rokemronnie, 2022-09-20:


neophyl, 2022-09-20:
I was adding the following when the issue was moved -

How Pause is handled is up to the printer firmware.  So any issue raised should be wherever handles your printers firmware.  
For a Mk3 that is https://github.com/prusa3d/Prusa-Firmware/issues  or for a mini https://github.com/prusa3d/Prusa-Firmware-Buddy/issues

If you don't have a Prusa printer then it needs to be with whoever made it.  So what printer do you have rokemronnie ?

Prusa-Support, 2022-09-24:
Thank you for reaching us.

Correct, that seems to be on the printer's firmware side, unless the printer is controlled by a 3rd party host software (e.g. Octoprint...) as explained here https://github.com/prusa3d/Prusa-Firmware/issues/3498#issuecomment-1182092698.

In this case, information about printer type, firmware version, upgrades/mods, and print modality (SD, USB, Octoprint...) may help.

Michele Moramarco
Prusa Research

closed 2023-11-16 by github-actions[bot]