prusa3d/Prusa-Firmware #3623: [BUG] Crash when stopping print sliced as "sequential printing"

Min-Tec, 2022-09-23:
### Description of the bug

When stopping a print which was sliced as "sequential printing" the extruder (nozzle) crashed into another (already printed) part while the printer was moving to its home position.

Gcode which had the problem is attached. The print was stopped when the fourth object started printing (1 to 3 already fully printed).

Could be solved when the z-axis would fully rise to the top before the x- and y-axis would move to their home position.

### Project file & How to reproduce

[Object_0.2mm_ASA_MK3S_2h35m.zip](https://github.com/prusa3d/PrusaSlicer/files/9617548/Object_0.2mm_ASA_MK3S_2h35m.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.5.0+win64

### Operating system

Windows 10

### Printer model

Prusa i3 MK3s

wavexx, 2022-09-25:
This is internally related to #3496. Looks like we need to keep track of the maximum Z seen on the current print to handle both.

ulab, 2023-11-09:
This is still something that should be looked at.

3d-gussner, 2023-11-29:
@ulab Thanks for the reminder, but it is a duplicate and I assigned the #3496 to myself to follow up this request.
Closing it.