prusa3d/Prusa-Firmware #3624: [BUG] FW crash during homing after closing file to print

SethM-3D, 2022-09-24:
**Printer type** - Mk3s
**Printer firmware version** - 3.11.0

**SD card or USB/Octoprint**
  SD card 

**Describe the bug**
  A Firmware crash occurs during homing when a file has been chosen to be printed from the SD card. 
  
**To Reproduce**
  Choose auto home from the menu, while homing is in progress a file from the sd card is chosen. Upon choosing the file the Firmware on the printer crashes, then a reboot occurs.
  
**Expected behavior**
  The file is to be chosen, and the printer starts the printing process 
  
**G-code**
  [coat_hook_0.2mm_PLA_MK3S_35m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware/files/9639479/coat_hook_0.2mm_PLA_MK3S_35m.gcode.zip)

** Log file from serial connection **
[serial log 9:24:2022.txt](https://github.com/prusa3d/Prusa-Firmware/files/9639482/serial.log.9.24.2022.txt)

**Video**
  https://youtu.be/Pl7a0s46020

leptun, 2022-09-26:
@wavexx This crash looks very similar to the autoload-print from SD crash. Does your fix also address this kind of crash of should we just prevent starting a print if any motion is happening when starting the SD print (automated or not)?

wavexx, 2022-12-14:
With #3564 we should now cover all the loading states, as well as wizard-related states that use the eFilamentAction flag. These required special treatment, since enqueue commands as they run.

I think adding a synchronize call before the print start should now resolve all the other cases, but we need to test this.

closed 2023-08-30 by SethM-3D