prusa3d/Prusa-Firmware #3625: [BUG]Firmware 3.13.0 not enabling MMU

ursius2020, 2022-09-25:
**Printer type** - MK3S+
**Printer firmware version** - 3.13.0

**MMU upgrade** - MMU2s
**MMU upgrade firmware version** - 1.0.3

**SD card or USB/Octoprint**
  USB/Pronterface

**Describe the bug**
  After compiling 3.13.0 from source and uploading the Firmware to my MK3S+ the MMU won't start/get enabled
  
**To Reproduce**
  Update MK3S+ to newest Firmware-Build 3.13.0
  
**Expected behavior**
  Comunication between MMU and MK3S+
  
**Output from Pronterface**
`echo:MMU2:>S0*c6.
echo:MMU2:>S0*c6.
echo:MMU2:>S0*c6.
echo:MMU2:<sokok.
echo:MMU2:>S0*c6.
echo:MMU2:<ok.
echo:MMU2:>S0*c6.
echo:MMU2:<ok.
echo:MMU2:>S0*c6.
echo:MMU2:>S0*c6.
echo:MMU2:>S0*c6.
echo:MMU2:>S0*c6.
echo:MMU2:>S0*c6.
echo:MMU2:>S0*c6.
echo:MMU2:>S0*c6.
echo:MMU2:<okokokokokokok.
echo:MMU2:>S0*c6.
echo:MMU2:<ok.
echo:MMU2:>S0*c6.
echo:MMU2:<ok.
echo:MMU2:>S0*c6.
echo:MMU2:<ok.
echo:MMU2:>S0*c6.
echo:MMU2:<ok.
echo:MMU2:>S0*c6.
echo:MMU2:>S0*c6.
echo:MMU2:<okok.
echo:MMU2:>S0*c6.
echo:MMU2:>S0*c6.
echo:MMU2:<okok.
echo:MMU2:>S0*c6.
echo:MMU2:>S0*c6.
echo:MMU2:>S0*c6.
echo:MMU2:<okokok.
echo:MMU2:>S0*c6.
echo:MMU2:>S0*c6.
echo:MMU2:>S0*c6.
echo:MMU2:<okokok.
echo:MMU2:>S0*c6.
echo:MMU2:<ok.
echo:MMU2:>S0*c6.
echo:MMU2:<ok.
echo:MMU2:>S0*c6.
echo:MMU2:>S0*c6.
echo:MMU2:>S0*c6.
echo:MMU2:>S0*c6.
echo:MMU2:>S0*c6.
echo:MMU2:>S0*c6.
echo:MMU2:>S0*c6.
echo:MMU2:<okokokokokokok.
echo:MMU2:>S0*c6.
echo:MMU2:<ok.
echo:MMU2:>S0*c6.
echo:MMU2:<ok.
echo:MMU2:>S0*c6.
echo:MMU2:<ok.
echo:MMU2:>S0*c6.
echo:MMU2:<ok.
echo:MMU2:>S0*c6.
echo:MMU2:<ok.
echo:MMU2:>S0*c6.
echo:MMU2:<ok.
echo:MMU2:>S0*c6.`

ursius2020, 2022-09-25:
@leptun 
@DRracer

DRracer, 2022-09-25:
... that was quite fast :wink: 
Nope, the current MMU FW 1.0.6 will not talk to FW 3.13. There is a brand new FW for the MMU in development, which will be released along with FW 3.13.
Therefore I'm closing this report for now.

closed 2022-09-25 by DRracer

ursius2020, 2022-09-25:
I hope this also adresses the issue MMU not talking back to the printer after an error is resolved. MMU is fine, printer still thinking and displaing MMU needs atention...
Is there a way to test the new FW3.13 and the new MMU-FW 1.0.7?