prusa3d/Prusa-Firmware #3636: [BUG] MK3S+ WITH REVO fails to calibrate new thermal model

darraghbr, 2022-09-29:
**Printer type** - MK3S+ WITH REVO 
**Printer firmware version** -  [3.12.0-BETA1](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.12.0-BETA1)

**Describe the bug**
  Attempting to start the new thermal model calibration WITH REVO (M310 A F1) results in an immediate THERMAL ANOMALY error with a warning tone and the calibration stops. 
  
**To Reproduce**
  Have a MK3S+ with REVO SIX installed, ensure RPi port is disabled, send M503 over USB with Pronterface, then send M310 A F1 and the printer almost immediately gives a THERMAL ANOMALY error and the calibration process stops. 

I did not attempt this calibration without filament loaded, I did have Prusament PETG loaded at the time, 
  
**Expected behavior**
  This was not unexpected, and serves as feedback for the developers.

darraghbr, 2022-09-29:
M503 response:

>>> M503
SENDING:M503
echo:Steps per unit:
echo:  M92 X100.00 Y100.00 Z400.00 E280.00
echo:UStep resolution:
echo: M350 X16 Y16 Z16 E32
echo:Maximum feedrates - normal (mm/s):
echo:  M203 X200.00 Y200.00 Z12.00 E120.00
echo:Maximum feedrates - stealth (mm/s):
echo:  M203 X100.00 Y100.00 Z12.00 E120.00
echo:Maximum acceleration - normal (mm/s2):
echo:  M201 X1000 Y1000 Z200 E5000
echo:Maximum acceleration - stealth (mm/s2):
echo:  M201 X960 Y960 Z200 E5000
echo:Acceleration: P=print, R=retract, T=travel
echo:  M204 P1000.00 R1250.00 T1000.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B0.00 X10.00 Y10.00 Z0.40 E4.50
echo:Home offset (mm):
echo:  M206 X0.00 Y0.00 Z0.00
echo:PID settings:
echo:   M301 P16.80 I3.20 D22.04
echo:PID heatbed settings:
echo:   M304 P126.13 I4.30 D924.76
echo:Retract: S=Length (mm) F:Speed (mm/m) Z: ZLift (mm)
echo:   M207 S3.00 F2700.00 Z0.00
echo:Recover: S=Extra length (mm) F:Speed (mm/m)
echo:   M208 S0.00 F480.00
echo:Auto-Retract: S=0 to disable, 1 to interpret extrude-only moves as retracts or recoveries
echo:   M209 S0
echo:Filament settings: Disabled
echo:Arc Settings: P:Max length(mm) S:Min length (mm) N:Corrections R:Min segments F:Segments/sec.
echo:  M214 P1.00 S0.50 N25 R20 F0
echo:Temperature Model settings:
echo:  M310 I0 R20.50
echo:  M310 I1 Rnan
echo:  M310 I2 Rnan
echo:  M310 I3 Rnan
echo:  M310 I4 Rnan
echo:  M310 I5 Rnan
echo:  M310 I6 Rnan
echo:  M310 I7 Rnan
echo:  M310 I8 Rnan
echo:  M310 I9 Rnan
echo:  M310 I10 Rnan
echo:  M310 I11 Rnan
echo:  M310 I12 Rnan
echo:  M310 I13 Rnan
echo:  M310 I14 Rnan
echo:  M310 I15 Rnan
echo:  M310 P38.00 C12.10 S0 B1 E1.74 W1.20 T-7.00

darraghbr, 2022-09-29:
M310 A F1 response:

>>> M310 A F1
SENDING:M310 A F1
LCD status changed
TM: autotune start
TM: cooling down to 50C
echo:busy: processing
TM: initial C estimation
echo:busy: processing
TM: error |1.280376|>1.200000
LCD status changed
TM: error |1.501300|>1.200000
LCD status changed
TM: error |1.709649|>1.200000
LCD status changed
TM: autotune failed
TM: error |1.874828|>1.200000
LCD status changed
TM: error triggered!
Received command cancel
// action:cancel
Error:Printer stopped due to errors. Supervision required.
[ERROR] Error:Printer stopped due to errors. Supervision required.

TM: error |2.014009|>1.200000
LCD status changed
tmc2130_home_enter(axes_mask=0x04)
TM: error |2.131985|>1.200000
LCD status changed
TM: error |2.211606|>1.200000
LCD status changed
TM: error |2.236551|>1.200000
LCD status changed
tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x04
TM: error |2.213689|>1.200000
LCD status changed
TM: error |2.136456|>1.200000
LCD status changed
TM: error |2.017257|>1.200000
LCD status changed
echo:busy: paused for user
TM: error |1.880923|>1.200000
LCD status changed
TM: error |1.880923|>1.200000
TM: error cleared
echo:busy: paused for user

oroz, 2022-09-29:
Same for me, MK3S with Revo Six

> Send: M310 A F1
Recv: LCD status changed
Recv: TM: autotune start
Recv: TM: initial C estimation
Recv: T:41.7 /230.0 B:43.8 /0.0 T0:41.7 /230.0 @:127 B@:0 P:32.7 A:38.7
Recv: echo:busy: processing
Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
Recv: TM: error |1.244095|>1.200000
Recv: LCD status changed
Recv: TM: error |1.423390|>1.200000
Recv: LCD status changed
Recv: TM: error |1.588214|>1.200000
Recv: LCD status changed
Recv: T:53.7 /230.0 B:43.9 /0.0 T0:53.7 /230.0 @:127 B@:0 P:32.7 A:39.1
Recv: TM: error |1.739812|>1.200000
Recv: LCD status changed
Recv: echo:busy: processing
Recv: TM: autotune failed
Recv: ok
Recv: TM: error |1.886143|>1.200000
Recv: LCD status changed

wavexx, 2022-09-29:
PTC-based heating elements (I'm aware of only the Revo using one at the moment) are known not to work in this beta (https://github.com/prusa3d/Prusa-Firmware/pull/3552#issuecomment-1229441032 for extra details). Nevertheless, we hope to support this and having a few extra traces would be very useful.

In order to submit a trace you should log all the serial output from the printer, with timestamps, while running the calibration. Using octoprint would be the perfect solution as it does everything right out of the box. To initiate a calibration with full log, issue the following commands in sequence:

```
M155 S1 C3
D70 S1
M310 A F0
```

These will run a full calibration without self-check, it should always succeed (and produce copious output). You can save _all_ the output (from M155 to the final result) and attach it here. We have just two samples at the moment from a revo hotend, but it would be great to have around ~20.

darraghbr, 2022-09-30:
> In order to submit a trace you should log all the serial output from the printer, with timestamps, while running the calibration. Using octoprint would be the perfect solution as it does everything right out of the box. To initiate a calibration with full log, issue the following commands in sequence:
> 
> ```
> M155 S1 C3
> D70 S1
> M310 A F0
> ```
> 
> These will run a full calibration without self-check, it should always succeed (and produce copious output). You can save _all_ the output (from M155 to the final result) and attach it here (as a zip file) along with the details of the printer (revo model and any other mods on the printer if any). We have just two samples at the moment from a revo hotend, but it would be great to have around ~20.

[serial.zip](https://github.com/prusa3d/Prusa-Firmware/files/9681903/serial.zip)

Is it just the serial logs that are required? I have uploaded mine please let me know if you want anything different.

Standard MK3S+ E3D Revo Six no other modifications.

oroz, 2022-09-30:
[serial.zip](https://github.com/prusa3d/Prusa-Firmware/files/9682345/serial.zip)

arekm, 2022-09-30:
Log for a newer 24V revo heater (one with blue thermistor wire):

```
M310
M155 S1 C3
D70 I1
M310 A
```

Obviously later M310 S1 and hotend to 200 degree -> error.

[newrevo.log.txt.gz](https://github.com/prusa3d/Prusa-Firmware/files/9682963/newrevo.log.txt.gz)

[copy of https://github.com/prusa3d/Prusa-Firmware/pull/3552#issuecomment-1263200210]

theloukou, 2022-10-01:
Here to confirm the same issue, only difference is i have a beta testing unit of the Revo (then V7), hence why i don't post any serial logs. If you think they would help though, even if i don't have an actual production spec Revo, let me know!

madproducts, 2022-10-08:
Same Issue. 
**M503 Log:**
Recv:16:21:34.898: echo:Steps per unit:
Recv:16:21:34.902: echo: M92 X100.00 Y100.00 Z400.00 E280.00
Recv:16:21:34.902: echo:UStep resolution:
Recv:16:21:34.906: echo: M350 X16 Y16 Z16 E32
Recv:16:21:34.910: echo:Maximum feedrates - normal (mm/s):
Recv:16:21:34.914: echo: M203 X200.00 Y200.00 Z12.00 E120.00
Recv:16:21:34.918: echo:Maximum feedrates - stealth (mm/s):
Recv:16:21:34.919: echo: M203 X100.00 Y100.00 Z12.00 E120.00
Recv:16:21:34.923: echo:Maximum acceleration - normal (mm/s2):
Recv:16:21:34.927: echo: M201 X1000 Y1000 Z200 E5000
Recv:16:21:34.931: echo:Maximum acceleration - stealth (mm/s2):
Recv:16:21:34.935: echo: M201 X960 Y960 Z200 E5000
Recv:16:21:34.939: echo:Acceleration: P=print, R=retract, T=travel
Recv:16:21:34.939: echo: M204 P1000.00 R1250.00 T1000.00
Recv:16:21:34.955: echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s), Z=maximum Z jerk (mm/s), E=maximum E jerk (mm/s)
Recv:16:21:34.959: echo: M205 S0.00 T0.00 B0.00 X8.00 Y8.00 Z0.40 E4.50
Recv:16:21:34.963: echo:Home offset (mm):
Recv:16:21:34.968: echo: M206 X0.00 Y0.00 Z0.00
Recv:16:21:34.968: echo:PID settings:
Recv:16:21:34.972: echo: M301 P22.40 I4.36 D28.75
Recv:16:21:34.972: echo:PID heatbed settings:
Recv:16:21:34.976: echo: M304 P126.13 I4.30 D924.76
Recv:16:21:34.980: echo:Retract: S=Length (mm) F:Speed (mm/m) Z: ZLift (mm)
Recv:16:21:34.984: echo: M207 S3.00 F2700.00 Z0.00
Recv:16:21:34.988: echo:Recover: S=Extra length (mm) F:Speed (mm/m)
Recv:16:21:34.988: echo: M208 S0.00 F480.00
Recv:16:21:34.996: echo:Auto-Retract: S=0 to disable, 1 to interpret extrude-only moves as retracts or recoveries
Recv:16:21:35.000: echo: M209 S0
Recv:16:21:35.001: echo:Filament settings: Disabled
Recv:16:21:35.013: echo:Arc Settings: P:Max length(mm) S:Min length (mm) N:Corrections R:Min segments F:Segments/sec.
Recv:16:21:35.013: echo: M214 P1.00 S0.50 N25 R20 F0
Recv:16:21:35.017: echo:Temperature Model settings:
Recv:16:21:35.017: echo: M310 I0 R20.50
Recv:16:21:35.021: echo: M310 I1 Rnan
Recv:16:21:35.021: echo: M310 I2 Rnan
Recv:16:21:35.025: echo: M310 I3 Rnan
Recv:16:21:35.025: echo: M310 I4 Rnan
Recv:16:21:35.029: echo: M310 I5 Rnan
Recv:16:21:35.029: echo: M310 I6 Rnan
Recv:16:21:35.029: echo: M310 I7 Rnan
Recv:16:21:35.033: echo: M310 I8 Rnan
Recv:16:21:35.033: echo: M310 I9 Rnan
Recv:16:21:35.037: echo: M310 I10 Rnan
Recv:16:21:35.037: echo: M310 I11 Rnan
Recv:16:21:35.037: echo: M310 I12 Rnan
Recv:16:21:35.041: echo: M310 I13 Rnan
Recv:16:21:35.041: echo: M310 I14 Rnan
Recv:16:21:35.045: echo: M310 I15 Rnan
Recv:16:21:35.049: echo: M310 P38.00 C12.10 S0 B1 E1.74 W1.20 T-7.00

**M310 A F1 Log:**
Recv:16:22:45.830: LCD status changed
Recv:16:22:45.859: TM: autotune start
Recv:16:22:45.859: TM: initial C estimation
Recv:16:22:49.808: TM: error |1.255663|>1.200000
Recv:16:22:49.808: LCD status changed
Recv:16:22:50.074: TM: error |1.458696|>1.200000
Recv:16:22:50.078: LCD status changed
Recv:16:22:50.344: TM: error |1.574352|>1.200000
Recv:16:22:50.349: LCD status changed
Recv:16:22:50.615: TM: error |1.657928|>1.200000
Recv:16:22:50.619: LCD status changed
Recv:16:22:50.885: TM: autotune failed
Recv:16:22:50.889: TM: error |1.799269|>1.200000
Recv:16:22:50.889: LCD status changed
Recv:16:22:50.893: TM: error triggered!
Recv:16:22:50.893: // action:cancel
Recv:16:22:50.922: Error:Printer stopped due to errors. Supervision required.
Recv:16:22:51.156: TM: error |1.951980|>1.200000
Recv:16:22:51.159: LCD status changed
Recv:16:22:51.274: tmc2130_home_enter(axes_mask=0x04)
Recv:16:22:51.426: TM: error |2.087927|>1.200000
Recv:16:22:51.429: LCD status changed
Recv:16:22:51.696: TM: error |2.204320|>1.200000
Recv:16:22:51.696: LCD status changed
Recv:16:22:51.966: TM: error |2.215564|>1.200000
Recv:16:22:51.967: LCD status changed
Recv:16:22:52.159: tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x04
Recv:16:22:52.163: E-motor current scaling enabled
Recv:16:22:52.356: TM: error |2.189073|>1.200000
Recv:16:22:52.356: LCD status changed
Recv:16:22:52.507: TM: error |2.145591|>1.200000
Recv:16:22:52.507: LCD status changed
Recv:16:22:52.777: TM: error |2.063081|>1.200000
Recv:16:22:52.778: LCD status changed
Recv:16:22:53.048: TM: error |1.952434|>1.200000
Recv:16:22:53.048: LCD status changed
Recv:16:22:53.318: TM: error |1.952434|>1.200000
Recv:16:22:53.318: TM: error cleared

madproducts, 2022-10-09:
Same issue.  Getting THERMAL ANOMALY message ion LCD.
Standard MK3S+ with original E3D Revo Six and no other modifications to the printer.
Serial log from Octoprint. SD card inserted. No MMU installed.
LCD display during logging was "Temp. model autotune" . Stopped logging when display returned to "Prusa i3 MK3S OK" and temperature started dropping back down.

[PrinterDetailsatBoot.txt](https://github.com/prusa3d/Prusa-Firmware/files/9740458/PrinterDetailsatBoot.txt)
[M503.txt](https://github.com/prusa3d/Prusa-Firmware/files/9740459/M503.txt)
[serial.zip](https://github.com/prusa3d/Prusa-Firmware/files/9740512/serial.zip)

wavexx, 2022-10-09:
@theloukou Might still be useful, if you don't mind please the post as well, writing that's a beta unit. I doubt the heating core changed in betas, and I can still use the log to see if that would pass the calibration.

theloukou, 2022-10-10:
Here are my logs as well. if any more info/measurements are needed, please just ask ^_^
MK3, running a beta version of Revo (E3D v7 then)
[serialLog.zip](https://github.com/prusa3d/Prusa-Firmware/files/9748195/serialLog.zip)

Yornik, 2022-10-14:
REvo six heated up first and then ran the calibration

`Send: M310 A F1
Recv: LCD status changed
Recv: TM: autotune start
Recv: TM: cooling down to 50C
Recv: T:215.0 /0.0 B:61.1 /60.0 T0:215.0 /0.0 @:0 B@:0 P:24.7 A:32.7
Recv: echo:busy: processing
Recv: T:213.0 /0.0 B:61.0 /60.0 T0:213.0 /0.0 @:0 B@:2 P:24.8 A:32.9
Recv: echo:busy: processing
Recv: T:209.6 /0.0 B:60.9 /60.0 T0:209.6 /0.0 @:0 B@:22 P:25.0 A:32.9
Recv: echo:busy: processing
Recv: T:206.3 /0.0 B:60.9 /60.0 T0:206.3 /0.0 @:0 B@:27 P:25.1 A:32.8
Recv: echo:busy: processing
Recv: T:202.4 /0.0 B:60.8 /60.0 T0:202.4 /0.0 @:0 B@:30 P:25.1 A:32.6
Recv: echo:busy: processing
Recv: T:198.5 /0.0 B:60.8 /60.0 T0:198.5 /0.0 @:0 B@:35 P:25.1 A:32.8
Recv: echo:busy: processing
Recv: T:195.2 /0.0 B:60.6 /60.0 T0:195.2 /0.0 @:0 B@:52 P:25.2 A:32.8
Recv: echo:busy: processing
Recv: T:191.3 /0.0 B:60.6 /60.0 T0:191.3 /0.0 @:0 B@:38 P:25.2 A:32.9
Recv: echo:busy: processing
Recv: T:187.9 /0.0 B:60.6 /60.0 T0:187.9 /0.0 @:0 B@:38 P:25.1 A:32.8
Recv: echo:busy: processing
Recv: T:184.1 /0.0 B:60.6 /60.0 T0:184.1 /0.0 @:0 B@:27 P:25.2 A:32.8
Recv: echo:busy: processing
Recv: T:180.5 /0.0 B:60.6 /60.0 T0:180.5 /0.0 @:0 B@:32 P:25.2 A:32.8
Recv: echo:busy: processing
Recv: T:177.4 /0.0 B:60.6 /60.0 T0:177.4 /0.0 @:0 B@:22 P:25.1 A:32.8
Recv: echo:busy: processing
Recv: T:173.8 /0.0 B:60.6 /60.0 T0:173.8 /0.0 @:0 B@:18 P:25.1 A:32.6
Recv: echo:busy: processing
Recv: T:170.2 /0.0 B:60.6 /60.0 T0:170.2 /0.0 @:0 B@:18 P:25.0 A:32.5
Recv: echo:busy: processing
Recv: T:167.1 /0.0 B:60.7 /60.0 T0:167.1 /0.0 @:0 B@:0 P:25.1 A:32.8
Recv: echo:busy: processing
Recv: T:164.0 /0.0 B:60.6 /60.0 T0:164.0 /0.0 @:0 B@:8 P:25.0 A:32.7
Recv: echo:busy: processing
Recv: T:160.5 /0.0 B:60.6 /60.0 T0:160.5 /0.0 @:0 B@:11 P:25.1 A:32.6
Recv: echo:busy: processing
Recv: T:157.8 /0.0 B:60.6 /60.0 T0:157.8 /0.0 @:0 B@:16 P:25.1 A:32.6
Recv: echo:busy: processing
Recv: T:155.0 /0.0 B:60.5 /60.0 T0:155.0 /0.0 @:0 B@:31 P:25.2 A:32.6
Recv: echo:busy: processing
Recv: T:151.8 /0.0 B:60.5 /60.0 T0:151.8 /0.0 @:0 B@:16 P:25.0 A:32.7
Recv: echo:busy: processing
Recv: T:149.0 /0.0 B:60.3 /60.0 T0:149.0 /0.0 @:0 B@:45 P:25.2 A:32.5
Recv: echo:busy: processing
Recv: T:146.1 /0.0 B:60.3 /60.0 T0:146.1 /0.0 @:0 B@:29 P:25.0 A:32.4
Recv: echo:busy: processing
Recv: T:143.4 /0.0 B:60.2 /60.0 T0:143.4 /0.0 @:0 B@:50 P:25.1 A:32.7
Recv: echo:busy: processing
Recv: T:140.3 /0.0 B:60.2 /60.0 T0:140.3 /0.0 @:0 B@:45 P:25.1 A:32.7
Recv: echo:busy: processing
Recv: T:137.9 /0.0 B:60.2 /60.0 T0:137.9 /0.0 @:0 B@:46 P:25.0 A:32.5
Recv: echo:busy: processing
Recv: T:135.6 /0.0 B:60.2 /60.0 T0:135.6 /0.0 @:0 B@:26 P:25.0 A:32.5
Recv: echo:busy: processing
Recv: T:132.8 /0.0 B:60.3 /60.0 T0:132.8 /0.0 @:0 B@:12 P:24.9 A:32.6
Recv: echo:busy: processing
Recv: T:130.4 /0.0 B:60.3 /60.0 T0:130.4 /0.0 @:0 B@:12 P:25.1 A:32.7
Recv: echo:busy: processing
Recv: T:127.9 /0.0 B:60.3 /60.0 T0:127.9 /0.0 @:0 B@:5 P:24.8 A:32.6
Recv: echo:busy: processing
Recv: T:125.7 /0.0 B:60.4 /60.0 T0:125.7 /0.0 @:0 B@:0 P:25.1 A:32.6
Recv: echo:busy: processing
Recv: T:123.5 /0.0 B:60.3 /60.0 T0:123.5 /0.0 @:0 B@:8 P:24.8 A:32.3
Recv: echo:busy: processing
Recv: T:121.0 /0.0 B:60.4 /60.0 T0:121.0 /0.0 @:0 B@:10 P:24.8 A:32.7
Recv: echo:busy: processing
[... 146 lines cut ...]
Recv: TM: initial C estimation
Recv: T:18.8 /230.0 B:18.3 /0.0 T0:18.8 /230.0 @:127 B@:0 P:18.9 A:26.6
Recv: echo:busy: processing
Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
Recv: TM: error |1.447402|>1.200000
Recv: LCD status changed
Recv: T:30.3 /230.0 B:18.7 /0.0 T0:30.3 /230.0 @:127 B@:0 P:19.0 A:26.7
Recv: TM: error |1.637214|>1.200000
Recv: LCD status changed
Recv: TM: autotune failed
Recv: ok
Send: M113 S2
Recv: TM: error |1.877606|>1.200000
Recv: LCD status changed
Recv: TM: error triggered!
Recv: // action:cancel
Cancelling on request of the printer...
Recv: Error:Printer stopped due to errors. Supervision required.
Changing monitoring state from "Operational" to "Error"
Send: M112
Send: N2 M112*35
Send: N3 M104 T0 S0*34
Send: N4 M140 S0*97
Changing monitoring state from "Error" to "Offline after error"
Connection closed, closing down monitor`

theloukou, 2022-10-14:
@Yornik Your fail condition is still the same, you are just waiting for the first cooldown, since you heated up.
As far as i understand, actual data collection starts the heating starts.

zbig-t, 2022-10-14:
Please see my dataset. Retail Revo Six (blue thermistor wires), filament loaded, no other mods that could affect thermal performance.

[serial.zbig-t.log.zip](https://github.com/prusa3d/Prusa-Firmware/files/9786056/serial.zbig-t.log.zip)

wavexx, 2022-10-14:
@Yornik see https://github.com/prusa3d/Prusa-Firmware/issues/3636#issuecomment-1262926997 and see if you can attach a log.

wavexx, 2022-10-14:
Is there anything regarding the thermistor wire colors I should be aware of since it has been mentioned a few times?

zbig-t, 2022-10-14:
> Is there anything regarding the thermistor wire colors I should be aware of since it has been mentioned a few times?

If anything, it's an indicator of whether the Revo Six specimen in question was made quite early after the Revo product launch or is a part of one of the later batches. As per https://e3d-online.com/blogs/news/heatercore-update-wire-they-blue , E3D has changed the wires from white to blue at some point. There were some initial reports of early Revos failing on people with inconsistent/inaccurate thermistor readings and there seem to be less of these as of recently. That might or might not be correlated (not caused, obviously) to the wire colour change.

AliceGrey, 2022-10-26:
Brand new Revo Six with otherwise stock MK3S+ - Both were purchased in the last month. Revo has the new colored cable.

[prusaLog.zip](https://github.com/prusa3d/Prusa-Firmware/files/9871082/prusaLog.zip)

AliceGrey, 2022-10-26:
Somewhat related question: Is it ok for us to use the new beta firmware with the revo six even though the new thermal model isn't working? Or is that not a safe idea?

wavexx, 2022-10-26:
@AliceGrey still perfectly safe. The thermal protection is just a safety feature on top of everything that was already available.

darraghbr, 2022-11-14:
First indications are good on 3.12.0-RC1 my printer goes through the entire TM calibration process when activated through the UI.

I will do it again while capturing the serial output and then I will try some printing and see what I find.

darraghbr, 2022-11-14:
[serial_RC.zip](https://github.com/prusa3d/Prusa-Firmware/files/10006553/serial_RC.zip)

darraghbr, 2022-11-14:
Ok here is the serial logs when I try to heat to 230°C, the first one I call the command from Octoprint, the second one I do through the printer itself (Preheat / PETG) 

[serial_when_heating.zip](https://github.com/prusa3d/Prusa-Firmware/files/10006644/serial_when_heating.zip)

Now I just need to figure out how to reset the thermal model....

wavexx, 2022-11-15:
@darraghbr yes, `M310 S0` just sets the value, then you need to use `M500` to save it.

darraghbr, 2022-11-15:
@wavexx Ah of course, I knew I was forgetting something stupid.

ulab, 2022-11-21:
Thanks for the hint on how to disable it again. Same issue with a Revo Six (old model) here. 

Initial calibration works, any heating then fails with a thermal anomaly.

QR4rigs, 2022-12-01:
I have the same issue, Revo six blue wires and Revo six white wires. 
Thermal Anomaly on heat or cool request, however initiated after a successful cal.

wavexx, 2022-12-08:
If you have some extra, yes please. Just label well the type.

QR4rigs, 2022-12-10:
So sorry, I thought it would be a simple process, but I need a degree in coding it seems to get a logged output. I don't have octoprint nor raspberry pi (yet). I do have Pronterface on Mac (which I'm a newbie on, but it is the only way to get a laptop to the remote printers currently). I have 3 Revo Six installations currently. 1 printer that has a MAXTEMP ERROR even with the stock hotend at anything over 240 degrees so is running 3.12 rc1 with the new thermal model disabled due to the Thermal Anomaly and one that works fine on 3.11 which I can update to 3.12 rc1 quickly.

I really would like Revo Six supported as there is another Mk3S+ incoming next month and the stock hotend is not an option for me with all the nozzle size changes I have to do for various orders.

If anybody has a readme/howto on how to get a logged output with my limited resources, I'm happy to provide them. 
TIA
Paul

wavexx, 2022-12-10:
@QR4rigs are you on windows or linux?

We have several scripts for logging that we used during development, @3d-gussner can help with these. On windows I cannot help much.

QR4rigs, 2022-12-10:
unfortunately, the laptop to hand is on Mac OS, so neither Linux nor Windows :)

ggppjj, 2023-01-03:
I am also having this issue and should be able to contribute logs. Is there a way to disable the hard error tone temporarily for the sake of my neighbors?

catadiel, 2023-01-06:
> PTC-based heating elements (I'm aware of only the Revo using one at the moment) are known not to work in this beta ([#3552 (comment)](https://github.com/prusa3d/Prusa-Firmware/pull/3552#issuecomment-1229441032) for extra details). Nevertheless, we hope to support this and having a few extra traces would be very useful.
> 
> In order to submit a trace you should log all the serial output from the printer, with timestamps, while running the calibration. Using octoprint would be the perfect solution as it does everything right out of the box. To initiate a calibration with full log, issue the following commands in sequence:
> 
> ```
> M155 S1 C3
> D70 S1
> M310 A F0
> ```
> 
> These will run a full calibration without self-check, it should always succeed (and produce copious output). You can save _all_ the output (from M155 to the final result) and attach it here (as a zip file) along with the details of the printer (revo model and any other mods on the printer if any). We have just two samples at the moment from a revo hotend, but it would be great to have around ~20.

MK3S+ with MMU2S and Revo Six (Original Version). Printer inside enclosure.
[serial.log](https://github.com/prusa3d/Prusa-Firmware/files/10359809/serial.log)

ulab, 2023-01-11:
I hope this helps. I wasn't sure when to stop the log :).

[tempcal-revo6-Balu.txt.zip](https://github.com/prusa3d/Prusa-Firmware/files/10392967/tempcal-revo6-Balu.txt.zip)

One thing that irritated me was that I tried doing the calibration from the menu and received a "temperature model error".

alienryes, 2023-01-11:
[serial.zip](https://github.com/prusa3d/Prusa-Firmware/files/10394265/serial.zip)
MK3S+ with Revo Six original. Printer in enclosure.

BootsSiR, 2023-01-11:
> @darraghbr yes, `M310 S0` just sets the value, then you need to use `M500` to save it.

so if I do the following, this will bypass this bug until it's resolved in firmware and then I can just do S1 to re-enable it?

```
M310 S0
M500
```

ksipple, 2023-01-11:
> > @darraghbr yes, `M310 S0` just sets the value, then you need to use `M500` to save it.
> 
> so if I do the following, this will bypass this bug until it's resolved in firmware and then I can just do S1 to re-enable it?
> 
> ```
> M310 S0
> M500
> ```

I've done this to disable the feature, but when the printer is turned on or reset it tells me thermal calibration needs to be run.  It then waits for me to press the button before OctoPrint can connect. Is there a timeout or way to turn the message off?

darraghbr, 2023-01-11:
> 

I haven't found a way to disable that message but I've raised an issue about it here #3894

joshuacant, 2023-01-11:
Here's the log from my MK3S with a Revo that I purchased the day it came out, so it's from the first batch. If additional information is needed, let me know. Will be rolling back to the previous FW until the thermal model works with Revo or the message on boot can be bypassed.

[serial.zip](https://github.com/prusa3d/Prusa-Firmware/files/10395056/serial.zip)

fridge-cat, 2023-01-11:
Brand new Revo Six HeaterCore on MK3S+. Printer inside enclosure.

[serial.zip](https://github.com/prusa3d/Prusa-Firmware/files/10395497/serial.zip)

cnlson, 2023-01-11:
@wavexx @QR4rigs 
1) do you have enough serial outputs now? do you need more?
2) can i execute a gcode to perform the commands to disable? in other words after reset or turn off and on my printer can i just run a gcode file with those commands or will i need to execute them by terminal only?
3) can i revert to 3.11 from 3.12 if it becomes enough of an annoyance. 

I have a mk3 upgraded to s+ with a revo, purchased in october, w/ white (original?) thermistor.

wavexx, 2023-01-11:
> do you have enough serial outputs now? do you need more?

If you ware willing to submit more, please do. I can use each run to test the model offline, so strictly speaking the more the better.

> can i execute a gcode to perform the commands to disable?

See https://github.com/prusa3d/Prusa-Firmware/issues/3891#issuecomment-1379287828 (and the rest of the thread). After a `M310 S0` and `M500` you can keep using 3.12 as before.

> can i revert to 3.11 from 3.12 if it becomes enough of an annoyance.

Yes, also possible.

> also any way to backup and restore live z for all stored plates?

Not in a nice way yet :( #2201
You can save the entire saved state of the printer using `D3`. If you're familiar with command line tools, see https://github.com/prusa3d/Prusa-Firmware/tree/MK3/tools (dump_eeprom in combination with update_eeprom will effectively save/restore everything)

cnlson, 2023-01-11:
> Not in a nice way yet :( #2201 You can save the entire saved state of the printer using `D3`. If you're familiar with command line tools, see https://github.com/prusa3d/Prusa-Firmware/tree/MK3/tools (dump_eeprom in combination with update_eeprom will effectively save/restore everything)

but, will it restore and cause null pointers for the thermal or other settings?

wwsean08, 2023-01-12:
24v Revo nozzle (revo six prusa edition), my room temperature is around 21c and the printer is in an enclosure, with no other mods in place on my mk3s+.

[serial.zip](https://github.com/prusa3d/Prusa-Firmware/files/10398599/serial.zip)

arekm, 2023-01-12:
3.12 works just fine after disabling TM with Revo (and offers the same thermal protection level as 3.11 + fixes in various areas).

What you need to do when using Revo is:
* issue
```
    M310 S0
    M500 
```
commands to disable TM enforcement, once. Remember, you will have old thermal protections active (like in 3.11) even if TM is disabled.

* make sure xyz calibration, z calibration and first layer calibration was already done on your printer and then issue ONLY ONCE
`  D3 Ax0ca6 X1f`
After that don't mess with D3 anymore (see https://github.com/prusa3d/Prusa-Firmware/issues/3891 for details)

That will set "TM calibration done" bit (but also xyz calibration, z calibration and first layer calibration bits as done as there is no command to set single bit unfortunately) to prevent printer from complaining about missing TM calibration every boot.

buliwyf42, 2023-01-12:
Another REVO Log Cheers

[serial.log.zip](https://github.com/prusa3d/Prusa-Firmware/files/10401619/serial.log.zip)

atom-b, 2023-01-12:
Calibration log for my Revo Six with the blue wire thermistor. No filament loaded. I'm using the [Delta-P fan duct mod](https://www.printables.com/model/62523-delta-p-fan-duct-v2-r2-for-mk3s-extruder), if that matters.

[mk3s-tempcal-revo-6.zip](https://github.com/prusa3d/Prusa-Firmware/files/10401817/mk3s-tempcal-revo-6.zip)

bkerler, 2023-01-13:
Here's another log. MK3S+ with MMU2S installed in original Prusa. No other mods, just Revo Six.
[mk3s+_mmu2s_revo6_log.txt](https://github.com/prusa3d/Prusa-Firmware/files/10409611/mk3s%2B_mmu2s_revo6_log.txt)

Yankee106, 2023-01-13:
My LOG
[prusa_3.12.0_revo_six.zip](https://github.com/prusa3d/Prusa-Firmware/files/10409989/prusa_3.12.0_revo_six.zip)

sdh2, 2023-01-13:
Here is my log. MK3s+ with no other mods, no enclosure, Revo Six with blue cable.
[serial.log.zip](https://github.com/prusa3d/Prusa-Firmware/files/10413640/serial.log.zip)

devsfan3, 2023-01-14:
Stock MK3S+ w/ exception of Revo 6 in an enclosure.
[serial.log.zip](https://github.com/prusa3d/Prusa-Firmware/files/10418919/serial.log.zip)

BobKarnes1, 2023-01-17:
MK3S+ with Revo Six log file
[MK3S+Revo log.txt](https://github.com/prusa3d/Prusa-Firmware/files/10430402/MK3S%2BRevo.log.txt)

zeroflow, 2023-01-17:
MK3S with Revo Six
[Delta-P Fan Duct Mod](https://www.printables.com/de/model/62523-delta-p-fan-duct-v2-r2-for-mk3s-extruder)

[serial.zip](https://github.com/prusa3d/Prusa-Firmware/files/10432031/serial.zip)

jdrozdz, 2023-01-20:
- MKS3+
- Revo Six
the rest of the parts were not changed.

[mks3p_revo_six.txt](https://github.com/prusa3d/Prusa-Firmware/files/10465841/mks3p_revo_six.txt)

enicky, 2023-01-20:
MKS3+
Revo (one of the beta units)

[log.txt](https://raw.githubusercontent.com/enicky/prusa/main/log.txt)

3d-gussner, 2023-01-20:
@jdrozdz @enicky Please follow these instructions https://github.com/prusa3d/Prusa-Firmware/issues/3636#issuecomment-1262926997

alexiri, 2023-01-21:
I have an MK3 mostly upgraded to S+ and a Revo with blue thermistor wire.

I set up serial logging in Octoprint (Settings -> Serial Connection -> Serial logging) in order to get full logs with timestamps and then I ran:

```
M155 S1 C3
D70 S1
M310 A F0
```

Much to my surprise, it worked more often than not! The calibration failed when run from the menu, so I was expecting 
this to fail too.

* `worked.log`: first attempt. the initial temperature was ~46º, so I thought maybe that's why it worked.
* `worked2.log`: second attempt from ~ambient, also successful.
* `failed.log`: ran calibration from LCD panel (which apparently runs `M310 A F1`), failed as I've seen in the past.
* `worked3.log`: if running with F0 works, can I save the calibrated model?

In my last attempt where the calibration did not fail, I tried to save the settings with `M310 S1` and `M500`. This doesn't seem to work, when I restarted the printer it continues to tell me that the calibration hasn't been run. However, then I started printing and immediately I got a thermal error. I had to disable the thermal model again (`M310 S0`, `M500`) to be able to keep printing.

Ben-James, 2023-01-22:
Here is my file.  My MMU2S power was disconnected because at first I thought it was the issue.  Revo is the only other upgrade.

[Prusa-w-revo-mmu-power-disconnected.zip](https://github.com/prusa3d/Prusa-Firmware/files/10473463/Prusa-w-revo-mmu-power-disconnected.zip)

turbotrix, 2023-01-22:
I'm having Same issue With Triangle Lab CHC Ceramic heating core on my MK3S+ Very Similar To Revo But Standard Nozzles ...As soon as it starts calibration,within 10 seconds i get "Thermal Anomaly" Error

[221 more replies]

closed 2023-07-28 by 3d-gussner