prusa3d/Prusa-Firmware #3637: [ENHANCEMENT]<Enter comprehensive title>

misko903, 2022-09-30:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

MK3

Please add the PA11 sheet to menu of possible sheets. now only Custom is available.

Thank you :)

Barnabasek, 2022-09-30:
You can rename ANY of the sheets in the menu to your liking. Click the sheet name and select Rename and change the name to PA sheet.

Prusa-Support, 2022-10-05:
> You can rename ANY of the sheets in the menu to your liking. Click the sheet name and select Rename and change the name to PA sheet.

Exactly. Besides this, these sheets are not that popular at the moment, but our developers may consider adding pre-named sheet profiles in the future.

Thank you for your suggestion.

Michele Moramarco
Prusa Research

misko903, 2022-10-06:
I know about the feature - renaming. But my sheet collection is growing and
there is already no reserve.

Even another topic - with Revo becomes nozzle changing too much hidden in
the menu.

Prusa-Support, 2022-12-03:
The Nylon specific sheet profile was added in FW [3.12.0-RC1](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.12.0-RC1).

This issue will be closed as completed now.

Michele Moramarco
Prusa Research

closed 2022-12-03 by Prusa-Support