prusa3d/Prusa-Firmware #3640: [BUG]<Gets stock in Prusa logo>

dlmsound, 2022-09-30:
**Printer type** - MK3S, 
**Printer firmware version** -  [ 3.10.1

**SD card 
   print via SD card 

**Describe the bug**
  I was trying to update to a 3.11. It get stock always in verifying firmware 100%. I try with to many different firmware, cables, PC/Mac,also I try to flash Xloader with no luck , I don’t know if my einsy Rambo is toasted. I’m a little frustrated because is practically a new machine.

  

**Error**
  eading 1##并＃##井##＃#井#＃并#井#＃＃#井并＃井##井#＃井##＃＃＃##＃＃#护＃我##我###1 100号 32.12S
avrdude-slic3r: verifying
avrdude-slic3r: verification
error, first mismatch at byte 0x0100
0x02 != 0x3f
avrdude-slic3r: verification error; content mismatch
avrdude-slic3r done.
Thank you.

dlmsound, 2022-09-30:
[image]
[image]

leptun, 2022-09-30:
Please upload the complete log from the firmware flasher

leptun, 2022-09-30:
From the log it looks like it's failing to verify the first upload stage data. I've never seen this error before. The only way for this to happen is if the hardware is faulty. Please contact Prusa support to get your board replaced.
You also said that you tried other cables, so that rules out a power issue. I don't think xloader works with this microcontroller. Only avrdude works using the "wiring" programmer type.

Did you flash the firmware _many_ times on that board? The board can only be flashed at least 10000 times before the flash starts dying in this way. I also see that you have different fuse settings than standard (hfuse 0xD0 vs the stock 0xD8).

dlmsound, 2022-09-30:
Is there any other way to get fixed. If I buy those main chip and replace it and try to program again it can work again?

dlmsound, 2022-09-30:
[image]

leptun, 2022-09-30:
Only the atmega2560 would need replacing for it to work again. And of course it would need the bootloader flashed again using a programmer.

dlmsound, 2022-09-30:
What kind of bootloader do I need and programmer ?

leptun, 2022-09-30:
Any AVR compatible programmer. It needs to be able to upload using the ICSP header.
The bootloader used is https://github.com/prusa3d/Arduino_Boards/blob/master/IDE_Board_Manager/prusa3dboards/bootloaders/prusa_einsy_rambo/stk500boot_v2_mega2560.hex
It is better if the Arduino IDE is used for "burning the bootloader" since that also sets the correct fuse settings.

Since this issue is not related to the 8bit firmware, I will be closing the issue. Please consider either contacting Prusa support if this is an original board or asking for technical help on the forum (for example, here: https://forum.prusa3d.com/forum/original-prusa-i3-mk3s-mk3-hardware-firmware-and-software-help/)

closed 2022-09-30 by leptun

Prusa-Support, 2022-10-01:
Thanks for bringing this to our attention.

You can reach the official Prusa Technical Support either via email or Live Chat.
https://help.prusa3d.com/article/customer-support_2287

I personally recommend writing an email to info@prusa3d.com, with reference to this issue (Prusa-Firmware/issues/3640), to my attention (Michele Moramarco).

We may want to start with installing the firmware on the Einsy board fully disconnected (except for PSU and computer), and taking this chance to take clear pictures of the Einsy board front and back without any connections.
Please continue via Tech Support.

Michele Moramarco
Prusa Resaerch