prusa3d/Prusa-Firmware #3642: Can not compile hex file

johandutoit1, 2022-10-01:
Arduino: 1.8.5 (Windows 10), Board: "PrusaResearch EINSY RAMBo"

In file included from sketch\mmu2_protocol.h:4:0,

                 from sketch\mmu2_protocol_logic.h:20,

                 from sketch\mmu2.h:3,

                 from sketch\Marlin.h:24,

                 from sketch\BlinkM.cpp:5:

sketch\mmu2_crc.h: In static member function 'static constexpr uint8_t modules::crc::CRC8::CCITT_updateCX(uint8_t, uint8_t)':

mmu2_crc.h:27: error: body of constexpr function 'static constexpr uint8_t modules::crc::CRC8::CCITT_updateCX(uint8_t, uint8_t)' not a return-statement

     }

     ^

sketch\mmu2_crc.h: In static member function 'static constexpr uint8_t modules::crc::CRC8::CCITT_updateW(uint8_t, uint16_t)':

mmu2_crc.h:38: error: body of constexpr function 'static constexpr uint8_t modules::crc::CRC8::CCITT_updateW(uint8_t, uint16_t)' not a return-statement

     }

     ^

In file included from sketch\mmu2_protocol_logic.h:20:0,

                 from sketch\mmu2.h:3,

                 from sketch\Marlin.h:24,

                 from sketch\BlinkM.cpp:5:

sketch\mmu2_protocol.h: In member function 'constexpr uint8_t modules::protocol::RequestMsg::ComputeCRC8() const':

mmu2_protocol.h:62: error: body of constexpr function 'constexpr uint8_t modules::protocol::RequestMsg::ComputeCRC8() const' not a return-statement

     }

     ^

sketch\mmu2_protocol.h: In constructor 'constexpr modules::protocol::RequestMsg::RequestMsg(modules::protocol::RequestMsgCodes, uint8_t)':

sketch\mmu2_protocol.h:71:5: sorry, unimplemented: calling a member function of the object being constructed in a constant expression

     }

     ^

sketch\mmu2_protocol.h: In constructor 'constexpr modules::protocol::RequestMsg::RequestMsg(modules::protocol::RequestMsgCodes, uint8_t, uint16_t)':

sketch\mmu2_protocol.h:82:5: sorry, unimplemented: calling a member function of the object being constructed in a constant expression

     }

     ^

sketch\mmu2_protocol.h: In member function 'constexpr uint8_t modules::protocol::ResponseMsg::ComputeCRC8() const':

mmu2_protocol.h:98: error: body of constexpr function 'constexpr uint8_t modules::protocol::ResponseMsg::ComputeCRC8() const' not a return-statement

     }

     ^

sketch\mmu2_protocol.h: In constructor 'constexpr modules::protocol::ResponseMsg::ResponseMsg(modules::protocol::RequestMsg, modules::protocol::ResponseMsgParamCodes, uint16_t)':

mmu2_protocol.h:108: error: constexpr constructor does not have empty body

     }

     ^

sketch\mmu2_protocol.h: In static member function 'static constexpr uint8_t modules::protocol::Protocol::Char2Nibble(uint8_t)':

mmu2_protocol.h:283: error: body of constexpr function 'static constexpr uint8_t modules::protocol::Protocol::Char2Nibble(uint8_t)' not a return-statement

     }

     ^

sketch\mmu2_protocol.h: In static member function 'static constexpr uint8_t modules::protocol::Protocol::Nibble2Char(uint8_t)':

mmu2_protocol.h:308: error: body of constexpr function 'static constexpr uint8_t modules::protocol::Protocol::Nibble2Char(uint8_t)' not a return-statement

     }

     ^

exit status 1
body of constexpr function 'static constexpr uint8_t modules::crc::CRC8::CCITT_updateCX(uint8_t, uint8_t)' not a return-statement

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

leptun, 2022-10-01:
The MK3 branch can no longer be compiled with gcc 5.4.0 that comes shipped with Arduino 1.8.5. You need to use arduino 1.8.19 as that one has gcc 7.3.0. The compile instructions in README.md still need to be updated.

Also, please note that the MK3 branch is the active development branch, so stuff may be broken there.

johandutoit1, 2022-10-01:
Thank you for the advice, what is the correct branch for the latest stable firmware for the MK3S+ with the MMU2

leptun, 2022-10-01:
Every time a firmware is released, a branch is created for it. At the moment, the latest branch that is considered stable enough for a beta release is https://github.com/prusa3d/Prusa-Firmware/tree/MK3_3.12. Otherwise, https://github.com/prusa3d/Prusa-Firmware/tree/MK3_3.11.1 would be an RC. But I recommend that you use 3.12. The MK3 branch (3.13.0) doesn't have support for the old MMU2S implementation, so if you need that to work, don't use this branch for now.

johandutoit1, 2022-10-02:
Getting this error now

Arduino: 1.8.19 (Windows 10), Board: "PrusaResearch EINSY RAMBo"

Y:\Z\Prusa\Firmware\Prusa-Firmware-MK3_3.12\Firmware\adc.cpp: In function 'void adc_reset()':

adc.cpp:31:58: error: expected ',' before ')' token

     static_assert((1 << first_channel_idx) & ADC_CHAN_MSK);

                                                          ^

adc.cpp:31:58: error: expected string-literal before ')' token

Y:\Z\Prusa\Firmware\Prusa-Firmware-MK3_3.12\Firmware\Marlin_main.cpp: In function 'void gcode_M701()':

Marlin_main.cpp:3639:73: error: expected ',' before ')' token

         static_assert(feed_mm_before_raising <= FILAMENTCHANGE_FIRSTFEED);

                                                                         ^

Marlin_main.cpp:3639:73: error: expected string-literal before ')' token

exit status 1

expected ',' before ')' token

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

wavexx, 2022-10-04:
@3d-gussner from the error this could be an issue with the boards file not setting the correct c++ standard. Can you help here?

leptun, 2022-10-04:
This problem is solved with version 1.0.6 of the Arduino Boards definition for the Einsy Rambo. Please update from 1.0.5 to that and see if it works now.

johandutoit1, 2022-10-04:
Just can not win, see below

 

 

Arduino: 1.8.19 (Windows 10), Board: "PrusaResearch Einsy RAMBo"

 

 

avr-gcc: error: du: No such file or directory

 

avr-gcc: error: Toit\AppData\Local\Arduino15\packages\PrusaResearch\hardware\avr\1.0.6/ldscripts/avr6.xn: No such file or directory

 

exit status 1

 

Error compiling for board PrusaResearch Einsy RAMBo.

 

 

 

This report would have more information with

"Show verbose output during compilation"

option enabled in File -> Preferences.

leptun, 2022-10-05:
Do you have a space character in your username?

3Delights, 2022-10-11:
> Every time a firmware is released, a branch is created for it. At the moment, the latest branch that is considered stable enough for a beta release is [...]

Hmmmm, does that mean there's a new MMU coming out soon?  MMU2S+? or MMU3?

3d-gussner, 2022-10-11:
> Hmmmm, does that mean there's a new MMU coming out soon? MMU2S+? or MMU3?

:eyes: [here](https://github.com/prusa3d/MM-control-01/issues/183#issuecomment-1252101304)

3Delights, 2022-10-11:
Ahh, OK, thanks for clearing that up for me Leptun.  Bit sad now got my hopes up! :(

UnseenAcademical, 2023-03-09:
Possibly related issue: I have downloaded the Arduino IDE 2.0.4 and confirmed that it is running gcc 7.3.0

I acquired the RAMBo files via the Board Manager feature in the ide following these instructions: https://reprap.org/wiki/Rambo_firmware

I've gotten the appropriate firmware for my MK3S+, renamed the file to configuration_prusa.h and opened the Firmware.ino and all related files.

When attempting to verify, I get the following error:

`Prusa-Firmware-MK3_3.12\Firmware\temperature.cpp:2340:1: error: body of constexpr function 'constexpr float temp_model::iir_mul(float, float, float, float)' not a return-statement
 }
 ^

exit status 1

Compilation error: body of constexpr function 'constexpr float temp_model::iir_mul(float, float, float, float)' not a return-statement`

The error points to this block:

`{
    const float a_ = !isnan(a) ? a : nanv;
    return (a_ * (1.f - f)) + (b * f);
}`

I'm assuming that I've got something configured/installed wrong, what should I check next?

Prusa-Support, 2023-08-09:
The new MMU firmware rendered the previous firmware and related builds obsolete.
Moreover, the Arduino building environment will be abandoned (compatible but not supported) in favor of CMake as per the recent [release notes](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.13.0).

Please find more information in [this section](https://github.com/prusa3d/Prusa-Firmware/blob/MK3/README.md#arduino-ide-deprecated) of the README notes.

You can surely keep discussing issues and solutions (here and on the [forum](forum.prusa3d.com)) but, because this build will no longer be maintained by us, please allow us to close this issue as obsolete now.

Michele Moramarco
Prusa Resaearch

closed 2023-08-09 by Prusa-Support