prusa3d/Prusa-Firmware #3645: [BUG]Revo Six M310 A F1 causes an Error with the thermal calibration

Plyn1x, 2022-10-01:
**MK3S+** - [e.g. MK3S, MK3, MK2.5S, MK2.5, MK2S, MK2]
**3.12.0 Beta 2** -  [e.g. 3.8.1, 3.8.1-RC1, ...]

**Describe the bug**
When trying to themaly calibrate the printer it beeps (running a revo six)
M310 A F1 was the command it started
  
**To Reproduce**
Install Revo Six and try to autone it 
  
**Expected behavior**
 Should tune itself 
  
**G-code**
  M310 A F1

Get this error log :
`SENDING:M310 A F1
LCD status changed
TM: autotune start
TM: initial C estimation
TM: error |1.222294|>1.200000
LCD status changed
TM: error |1.411651|>1.200000
LCD status changed
TM: error |1.599723|>1.200000
LCD status changed
TM: autotune failed
TM: error |1.803040|>1.200000
LCD status changed
TM: error triggered!
Received command cancel
// action:cancel
Error:Printer stopped due to errors. Supervision required.
Error:Printer stopped due to errors. Supervision required.

TM: error |1.997395|>1.200000
LCD status changed
tmc2130_home_enter(axes_mask=0x04)
TM: error |2.207648|>1.200000
LCD status changed
TM: error |2.309392|>1.200000
LCD status changed
TM: error |2.375035|>1.200000
LCD status changed
tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x04
TM: error |2.368621|>1.200000
LCD status changed
TM: error |2.340414|>1.200000
LCD status changed
TM: error |2.286039|>1.200000
LCD status changed
echo:busy: paused for user
TM: error |2.181078|>1.200000
LCD status changed
TM: error |2.181078|>1.200000
TM: error cleared
echo:busy: paused for user 
... `

arekm, 2022-10-01:
There is already a bug about that
https://github.com/prusa3d/Prusa-Firmware/issues/3636

Please read it, produce serial.log and attach to that issue. Close this one to avoid duplication. Thanks.

wavexx, 2022-10-09:
Yes please, see the discussion in #3636 and attach a log there if you can so we can have all logs in one place. Closing this issue.

closed 2022-10-09 by wavexx