prusa3d/Prusa-Firmware #3670: [BUG] 3.11. Cooling Fan Stayes on while filament stucked an printer asks to remove and reinsert filament

GMeyerGMeyer, 2022-10-07:
--> MK3S, 3.11, no MMU, Sd card, 

I printed a larger model and somehow is forgot to check if my filament is ordered rightli on the role. So 

1. somehow my filament got stuck on the role. 
2. Printer detecet it and asked me to remove filament
3. he parked the printer head above the I3 sign of the prusa metal sheet. 
4. the printing cooling fan was still active. 
5. i was somewhere else, so i detected that message an hour later. 
6. While the printer wanted me to unload filament, the front fan was all the time engaged. Thus, it was cooling down my printed element, which was, unfortunatley on that corner lifted up. -> BUG
7. i unloaded filament, reloaded it, continued printing, still happy. 

Wanted Behaviour: If the printer detects that the filament stucks or is empty, ther printer does not print anymore, so 

- please turn off the front cooling fan, as it is not needet anylonger (cause we cant currently print) until we reload and restart printing. Thats all :)

Thanks a lot!

[image]

Cheers George

GMeyerGMeyer, 2022-10-19:
Thanks for your answer. 

The filament change came from a non reproducable slicing operation. When i resliced it the next run ( no changed settings) it produced code without the filament change. 

Still would be lovely if on next firmware the fan would stop working when filament change is in progress... 

I was capable of reproducing the source of lifting. Its the temperature sensor. If your environment is a bit cool (18°C), and you print large objects, the Heatbed does not lose as much energy above the sensor than in the corners of the bed. Thus, it heats the center to 60°C, but you have lower temperatures in the corners. When changing bed temperature to 68°C it worked all well.

Prusa-Support, 2022-10-29:
Interesting how the color change occurred unintentionally. It would be interesting to study if reproducible. It may have been the case of an accidental click in Prusa Slicer though (it happened to me).

Exactly. The heat dissipation is stronger near the edges.
I think the heat dissipation is even slightly faster in the case of the Textured and Satin sheets compared to the Smooth sheet. Also, the temperature is only probed in the middle and the heater circuit is subdivided into three zones (non-concentric) - it wouldn't be possible to provide the perimetrical area with a higher power, that is not necessary though. 
The temperature gradient is usually negligible, but increasing the temperature is something I often recommend to be on the safe side, in case of large prints; plus, possibly use the most convenient sheet according to the print scenario. Careful though, a too-high bed temperature may cause warping and poor plastic segments blending.

For the curious ones, together with the announced series of "business" printers (farm printers in Dubai Expo 2020 and Prusa XL), we introduced a modular heatbed made of individually controlled 9×9 cm tiles 🙂 

Michele Moramarco
Prusa Research

Prusa-Support, 2022-10-29:
I think the cause and complications of this issue are sorted out, but the issue will stay open so that our developers can consider it as an enhancement to improve the print fan behavior during the filament change.

Michele Moramarco
Prusa Research

GMeyerGMeyer, 2022-11-05:
Hi Michele, 

yes, its been an unintentional click in the slicer. A checkbox would be nice when exporting like: 
"Do you realle want to use multiple colors?"
In Edinc CNC i have those checkboxed for multiple G-Code Features. Very nice, clearyfies my actions... 
[image]

PR #3746 (2022-11-18): M600: Fix bug where fan is not turned off
The firmware saves the current fanspeed before parking, and the fanspeed is restored after unparking. The problem is the fan was never actually paused.

Turning the fan off just before the retraction is consistent with Marlin 2.

Fixes #3670
commit c86c961d (2022-11-18): M600: Fix bug where fan is not turned off

The firmware saves the current fanspeed before parking,
and the fanspeed is restored after unparking. The problem is the fan
was never actually paused.

Fixes #3670
closed 2022-11-18 by DRracer